pbatard/rufus #1409: "Add fixes for old BIOSes" crashes disk creation (VHD) on new win10 build

JuniorJPDJ, 2019-11-14:
Hi,
I just installed Windows 10 1909 and tried to do VHD with installer from rufus.
Enabled EFI+BIOS dual usb, checked fixes, rufus MBR, quick format and unchecked extended label.
I'm trying to do win10 1909 (19H2) bootable VHD image from rufus-downloaded ISO.
When old BIOSes fixes are checked rufus is unable to create bootable image.
When I uncheck it, It works just good. It's probably related with using new windows build to run rufus.

There's a log:
```
Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive1':
● Partition 1 (offset: 1048576, size: 6.0 GB)
● Partition 2 (offset: 6441378304, size: 512 KB)
Opened \\.\PhysicalDrive1 for exclusive write access
Could not get Disk Extents: [0x000001B1] A device which does not exist was specified.
Could not get Disk Extents: [0x000001B1] A device which does not exist was specified.
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive has a Rufus Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 32256, size: 6.0 GB)
● Creating UEFI:NTFS Partition (offset: 6441361920, size: 512 KB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
Closed Windows format prompt
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
FormatExCallback: Received unhandled command 0x0225 - aborting
[0x00000001] Incorrect function. - Retrying...
[0x00000001] Incorrect function. - Retrying...
[0x00000001] Incorrect function. - Retrying...
[0x00000001] Incorrect function. - Retrying...
Format error: An unsupported command was received.
\\?\Volume{00a0be15-0000-0000-007e-000000000000}\ is already mounted as F:
Re-mounted volume as F: after error

Found non-USB removable device 'QEMU QEMU HARDDISK SCSI Disk Device' => Eliminated
Found VHD device 'Microsoft Virtual Disk'
1 device found
No volume information for drive 0x81
Disk type: FIXED, Disk size: 6.4GB, Sector size: 512 bytes
Cylinders: 783, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x00A0BE15
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 6.0 GB (6441329664 bytes)
  Start Sector: 63, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI FAT (0xef)
  Size: 512 KB (524288 bytes)
  Start Sector: 12580785, Boot: No
```

pbatard, 2019-11-14:
Thanks for the report but you truncated your log (the checklist __clearly__ states not to do that), so I'm missing some information that I'd very much like to see. Can you please retry and make sure you post the __full__ log from Rufus?

pbatard, 2019-11-14:
Also can you please post the SHA-1 of your ISO, which is another thing that was __strongly__ suggested from the checklist you ignored... Please understand, even if you have reasons to believe that what is requested from the checklist does not apply to you, the checklist exists for a reason, which is to ensure that I can help YOU in the best possible conditions. The minute you choose to ignore the checklist, you are actually make it harder on yourself to get help. So please don't do that again, okay?

JuniorJPDJ, 2019-11-14:
Of course, sorry.
I truncated the log because I cleared it accidentaly before, but I'll do again.

JuniorJPDJ, 2019-11-14:
```
Rufus x86 v3.8.1580
Windows version: Windows 10 64-bit (Build 18362)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found non-USB removable device 'QEMU QEMU HARDDISK SCSI Disk Device' => Eliminated
0 devices found
Checking for Rufus updates...
Checking release channel...
No new release version found.
Found VHD device 'Microsoft Virtual Disk'
No logical drive found (unpartitioned?)
Found non-USB removable device 'QEMU QEMU HARDDISK SCSI Disk Device' => Eliminated
1 device found
No volume information for drive 0x81
Disk type: FIXED, Disk size: 6.4GB, Sector size: 512 bytes
Cylinders: 783, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 0
Disk ID: 0x00000000
Drive does not have an x86 Master Boot Record
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CCCOMA_X64FRE_PL-PL_DV9'
  Size: 4.9 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Win10_1909_Polish_x64.iso (5.0 GB)

Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'F:' as volume mountpoint
Deleting partitions...
Notice: Could not delete partitions: [0x00000000] The operation completed successfully.
Opened \\.\PhysicalDrive1 for exclusive write access
No logical drive found (unpartitioned?)
Drive does not appear to be partitioned
Analyzing existing boot records...
Drive does not have an x86 Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 32256, size: 6.0 GB)
● Creating UEFI:NTFS Partition (offset: 6441361920, size: 512 KB)
Writing UEFI:NTFS data...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Closed Windows format prompt
FormatExCallback: Received unhandled command 0x0225 - aborting
[0x00000001] Incorrect function. - Retrying...
[0x00000001] Incorrect function. - Retrying...
[0x00000001] Incorrect function. - Retrying...
[0x00000001] Incorrect function. - Retrying...
Format error: An unsupported command was received.
\\?\Volume{0004ef8c-0000-0000-007e-000000000000}\ is already mounted as F:
Re-mounted volume as F: after error

Found VHD device 'Microsoft Virtual Disk'
Found non-USB removable device 'QEMU QEMU HARDDISK SCSI Disk Device' => Eliminated
1 device found
No volume information for drive 0x81
Disk type: FIXED, Disk size: 6.4GB, Sector size: 512 bytes
Cylinders: 783, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x0004EF8C
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 6.0 GB (6441329664 bytes)
  Start Sector: 63, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI FAT (0xef)
  Size: 512 KB (524288 bytes)
  Start Sector: 12580785, Boot: No
Found VHD device 'Microsoft Virtual Disk'
Found non-USB removable device 'QEMU QEMU HARDDISK SCSI Disk Device' => Eliminated
1 device found
No volume information for drive 0x81
Disk type: FIXED, Disk size: 6.4GB, Sector size: 512 bytes
Cylinders: 783, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x0004EF8C
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: NTFS/exFAT/UDF (0x07)
  Size: 6.0 GB (6441329664 bytes)
  Start Sector: 63, Boot: Yes
Partition 2 (UEFI:NTFS):
  Type: EFI FAT (0xef)
  Size: 512 KB (524288 bytes)
  Start Sector: 12580785, Boot: No
```

ISO Checksums:
```
  File: Win10_1909_Polish_x64.iso
CRC-32: df49f19b
   MD4: 6d1fa496da167f8fb2b22cf68e89e923
   MD5: 7366a54759a631d5b80cc53927f28f73
 SHA-1: 7955c8af2af9087bd4af42c234edfb0680f81b80
```
ISO obtained by setting Fido to:
Windows 10 -> 19H2 -> Windows 10 Home/Pro -> Polish -> x64
x86 ISO works good, x64 also works good when fixes for old BIOSes are disabled

pbatard, 2019-11-14:
> `Windows 10 64-bit (Build 18362)`

Hold on, you are running Rufus on Windows 10 19H1?

From the way you described your issue, it looked to me like you were stating that, after you upgraded Windows 10 to 19H2, and ran Rufus on the updated OS, you were experiencing an issue with 19H2 ISOs that you weren't experiencing when trying to use the same ISOs on Windows 19H1.

Can you therefore confirm that you are seeing the same issue regardless of whether the platform you are running Rufus on (I'm not talking about the ISO being used here, which I do understand is 19H2) is 19H1 or 19H2?

I'm actually in the process of upgrading a platform to 19H2 to test what I originally thought your issue was, so it may be a while.

I also don't like that `Notice: Could not delete partitions: [0x00000000] The operation completed successfully.` message you seem to be getting (even if it ultimately says that the operation was successful). Something weird is going on, which I am not sure has anything to do with using a 19H2 ISO (because there's really very little the ISO selected can do to influence partitioning or formatting in Rufus, which are ISO agnostic operations). Instead, it looks like this is an issue when trying to create 3 partitions (main partition + UEFI:NTFS partition + extra partition added when old BIOS fixes is enabled), that has very little to do with the current ISO.

It may a bug I introduced recently (I overhauled the partition creation not that long ago, and it's possible I didn't properly test the 3-partitions scenario above), but I also should state that, despite what many people seem to think, the "fixes for old BIOSes" is hardly the magic bullet they are hoping for, and mostly exists for the very few people who are using Rufus to create a drive they plan to boot on a pre year 2000 system (when I say "old", I mean  "old"). Plus that is mostly for DOS boot issues, not Windows boot issues.

I seriously doubt you are going to be able to install Windows 10 on any pre-2000 system, so, really, you shouldn't be using that option in the first place, unless you have some very compelling reason to do so.

Still, I'll look into this issue to see what I can find. But it definitely shouldn't be s showstopper. And, by the looks of it, I don't believe it has anything to do with using the latest Windows 10 ISO, because there's really no logic where the ISO being selected could have an incidence on the parameters Rufus uses to partition and format a drive, apart from possibly the 19H1 ISO using an `install.wim` that is smaller than 4 GB (in which case Rufus won't need to create a UEFI:NTFS partition) and 19H2 having an `install.wim` larger than 4 GB, in case the difference is that, if you don't select NTFS as the file system for 19H1, it will create only 2 partitions instead of 3, hence, if it's a leftover bug with extra partition handling, it may look like it's ISO related (but then you would see the exact same issue with 19H1 if using UEFI as the target system and NTFS as the file system).

I'll update this issue when I've had a chance to test it and have more data.

pbatard, 2019-11-14:
Okay, I tested and replicated your issue. A few things:
- As I suspected, this is __not__ related to the ISO used as I was able to replicate the issue with 1903 and other ISOs
- It only happens when the target is a VHD. When the target is a regular USB drive, I see no issue
- Dual BIOS + UEFI must be enabled, so that MBR partitioning can be used along with UEFI:NTFS, which also seriously limits the number of people this issue is likely to affect.

Considering that, as I explained, it makes little sense to use "Add fixed for old BIOSes" when using a VHD and even more so when installing Windows 10, and that you really have to trigger a very specific set of conditions that are unlikely to affect many people, I'm going to seriously lower the priority of my looking into this issue.

That doesn't mean that I'm not going to look at it ever, but, obviously, my focus is on features and problems that are likely to affect a lot of Rufus users and this is certainly not one of them when, again, it makes absolutely no sense to enable _Add fixes for old BIOSes_ in the scenario you encountered the issue with.

&rarr; Deferred.

JuniorJPDJ, 2019-11-14:
Oh sorry, you are right. I'd to miss-install older ISO on VM, I tough I installed 1909.
Thank you for feedback and you are right that issue is not high priority :)
It couldn't delete partitions bcs.. there was no partitions. It was newly created VHD.

DurtyDev, 2020-03-26:
just was reading and noticed that qemu was mentioned. i happened to download qemu yesterday as I was gonna use  it instead of virtualbox, well installer is not signed and digital signature of certificate has problems and considered untrusted. Maybe a bad qemu is going around

JuniorJPDJ, 2020-04-07:
Windows is running inside QEMU, not QEMU inside windows ;)
I use linux with qemu-kvm occasionally running windows, it runs great or even better.
But It seems to be offtopic, that's not the best place to write about it.

DurtyDev, 2020-04-11:
Well that certainly was a shocking answer. Maybe all those virus total
detections of Rufus as Trojan/Rat
https://amp.reddit.com/r/techsupport/comments/8te965/rufus_usb_tool_detected_by_windows_defender/
Rufus USB Tool detected by Windows Defender : techsupport
https://malwaretips.com/threads/rufus-3-8-infected-with-malware.95772/
Rufus 3.8 | Infected with malware? | MalwareTips Community
https://www.microsoft.com/en-us/wdsi/threats/malware-encyclopedia-description?Name=TrojanDropper:BAT/Rufus!lnk&ThreatID=2147753002
TrojanDropper:BAT/Rufus!lnk threat description - Microsoft ...

pbatard, 2020-04-11:
> Maybe all those virus total detections of Rufus as Trojan/Rat

You mean the ones that were __officially declared as false positives__ once they were properly analysed?

> https://malwaretips.com/threads/rufus-3-8-infected-with-malware.95772/

Then you've seen my reply in [that same thread](https://malwaretips.com/threads/rufus-3-8-infected-with-malware.95772/page-2#post-871631), which basically reiterates the same conclusion as the other people in the thread reached (yet another false positive).

> TrojanDropper:BAT/Rufus!lnk threat description

So because it has "Rufus" in its name (which is a common name, and which, even if it is somehow related to this Rufus application, may be related to the many attempts I am seeing by malware authors to repackage their own version of Rufus as a malware application, whilst trying to trick user into believing that it is an official version), you think it means that it is associated with the official release? How about you run Defender against recent versions of Rufus and see for yourself if Defender pops out that one? Because if it doesn't, then clearly something might be wrong in your line of thinking.

Or is "Innocent until proven guilty" something that doesn't apply to software applications, and instead, you should be entitled to use flimsy "evidence" and disproven reports (like the false positives you mention) to try to tarnish someone else's work?

Please try to come back when you have actual evidence that the official version of Rufus is malicious, and don't try to hijack a completely separate thread.

shwegyi198361, 2020-07-03:
> Hi,
> I just installed Windows 10 1909 and tried to do VHD with installer from rufus.
> Enabled EFI+BIOS dual usb, checked fixes, rufus MBR, quick format and unchecked extended label.
> I'm trying to do [...]

pbatard, 2020-07-03:
It's an alignment issue. Microsoft's format functions don't like it when you try to format a partition on a VHD, where the partition start offset isn't aligned to the cluster size (which isn't an issue with regular disks). Probably comes from trying to improve write access speeds to VHDs.

It has __NOTHING__ to do with _"using new windows build to run rufus"_ and I would appreciate if people weren't getting into the pitfall of always assuming, without a shred of evidence (or worse, as in this case, in direct contradiction of evidence that demonstrated otherwise), that Rufus must be incompatible with new versions of Windows...

closed 2020-07-03 by pbatard

JuniorJPDJ, 2020-07-03:
I just thought it's related to new windows bcs I was doing similar thing on earlier version and it worked.
Thank you very much for fix :D 
I really appreciate your work, it's the best part of essential software for any IT guys ;)

commit 500172a7 (2020-07-03): [core] fix format error when trying to use old BIOS fixes with VHDs

* And other improvements
* Closes #1409