pbatard/rufus #1411: rufus on wine

jaeh, 2019-11-18:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I do not need nor expect any fixes, just thought there might be interest in my experience running rufus using wine.
i just started writing the iso to the usb on a windows machine and it is working fine.

Just downloaded rufus (both portable and regular from rufus.ie via https) and started it using:
wine rufus-3.8.exe

i did not expect it to work, because of usb, but was curious.

first thing that greeted me was the
"This Version of Rufus was not produced by it's official developer(s)."
warning popup.
i guess the hash check does not work on linux?

maybe consider adding a warning before that warning that checks for wine and shows a different error that terminates the process?

after i clicked ok (see hashes below) the program started and i could select the iso and rufus read it correctly,
however, as the log below shows, the usb drives did not get detected.

if that usb detection would work, and the wrong warning mentioned above is fixed, everything should be fine when ran on wine.

thanks a lot for keeping this free software,
liked what i read in the faq.

rufus-3.8.exe non portable
windows md5 hash: a0b4e4566aa96b6658b32075492b43b7
linux md5 hash:       a0b4e4566aa96b6658b32075492b43b7  

versions:
wine 4.2
lubuntu 19

Log
---
```
Rufus x86 v3.8.1580 (Portable)
Windows version: Windows Server 2008_R2 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: CoCreateInstance failed; hr = 80040154
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
CoCreateInstance for TaskbarList failed: error 80040154
0 devices found
```

jaeh, 2019-11-18:
oh and i am closing this,
since it is an optional-to-have feature at best.

closed 2019-11-18 by jaeh

pbatard, 2019-11-18:
> first thing that greeted me was the
"This Version of Rufus was not produced by it's official developer(s)."
warning popup.

Well, Wine is clearly missing the APIs that allow us to check if the application is official or not. So that's one of the many limitations of trying to use Wine.

I have no plans to add specific code for Wine, or even investigate it, since Wine is missing way too much of the Windows APIs to enable rufus to run. In other words, doing anything to improve the Wine "user experience" would be pointless at this stage. If anything, I'd rather just add code that detects if Wine is in use and prevents Rufus from launching altogether...

> however, as the log below shows, the usb drives did not get detected.

Yes, that's the thing with Wine. I doubt it will ever have the low level APIs we need for successful enumeration

> if that usb detection would work, and the wrong warning mentioned above is fixed, everything should be fine when ran on wine.

Doubtful, as there probably won't be any `FormatEx` API emulation, so even if you have detection, you won't be able to format drives anyway...

Wine is just too limited as an emulation platform for software like Rufus. It may work fine for regular UI applications that don't need to access much but userspace data, but as soon as you need something a bit more low level, such as the operations needed to detect, partition and format a drive, Wine is pretty much hopeless, because it was never designed to emulate that far. Heck, considering that even ReactOS, which is meant to provide a full Windows API implementation (by also reusing some of Wine's implementation) [can't get Rufus to list devices](https://jira.reactos.org/browse/CORE-14187), it'll probably be long time before you find something that wasn't written by Microsoft with the capability to actually run Rufus.

jaeh, 2019-11-18:
it's very important to have something like rufus (foss usb writer) on windows,
and it will be my goto when creating bootable usb on windows from now on.

also, thank you for elaborating, i used neither windows nor wine as target platforms before, but this confirmed what i assumed.

> I'd rather just add code that detects if Wine is in use and prevents Rufus from launching altogether...

that would be fine imho. maybe log something to bash too, then it should be golden.