pbatard/rufus #1415: NVMe PCIe SSD and NVMe SSD to USB 3 Adapter

johnthecracker, 2019-11-29:
﻿

Checklist
---------
- [ x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [x ] Using a different USB drive.
  - [ x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------

Would it be possibe for Rufus to work with NVMe PCIe SSD on an NVMe SSD to USB 3.1 Adapter ?

I can't provide any log file because Rufus doesn't recognize the device,

Kingston NVMe PCIe SSD KC1000 on "NVMe SSD to USB 3.1 Adapter for PCI-E M.2 2280 SSD"

https://www.ebay.co.uk/itm/NVMe-SSD-to-USB-3-1-Adapter-Converter-For-PCI-E-M-2-2280-SSD-External-Drive-New/273818074864

https://www.kingston.com/us/company/press/article/48938

The result is look of an USB key working on USB3 port to its absolute maximum speed of USB3 port:

```
kc1000
   Sequential Read (Q= 32,T= 1) :   440.540 MB/s
  Sequential Write (Q= 32,T= 1) :   451.346 MB/s
  Random Read 4KiB (Q=  8,T= 8) :   187.670 MB/s [  45817.9 IOPS]
 Random Write 4KiB (Q=  8,T= 8) :   206.386 MB/s [  50387.2 IOPS]
  Random Read 4KiB (Q= 32,T= 1) :   186.887 MB/s [  45626.7 IOPS]
 Random Write 4KiB (Q= 32,T= 1) :   205.448 MB/s [  50158.2 IOPS]
  Random Read 4KiB (Q=  1,T= 1) :    24.431 MB/s [   5964.6 IOPS]
 Random Write 4KiB (Q=  1,T= 1) :    41.365 MB/s [  10098.9 IOPS]
```

It is desirable for program like Rufus to be able to utilize this gadget.

Log
---
```
Rufus doesn't recognize the device, that is the problem
```

pbatard, 2019-11-29:
https://github.com/pbatard/rufus/wiki/FAQ#Help_my_USB_drive_is_not_detected

If you had actually looked at the log, you would have seen a message that pointed you to the answer you sought.

Never assume that the log is useless or won't be needed. __ALWAYS__ attach a log, even if your issue is that the device is not recognized, as even if it displays `0 devices` it contains data that may be very relevant to troubleshooting your issue, and not just for you but for developers as well (for instance, how am I supposed to know what version of Windows or Rufus you are using, which can be exceedingly relevant?).

If your issue is "I'm running Rufus and XYZ happens or does not happen", __ALWAYS__ provide a log. I really can't stress this enough.

Closing this issue as it should be answered. If not, make sure to provide a log.

closed 2019-11-29 by pbatard

johnthecracker, 2019-11-29:
Thank you for the reply.
It was helpful as you pointed me in the right direction.
Selecting "List USB Hard Drives" appeared to solve this issue.
Rufus does keep pace with latest developments in technology.

Here is log file only after I actually found out how to get one
after you insisted it must be somewhere there.

This post will be useful reference to somebody else in the future.

```
Rufus x86 v3.8.1580 (Portable)
Windows version: Windows 10 64-bit (Build 17763)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
0 devices found
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CENA_X64FREE_EN-GB_DV9'
  Size: 4.6 GB (Projected)
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: 18363.418.191007-0143.19h2_release_svc_refresh_CLIENTENTERPRISEEVAL_OEMRET_x64FRE_en-gb.iso (4.6 GB)
0 devices found
Found UAS (USB 3.0) device 'JMicron Tech SCSI Disk Device' (152D:0562)
1 device found
Disk type: FIXED, Disk size: 240 GB, Sector size: 512 bytes
Cylinders: 29185, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x460C8A20
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 223.6 GB (240054697984 bytes)
  Start Sector: 2048, Boot: No
```

[image]