pbatard/rufus #1416: Rufus doesn't allow me to write an image in dd mode if it can't scan it

voltagex, 2019-12-01:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I need an override to allow me to write an image directly in dd mode, even if Rufus doesn't know what the image is. I'm trying to write a uboot/Rockchip loader to my SD card for testing
Log
---
```
Rufus x86 v3.8.1580
Windows version: Windows 10 64-bit (Build 18363)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: OpenLocalMachineGPO failed - error 80004005
Localization set to 'en-US'
Found USB 3.0 device 'Lexar uSD UHS2 RDR USB Device' (05DC:B055)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3891, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {1EF55713-C2E0-44F1-9235-087163D558C4}
Max parts: 128, Start Offset: 17408, Usable = 32010893824 bytes
Partition 1:
  Type: {0FC63DAF-8483-4772-8E79-3D69D8477DE4}
  Name: 'bootloader'
  ID: {EBA7F93E-D660-41F3-B78A-D939034EE403}
  Size: 3.9 MB (4128768 bytes)
  Start Sector: 64, Attributes: 0x0000000000000000
Partition 2:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'boot'
  ID: {509B845C-5703-4F12-B63F-923FB1ECC602}
  Size: 12.0 MB (12566016 bytes)
  Start Sector: 8192, Attributes: 0x0000000000000004

Image provided: '\\debian.local\test\rockchip-u-boot\rksd_loader.img'
Scanning image...
ISO analysis:
libcdio: could not retrieve file info for `\\debian.local\test\rockchip-u-boot\rksd_loader.img': No such file or directory
libcdio: could not retrieve file info for `\\debian.local\test\rockchip-u-boot\rksd_loader.img': No such file or directory
  '\\debian.local\test\rockchip-u-boot\rksd_loader.img' doesn't look like an ISO image
Disk image analysis:
  Could not open image '\\debian.local\test\rockchip-u-boot\rksd_loader.img'
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3891, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {1EF55713-C2E0-44F1-9235-087163D558C4}
Max parts: 128, Start Offset: 17408, Usable = 32010893824 bytes
Partition 1:
  Type: {0FC63DAF-8483-4772-8E79-3D69D8477DE4}
  Name: 'bootloader'
  ID: {EBA7F93E-D660-41F3-B78A-D939034EE403}
  Size: 3.9 MB (4128768 bytes)
  Start Sector: 64, Attributes: 0x0000000000000000
Partition 2:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'boot'
  ID: {509B845C-5703-4F12-B63F-923FB1ECC602}
  Size: 12.0 MB (12566016 bytes)
  Start Sector: 8192, Attributes: 0x0000000000000004
Failed to scan image
Checking for Rufus updates...
Checking release channel...
No new release version found.
Scanning image...
ISO analysis:
  '\\debian.local\test\rockchip-u-boot\rksd_loader.img' doesn't look like an ISO image
Disk image analysis:
  Image does not have an x86 Master Boot Record
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3891, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {1EF55713-C2E0-44F1-9235-087163D558C4}
Max parts: 128, Start Offset: 17408, Usable = 32010893824 bytes
Partition 1:
  Type: {0FC63DAF-8483-4772-8E79-3D69D8477DE4}
  Name: 'bootloader'
  ID: {EBA7F93E-D660-41F3-B78A-D939034EE403}
  Size: 3.9 MB (4128768 bytes)
  Start Sector: 64, Attributes: 0x0000000000000000
Partition 2:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'boot'
  ID: {509B845C-5703-4F12-B63F-923FB1ECC602}
  Size: 12.0 MB (12566016 bytes)
  Start Sector: 8192, Attributes: 0x0000000000000004
Failed to scan image
Scanning image...
ISO analysis:
  'C:\Users\Adam\Desktop\rksd_loader.img' doesn't look like an ISO image
Disk image analysis:
  Image does not have an x86 Master Boot Record
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3891, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {1EF55713-C2E0-44F1-9235-087163D558C4}
Max parts: 128, Start Offset: 17408, Usable = 32010893824 bytes
Partition 1:
  Type: {0FC63DAF-8483-4772-8E79-3D69D8477DE4}
  Name: 'bootloader'
  ID: {EBA7F93E-D660-41F3-B78A-D939034EE403}
  Size: 3.9 MB (4128768 bytes)
  Start Sector: 64, Attributes: 0x0000000000000000
Partition 2:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'boot'
  ID: {509B845C-5703-4F12-B63F-923FB1ECC602}
  Size: 12.0 MB (12566016 bytes)
  Start Sector: 8192, Attributes: 0x0000000000000004
Failed to scan image
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3891, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {1EF55713-C2E0-44F1-9235-087163D558C4}
Max parts: 128, Start Offset: 17408, Usable = 32010893824 bytes
Partition 1:
  Type: {0FC63DAF-8483-4772-8E79-3D69D8477DE4}
  Name: 'bootloader'
  ID: {EBA7F93E-D660-41F3-B78A-D939034EE403}
  Size: 3.9 MB (4128768 bytes)
  Start Sector: 64, Attributes: 0x0000000000000000
Partition 2:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'boot'
  ID: {509B845C-5703-4F12-B63F-923FB1ECC602}
  Size: 12.0 MB (12566016 bytes)
  Start Sector: 8192, Attributes: 0x0000000000000004
Scanning image...
ISO analysis:
  'C:\Users\Adam\Desktop\rksd_loader.img' doesn't look like an ISO image
Disk image analysis:
  Image does not have an x86 Master Boot Record
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3891, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {1EF55713-C2E0-44F1-9235-087163D558C4}
Max parts: 128, Start Offset: 17408, Usable = 32010893824 bytes
Partition 1:
  Type: {0FC63DAF-8483-4772-8E79-3D69D8477DE4}
  Name: 'bootloader'
  ID: {EBA7F93E-D660-41F3-B78A-D939034EE403}
  Size: 3.9 MB (4128768 bytes)
  Start Sector: 64, Attributes: 0x0000000000000000
Partition 2:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'boot'
  ID: {509B845C-5703-4F12-B63F-923FB1ECC602}
  Size: 12.0 MB (12566016 bytes)
  Start Sector: 8192, Attributes: 0x0000000000000004
Failed to scan image
Found USB 3.0 device 'Lexar uSD UHS2 RDR USB Device' (05DC:B055)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3891, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {1EF55713-C2E0-44F1-9235-087163D558C4}
Max parts: 128, Start Offset: 17408, Usable = 32010893824 bytes
Partition 1:
  Type: {0FC63DAF-8483-4772-8E79-3D69D8477DE4}
  Name: 'bootloader'
  ID: {EBA7F93E-D660-41F3-B78A-D939034EE403}
  Size: 3.9 MB (4128768 bytes)
  Start Sector: 64, Attributes: 0x0000000000000000
Partition 2:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'boot'
  ID: {509B845C-5703-4F12-B63F-923FB1ECC602}
  Size: 12.0 MB (12566016 bytes)
  Start Sector: 8192, Attributes: 0x0000000000000004

```

pbatard, 2019-12-01:
There's a convention to indicate that an image is a bootable one, that applies to all architectures (including ARM64), and that is to have a boot marker (`0x55`, `0xAA` at offset `0x1FE`) on your image. All of the ARM64 images I know (e.g. look at Armbian, Raspbian, etc.) follow this.

If the method you used to create your image doesn't do that, then they aren't compliant with established conventions, and you will need to add a step that add the `0x55`, `0xAA` as part of your image generation process, which should be easy to do with `dd`.

If on the other hand, I were to allow images that are not __explicitly__ flagged as bootable, even as a cheat mode, I'd have to waste time trying to help people who are using random images that are simply not bootable. Now, if you still disagree with the idea that bootable images should always have a boot marker, and that an application like Rufus should just do what `dd` does an accept anything, please understand that , as opposed to `dd` (which is designed to move bytes around and that's about it) the purpose of Rufus is first and foremost to create bootable drives, which is a very different mandate. Therefore, by only accepting images that have a boot marker, we are also ensuring that the person who created the image is actually familiar enough with established boot processes, which, even if you may disagree with that as someone who is trying to create a boot image, __is__ actually beneficial for users.

As I believe I have answered the question, I will close this issue.

closed 2019-12-01 by pbatard

voltagex, 2019-12-02:
You are correct. I was missing that marker - I didn't realise it was still used in GPT. Thank you very much for your detailed answer. 
Thanks,
Adam