pbatard/rufus #1417: Created Rufus USB drive not working

esmaeilmirzaee, 2019-12-06:
First and foremost, thank you so much for your effort. I use * Rufus * to create bootable installation Windows 10 but, unfortunately, it does not work on my machine. It is worth mentioning, I am exhausted. I Googled the issue, read the FAQ, disabling/enabling Secure boot. I believe that my machine is `UEFI`-based however I checked all the applicable options and the result is nothing. My machine does not start to boot the windows installer on the USB drive. My machine recognises two USB drive (ie created during the process by Rufus) and non-works.

My machine is 'Asus Transformer Book T100'
Windows os 10 -- 1909

pbatard, 2019-12-06:
https://github.com/pbatard/rufus/wiki/FAQ#Help_I_created_a_Windows_installation_drive_but_it_doesnt_boot

The checklist is __NOT__ optional. You were warned that the issue would be closed if you didn't follow it, therefore I will close it.

closed 2019-12-06 by pbatard

esmaeilmirzaee, 2019-12-07:
Believe it or not, I read that FAQ section twice. Honestly, I don't know what is the solution, and I am unable to change Windows OS.