pbatard/rufus #1418: “Warning: no configuration file found” with latest KNOPPIX DVD-en and Lenovo u410

ilyaza, 2019-12-06:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------

-  Write the latest KNOPPIX DVD-en to an SDXC in a card reader
-  Boot this in Lenovo u410 (Choosing GENERIC STORAGE device 2 from the menu of media)
-  Get: “Warning: no configuration file found” and a “boot: ” prompt.

Log
---
```
<FULL LOG>
Rufus x86 v3.8.1580 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 3.0 device 'Generic STORAGE DEVICE USB Device' (05E3:0732)
Device eliminated because it appears to contain no media
Found USB 3.0 device 'Generic STORAGE DEVICE USB Device' (05E3:0732)
Device eliminated because it appears to contain no media
Found USB 3.0 device 'Generic STORAGE DEVICE USB Device' (05E3:0732)
Found USB 3.0 device 'Generic STORAGE DEVICE USB Device' (05E3:0732)
Device eliminated because it appears to contain no media
Found USB 3.0 device 'Generic STORAGE DEVICE USB Device' (05E3:0732)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'SanDisk Ultra Fit USB Device' (0781:5583)
Found USB 3.0 device 'Seagate Backup+ Hub BK USB Device' (0BC2:AB38)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
2 devices found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7881, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x4032FAA4
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 60.4 GB (64825065472 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/boot/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20150822 (from '/boot/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'KNOPPIX_8'
[... 125 lines cut ...]
Found USB 3.0 device 'Generic STORAGE DEVICE USB Device' (05E3:0732)
Device eliminated because it appears to contain no media
Found USB 3.0 device 'Generic STORAGE DEVICE USB Device' (05E3:0732)
Device eliminated because it appears to contain no media
Found USB 3.0 device 'Generic STORAGE DEVICE USB Device' (05E3:0732)
Found USB 3.0 device 'Generic STORAGE DEVICE USB Device' (05E3:0732)
Device eliminated because it appears to contain no media
Found USB 3.0 device 'Generic STORAGE DEVICE USB Device' (05E3:0732)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'SanDisk Ultra Fit USB Device' (0781:5583)
Found USB 3.0 device 'Seagate Backup+ Hub BK USB Device' (0BC2:AB38)
Device eliminated because it was detected as a Hard Drive (score 33 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
2 devices found
Disk type: Removable, Disk size: 64 GB, Sector size: 512 bytes
Cylinders: 7881, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x4046CB6A
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 60.4 GB (64825065472 bytes)
  Start Sector: 2048, Boot: Yes
```

pbatard, 2019-12-07:
This looks like a pure Knoppix issue with SD card reader support.

Unless you can confirm that you see the same issue when using a USB Flash Drive rather than an SD card, I'm just going to invite you to take this to the Knoppix maintainers, as it just looks like the Knoppix installation scripts are not designed to handle installation from SD, and there's nothing Rufus can (or should) do about that as this is an issue with Knoppix itself.

ilyaza, 2019-12-07:
Well, I remembered that this PC also has an SD card slot.  So I remade the card, and checked it in this slot.

Result: the boot entry in the “BIOS” menu looks like this: 
  USB HDD: Multiple Card  Reader
(so no BS with multiple entries, only one of which active).  However, it still does not boot, the same way as above.

pbatard, 2019-12-07:
Again, what I want to know, are you testing with a USB media on the target PC.

I really don't like all this talk about using an SD card, because there's a good chance that Knoppix is not designed to be installed from an SD card.

Please understand that the message you are getting comes from Knoppix, therefore we need to eliminate the possibility that it's an SD vs USB Flash Drive issue.

Also, you __did__ see the prompt that asks you to select between ISO and DD mode when creating the drive in Rufus, and that __advises__ you to retry writing the image in DD mode if ISO mode doesn't work. Have you tried that?

pbatard, 2019-12-19:
No update &rarr; Closing

closed 2019-12-19 by pbatard