pbatard/rufus #1420: Persian language on the website should be spelled فارسی

kaysond, 2019-12-15:
There's a typo in the drop down box for language on the rufus website. Wasn't sure where to put this so github issue it is...

Persian, or Farsi, should be spelled فارسی, :) 

See https://en.wikipedia.org/wiki/Persian_language
and also https://github.com/pbatard/rufus/blob/72cfb89154ae5451b517ab94ecaf441da4e39081/res/loc/rufus.loc#L8464

pbatard, 2019-12-15:
Issues with the web site should be reported [here](https://github.com/pbatard/rufus-web/issues).

Also, I literally changed from فارسی to پارسی a few weeks ago (see [here](https://github.com/pbatard/rufus-web/commit/fad0c781c633d97d9eab435f8a7758b6bc94a712)) because a someone (who I assume is a Persian speaker) told me exactly this:

> The correct pronunciation of Persian Language in Persian Alphabet is : پارسی
>
> " فارسی " is the Arabic translation of Persian.

So, no offence here, but can you confirm that you are a native Persian speaker, and if so, provide an explanation as to why other person would say that پارسی is the correct spelling? I don't mind changing the spelling back, but I'd rather not have to do that every other week on account on who is assuring me that _"No, __this__ is the actual proper spelling!"_

If we could settle this matter once and for all, with sources other than Wikipedia, I would grealy appreciate it.

kaysond, 2019-12-16:
Farsi was my first language. So what that person said is sort of true, in that the "فارسی" spelling is Arabic influence. However, its still the technically correct spelling. If you type "persian" into an English->Farsi dictionary, that's more than likely what you'll see. (And is in the three I checked). Out of curiosity, I asked a friend who is more in touch with current Iranian culture. He said you would see  "فارسی" in anything official, and this is why you see it spelled that way on places like Wikipedia. But recently there's been a cultural movement to remove Arabic influence from the language and embrace more "pure" Persian words, hence a lot of changes from "ف" to  "پ" (which is "f" to "p", in case you're wondering). Also, I asked my grandmother, and she said she'd spell it " فارسی ". So I guess its coming from a somewhat political place...

Probably more than you ever wanted to know :)

pbatard, 2019-12-16:
Thanks for clarifying. I think I'm starting to see a bit clearer, because, at the time where I added the language, my original plan was to add it as "Farsi (فارسی)", similar to what I saw 7-zip and other applications do.

But then I looked around and saw that there was a push to call the English translation "Persian" rather than "Farsi", which I guess is also where the "f" to "p" push you indicate above comes from.

Therefore, to be consistent with calling the language "Persian" rather than "Farsi" in English, I think I'm going to keep پارسی, since it appears closer to the direction which I am seeing people trying to currently push the language towards.

Of course, if there is a huge pushback, I'm willing to review those choices. But it looks more consistent to me to use پارسی when one also uses Persian.

I'm also going to reopen this issue, since, even as I'm not planning to alter the website back, I still need to change the Persian spelling in Rufus.

commit 5dce5fb5 (2019-12-27): [loc] update spelling of Persian to پارسی

Closes #1420
closed 2019-12-27 by pbatard