pbatard/rufus #1422: "Error: Write error." when writing a bz2 compressed image directly to SD card.

chazste, 2019-12-19:
﻿

Checklist
---------
- [X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------

I am trying to write an image generated by the Yocto build system to an SD card. If I manually decompress the image first it writes correctly. If I try to let Rufus decompress it I get the above error.

Command line was "rufus-3.8p.exe -g -i core-image-weston-logicpd-imx8mm-kit.sdcard.bz2"

Log
---
```
Rufus x86 v3.8.1580 (Portable)
Windows version: Windows 10 64-bit (Build 18362)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Generic- Compact Flash USB Device' (058F:6361)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Multiple Flash Reader USB Device' (058F:6361)
1 device found
Could not get Disk Extents: [0x000001B1] A device which does not exist was specified.
No volume information for drive 0x83
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 483, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 0
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record

Image provided: 'core-image-weston-logicpd-imx8mm-kit.sdcard.bz2'
Scanning image...
ISO analysis:
  'core-image-weston-logicpd-imx8mm-kit.sdcard.bz2' doesn't look like an ISO image
Disk image analysis:
  Image is a compressed bootable disk image
Using image: core-image-weston-logicpd-imx8mm-kit.sdcard.bz2 (89.2 MB)
Found USB 2.0 device 'Generic- Compact Flash USB Device' (058F:6361)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Multiple Flash Reader USB Device' (058F:6361)
1 device found
No volume information for drive 0x83
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 483, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 0
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Found USB 2.0 device 'Generic- Compact Flash USB Device' (058F:6361)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Multiple Flash Reader USB Device' (058F:6361)
1 device found
No volume information for drive 0x83
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 483, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 0
Disk ID: 0x00000000
Drive has a Zeroed Master Boot Record
Checking for Rufus updates...
Checking release channel...
No new release version found.

Format operation started
Requesting disk access...
Will use 'G:' as volume mountpoint
Deleting partitions...
No partition to delete on disk '\\?\PhysicalDrive3'
Opened \\.\PhysicalDrive3 for exclusive write access
Writing compressed image...
Error: write error - -1 bytes written but 2976 expected
Could not write compressed image: -4
Remounted \\?\Volume{1fd4aeea-2134-11ea-885b-d89ef3499489}\ as G:
\\?\Volume{1fd4aeea-2134-11ea-885b-d89ef3499489}\ is already mounted as G:
Re-mounted volume as G: after error

Found USB 2.0 device 'Generic- Compact Flash USB Device' (058F:6361)
Device eliminated because it appears to contain no media
Found USB 2.0 device 'Multiple Flash Reader USB Device' (058F:6361)
1 device found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 483, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0xE138F949
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 64 MB (67108864 bytes)
  Start Sector: 16384, Boot: No
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 448 MB (469762048 bytes)
  Start Sector: 147456, Boot: No

```

pbatard, 2019-12-19:
I'm gonna need to have access to that image to be able to look into this.

Any chance you can upload it somewhere?

If you don't want to post a public link (in case the image contains data that you don't want to share), you can send an e-mail with the link to pete@akeo.ie.

chazste, 2019-12-19:
Thanks for the quick response. This image is directly built from an NXP provided recipe so there is no problem sharing it. I will just need to make sure I can upload it to Google Drive and send you a link. Now that I know it is supposed to work I will double check with a different reader and/or card. If that does not work I will send you a link.

chazste, 2019-12-20:
I have shared the image file link via email. Hopefully that makes it through.

I have also tested with a different card and reader with the same result.

If you have any ideas on what is going on I can certainly attempt to debug this.

pbatard, 2019-12-20:
Thanks, I got your link. I suspect this is a bug in [Bled](https://github.com/pbatard/bled) which is the library we use for decompressing bzip2, so that's probably where I'll be testing things.

It may be a while before I do that though...

chazste, 2019-12-20:
I've built 3.8 here as well. I agree with your assessment, BLED is getting an INVALID_PARAMETER error back from WriteFile (Windows API error 87). It seems to successfully write a bunch of 4KiB blocks and then fails when writing one that is smaller.

pbatard, 2019-12-20:
Yeah it does look like a minimum block size issue as I've tested using the ` uncompress`  commandline application from Bled, and it was able to extract the whole image without issue:
``` 
D:\Projects\bled\x86_32\Debug>uncompress C:\Downloads\core-image-minimal-imx8mmevk.sdcard.bz2 core.img
Uncompressing: 100.00%
Uncompressed size: 236746752 bytes
```

The thing is we're calling ` bled_uncompress_with_handles()`  in both cases, except in Bled's `uncompress` case, we're writing to a file whereas in Rufus we're writing to a device, so I guess we're hitting either a block alignment or block size issue when it comes to disk device, which we may have not seen until now because most compressed images work with a block size that's a multiple of 512 when compressing (i.e. if you compress something using blocks of 4KB or something, then you logically get a bunch of nice 4KB blocks when decompressing, and you won't see the issue).

This may actually be a real pain to fix or work around, because it means we're gonna have to alter the behaviour of [` transformer_write`](https://github.com/pbatard/bled/blob/master/src/open_transformer.c#L30-L54) (can't switch to using `bled_uncompress_to_buffer()`  as it's only designed to be used with buffers that can hold the full decompressed images, not chunks)...

chazste, 2019-12-24:
OK. I've done some more looking at the code (and in this case the comments which actually point to the issue) as well as the structure of the .bz2 file itself. It appears that Yocto uses a native version of pbzip2 to build this file. In the comments it is mentioned that pbzip2 creates multiple streams inside a single file (basically one stream for each 900000 bytes (not evenly dividable by 512)) So instead of crossing nicely between compression block boundaries in the inner while(1) loop, it has to break out and start a whole new stream with the associated set-up and tear-down. 

One test that I did was to expand and recompress the file using 7-zip file manager. The resulting .bz2 file is a little bit smaller and contains only one stream. Rufus seems to be able to write this file to a USB thumb drive OK.

I agree that fixing this in transformer_write may be a pain. I was looking at a fix a level above that, but that does not look promising either. The handling of End Of Stream is quite disruptive and requires flushing things to disk.

I am OK marking this as "deferred" as this does seem like kind of a weird corner case.

pbatard, 2019-12-24:
Yeah, I think I'm going to defer this for now. The one way I currently see out of this at the moment is to convert the `full_write()` that's invoked as part of `transformer_write()`, and that currently is just an alias to `_write()` to a buffered write that is aware of the target block size.

Since all of the images processed by Rufus should be multiples of 512-bytes (if needed I can try to add an extra check that reports an error to the user if it isn't, though the one thing that makes this a bit more complex is that some compression schemes don't provide the uncompressed size until you've actually gone through the full decompression), we shouldn't have to worry about flushing a partial block at the end of the stream. In other words, `full_write` would basically be _"write as many blocks you can from the data, and keep the remainer into a sector-sized buffer, that you'll reuse with the next call"_.

So I think I'll add a new call to Bled, for users to provide their own version of `full_write()` (since my worry is that Windows may want to see a boundary that is the actual sector size, and forcing 512-bytes might not always work for disks with 4K sectors), and once we have that, we should hopefully be able to properly support the `.bz2` images above in Rufus.

pbatard, 2019-12-26:
I think I should have a solution for this issue with the commits I just pushed (but I can't say I have tested it thoroughly).

If you are recompiling Rufus from source, you should be able to test and report (don't worry about the issue being closed from the commit).

closed 2019-12-26 by pbatard

chazste, 2019-12-27:
Built from source. Caveat I built with VS2017.

My test image seems to write correctly to the card. The ARM board boots and I can log in. So a lot of the bytes are correct :)

Timing observation:
uncompressed write 0:58
bz2 write                   3:18 

I'm going account about 1:18 of that to weirdness with USB on my desktop causing long delays at the start and end of writing. (Virtual Box filter driver?)

I'll keep using it and post again in a few days.

Thanks again for looking into this and fixing it!

pbatard, 2019-12-27:
Thanks for testing.

The much slower write of compressed images is expected. The decompression code comes from busybox and is __not__ optimized at all (no multithreading, no double buffering, etc.) because it is designed first and foremost to be small, so it will be slow. But it has the advantage of being compact and easy to reuse as integrated code, which is precisely what allows me to apply fixes such as the above, that I suspect faster more optimized dedicated decompression libraries would also have needed, since they aren't designed to write straight to block devices. In other words, decompression of images in Rufus is never going to break write speed records because it's only provided as a convenience.

chazste, 2019-12-30:
One final comment on speed. (using an image like I provided)
using 7z to extract from the command line ~7 sec
using BLED uncompress from the command line ~9 sec

As I mentioned my workstation sometimes has weird USB behavior so I created a 600 MB VHD file and mounted that.
Using rufus (built from source still)
Write the uncompressed file ~5 sec
Write the .bz2 file ~60 sec

While I'm not surprised that the final number is > 15 sec I am surprised that it is so much larger than 30 seconds. 

As you said this is a convenience feature so speed may not be such a big issue. I still thought you might be interested in the results.

Thanks again for making this work.

chazste, 2020-01-02:
Since I had a couple of days off. I did look at this a little more. 

Reading your comments for the uncompressed case you assert that doing overlapped I/O is not worth the hassle and might not even be faster. In that case you chose DD_BUFFER_SIZE as 64K.

For the compressed case overlapped I/O should have benefit, but more importantly IOBUF_SIZE is 4K. Changing IOBUF_SIZE 64K gives my test file comparable performance (to USB) regardless of whether I pick the compressed or uncompressed version of the file. ~1 min to write 512MB over HS-USB.

Is there a reason to keep IOBUF_SIZE small? This is local to bunzip2 and the "chunk" size is between 100,000 and 900,000 bytes, so 64K does not seem unreasonable. And it gives good I/O performance.

pbatard, 2020-01-02:
> Is there a reason to keep IOBUF_SIZE small?

There are two: Maintainability and reliability.

The code comes from busybox and as a mere re-user of the code, I'd rather not have to spent my time trying to second guess what the busybox people did to see if I can optimize their work, as this is usually the shortest thing to run into issues or limitations that those guys ran into and that made them decide to go with a specific constant (reliability). And, since the busybox code is a living thing, where bugs are fixed and improvements are applied, the closer I keep my code to the busybox original one, the easier it is to apply those original improvements and bugfixes (maintainability).

This being said, I would also assert that the most likely reason busybox chose to go with small buffers is because their software is usually targeted at running on environments with limited RAM, so their constant choice probably derives from this design consideration more than anything else (but again, short of having participated in the discussion where the constant was chosen, there's no way to be 100% positive about that). As such, I don't really have an issue trying to increase bzip2's `IOBUF_SIZE` and see how that goes. But if I get reports of any issues that hint at being related to increasing the buffer size, I'll revert to the busybox original constant in a heartbeat and that will be the end of it.

If you don't mind, can I ask you to please create a new issue in [bled](https://github.com/pbatard/bled) requesting an enhancement to increase `IOBUF_SIZE` so that it's easier for me to track it?

Once I have applied that change in Bled, I'll update Rufus to use the latest version of the library.

commit 4f0335a4 (2019-12-26): [cmp] fix writing of compressed streams that don't end on sector boundary

* Closes #1422