pbatard/rufus #1423: Make a "Write anyway" option, for images that the program fails to scan

goofyseeker311, 2019-12-20:
test it with Android 10 aosp_arm64-Q "system.img" file, which the program fails to scan

pbatard, 2019-12-20:
Please provide a log as well as a direct download link to the image that fails to scan.

As indicated in the checklist you ignored, failure to produce a log may result in an issue being closed without further notice.

goofyseeker311, 2019-12-20:
How about I just convert this to a feature request, then both log and image
are not needed?

goofyseeker311, 2019-12-20:
The images can be found at (Android 10 GSI):

https://developer.android.com/topic/generic-system-image/
https://developer.android.com/topic/generic-system-image/releases

https://dl.google.com/developers/android/qt/images/gsi/aosp_arm64-exp-QJR1.191112.001-6004257.zip

goofyseeker311, 2019-12-20:
Android 10 does not use a standard Linux file system, even a standard Linux kernel cannot read it.

pbatard, 2019-12-27:
Exact duplicate of #1416.

Google should know better than create bootable images that don't have the `55` `AA` marker, that is the __de facto established way of signifying that an image is bootable__.

If Google decided that they are above established rules, then you need to complain to them as Rufus will never be designed to be a `dd` replacement (where any image is accepted). You may not realize this, but accepting any image would have a lot of negative ramifications.

Closed as _"not my problem"_.

closed 2019-12-27 by pbatard