pbatard/rufus #1429: Bad syslinux download URL in rufus 2.18 (and maybe others)

positron96, 2019-12-29:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Issue description
-----------------
I am using rufus 2.18.1213 on windows XP and tried to burn xubuntu 18.04.3. Rufus said that it only contained 2014 isolinux and tried to download more recent one. It failed with this message:
```
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.03/20171017/ldlinux.sys
Unable to send request: This system's SSL library is too old to be able to access this website.
```
I understand that my windows has no modern cyphers, but when I tried to download these files manually, Firefox gave me **404**, so, even if windows was able to use modern ssl, it would fail.

My proposal is to either backport new working URLs to 2.x branch (sounds like a lot of work, since old SSL would need to be overcame) or just publish links on the official website so that users of old versions of rufus could download them manually. This might also be useful for users of up-to-date versions of rufus and OSes (I remember a case when I did not have good internet on computer running rufus but I was able to pull some files from another PC with internet).

Log
---
```
Rufus version: 2.18.1213 (Portable)
Windows version: Windows XP SP3 32 bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02
System locale ID: 0x0419
Will use default UI locale 0x0419
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'ru-RU'
NOTE: The Russian (Русский) translation requires an update, but the current translator hasn't submitted one. Because of this, some messages will only be displayed in English.
If you think you can help update this translation, please e-mail the author of this application
Found USB 2.0 device 'USB Device' (058F:6387)
Found USB 1.1 device 'Generic USB CF Reader USB Device' (058F:6377)
Device eliminated because it appears to contain no media
Found USB 1.1 device 'Generic USB MS Reader USB Device' (058F:6377)
Device eliminated because it appears to contain no media
Found USB 1.1 device 'Generic USB SD Reader USB Device' (058F:6377)
Device eliminated because it appears to contain no media
Found USB 1.1 device 'Generic USB SM Reader USB Device' (058F:6377)
Device eliminated because it appears to contain no media
1 device found
Could not get Disk Extents: [0x00000015] Устройство не готово.
Could not get Disk Extents: [0x00000015] Устройство не готово.
Could not get Disk Extents: [0x00000015] Устройство не готово.
Could not get Disk Extents: [0x00000015] Устройство не готово.
Disk type: Removable, Disk size: 7.8 Гб, Sector size: 512 bytes
Cylinders: 949, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x004AB500
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 (0x0b)
  Size: 7.3 GB (7810842624 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20171017 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Xubuntu 18.04.3 LTS amd64'
  Size: 1.4  Гб (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
[... 948 lines cut ...]
Using autorun.inf label for drive F: 'Xubuntu 18.04.3 LTS amd64'
Found USB 1.1 device 'Generic USB CF Reader USB Device' (058F:6377)
Device eliminated because it appears to contain no media
Found USB 1.1 device 'Generic USB MS Reader USB Device' (058F:6377)
Device eliminated because it appears to contain no media
Found USB 1.1 device 'Generic USB SD Reader USB Device' (058F:6377)
Device eliminated because it appears to contain no media
Found USB 1.1 device 'Generic USB SM Reader USB Device' (058F:6377)
Device eliminated because it appears to contain no media
1 device found
Could not get Disk Extents: [0x00000015] Устройство не готово.
Could not get Disk Extents: [0x00000015] Устройство не готово.
Could not get Disk Extents: [0x00000015] Устройство не готово.
Could not get Disk Extents: [0x00000015] Устройство не готово.
Disk type: Removable, Disk size: 7.8 Гб, Sector size: 512 bytes
Cylinders: 949, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001F9479
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.3 GB (7810842624 bytes)
  Start Sector: 2048, Boot: Yes

```

pbatard, 2019-12-29:
> or just publish links on the official website so that users of old versions of rufus could download them manually

https://github.com/pbatard/rufus/wiki/Usage-Notes#Where_to_manually_copy_the_SyslinuxGRUB_files_if_Rufus_cant_download_them

Couple things you need to understand:
1. The 2.x branch will not receive any updates, EVER. No backports, no nothing. That's because anybody who is using an OS that hasn't reached __official__ End of Life from Microsoft can use Rufus 3.x.
2. I am no longer providing help or planning to provide help to Windows XP users in any shape or form [as per the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Windows_XP_Windows_Vista_NO_LONGER_SUPPORTED). I already supported that OS __way past official End Of Life__ because I understood that, since Rufus could be used to migrate __away__ from that OS, it put me in a position where I had to go the extra mile. However, that boat has sailed. It has been years now since Windows XP has reached End Of Life, so if you __choose__ to keep using that OS, you are 100% on your own, and you completely forfeit your right to receive help from busy Open Source developers, who already have enough to do with OSes that, as opposed to XP, haven't reached End Of Life.
3. Because the website was subject to obnoxious DDoS attacks, we had to move all the content to GitHub, which means we have no control over the SSL layer being used at all. If GitHub chooses to use SSL algorithms that are incompatible with the native libraries of your OS (Firefox and Chrome work around that by using their own SSL libraries, but of course, it would make no sense to do the same for Rufus, since that would be A LOT of extra work and wasted space just for people who choose to use an OS that is officially no longer supported),

TL;DR: If you choose to use Windows XP, you are 100% on your own.

closed 2019-12-29 by pbatard

positron96, 2019-12-30:
Reasonable. Thanks for response (the wiki page was what I actually seeked) and amazing program. 

My created USB drive worked fine even without downloaded syslinux, btw

pbatard, 2019-12-30:
> My created USB drive worked fine even without downloaded syslinux, btw

Yes, that's because Rufus already embeds a single version of the Syslinux and GRUB bootloaders (usually the latest) that it uses when it can't access the network ones. In most cases (recent ISOs) these should work just fine. But now always (especially in the case of Syslinux since the recent versions of Syslinux are designed in such a manner that they are incompatible with one another), which is why Rufus needs to perform a network lookup to try to locate and download the most appropriate binary, even if it's often the same as the embedded one.

You can actually see the versions of Syslinux and GRUB embedded by Rufus on startup which are the ones that are used as a fallback if the network is inaccessible:

```
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
```