pbatard/rufus #1430: Rufus stalled at "Writing Master Boot Record"

aborb, 2019-12-29:
Hi, I am creating an Ubuntu boot USB using Rufus. However. Rufus has stuck on Writing Master Boot Record for over an hour. Cancelling the process would render the USB unusable. I have disabled all firewall / virus programs.

Log :
>Rufus x86 v3.8.1580
Windows version: Windows 7 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Specific STORAGE DEVICE USB Device' (1F75:0918)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1937, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x00000000
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.8 GB (15938322432 bytes)
  Start Sector: 64, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20171017 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 18.04.3 LTS amd64'
  Size: 1.9 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.03
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: ubuntu-18.04.3-desktop-amd64.iso (1.9 GB)
Downloading https://rufus.ie/files/syslinux-6.03/20171017/ldlinux.sys
Unable to send request: This system's SSL library is too old to be able to access this website.
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.03/ldlinux.sys
Unable to send request: This system's SSL library is too old to be able to access this website.
Could not download the file - will try to use embedded 6.03 version instead
Downloading https://rufus.ie/files/syslinux-6.03/20171017/ldlinux.bss
Unable to send request: This system's SSL library is too old to be able to access this website.
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.03/ldlinux.bss
Unable to send request: This system's SSL library is too old to be able to access this website.
Could not download the file - will try to use embedded 6.03 version instead

>Format operation started
Requesting disk access...
Will use 'D:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive1':
● Partition 1 (offset: 32768, size: 14.8 GB)
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Windows 7 Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 64 sectors
Deleting partitions...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 14.8 GB)
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (FAT32)...
Using cluster size: 8192 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Partition is already FAT32 LBA...
Set bootable USB partition as 0x80
write_sectors: Write error [0x0000045D] The request could not be performed because of an I/O device error.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Using Syslinux MBR
read_sectors: Read error [0x00000079] The semaphore timeout period has expired.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200

What should I do?
[image]

pbatard, 2019-12-29:
Well, you have Read/Write errors and the log reports an I/O error, so you have an issue with that drive anyway. You should cancel it, knowing that any hardware issue you may get afterwards will not have been caused by Rufus, since Rufus already detected a problem.

Most likely, your issue is caused by another application or Windows trying to access the drive while Rufus is trying to write it. You should retry and see what happens.

aborb, 2019-12-31:
There was most likely an issue with the usb I was using. Used a different one and it worked.
Thanks though

closed 2019-12-31 by aborb