pbatard/rufus #1431: I killed 5 USB PEN drives flashing an ISO

weworkhere, 2020-01-01:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I do not know if it is an Rufus or Windows issue, or even an issue in the ISO.
I saw issue 1332.
I tested this on two computers Windows 7 (desktop and a laptop)
I killed 5 USB PEN drives doing this procedure.
Yes, my 5 USB drives were used and tested many many times - never had an issue.
No, I did not performed an check disk (as I said, never had issues with these 5 USB drives)
No, I cannot see them anymore on diskpart ("list disk" only lists my main hard drive)
I tried to plug them on a freeBSD box too - no chance, USB is not recognized at all
I did this on both computers - some USB drive on desktop and others on laptop.

I used Rufus 3.4 for the 1st USB drive and Rufus 3.8 for last 4.

Procedure:
1 - I downloaded FreeNAS ISO (FreeNAS-11.2-U7_1.iso)
2 - I checked SHA256 with Windows tool "certUtil" (not rufus)
3 - for each 5 runs, At 99%, Windows go in a BSoD with STOP 7E error

In all 5 runs, I got same result, a nice BSoD and a dead PEN.

Some USB drives are not recognized by diskpart - I not even get the "USB icon" to eject it.
Some of them, when plug, crash Windows again, with same BSoD.

Since Windows crash, I am not able to send any kind of log.

To finish, although the ISO's SHA256 is correct, i noticed that when extracting it to a directory, 7Zip says "there are some data after the end of the payload data".

Please note, I do not know if it is and Rufus or Windows issue, but the fact is that I do have 5 dead USB drives after trying to flash FreeNAS ISO with Rufus.

Log
---
```
<FULL LOG>
```

pbatard, 2020-01-01:
> after trying to flash FreeNAS

There's your problem. Creating a FreeNAS drive and Windows 7 are incompatible.

IIRC this is because of a Windows 7 bug with backup GPT, on account that FreeNAS uses GPT as its partition scheme and, when you write a GPT image in DD mode on a drive that is not the exact same size as the image, the backup GPT table, which is normally situated very close to the end of the disk, and which Windows 7 attempts to process (instead of simply validating whether it's the same as the main GPT from the beginning of the disk and ignore it if it isn't), causes Windows 7 to crash.

This has been fixed in Windows 8 and later, but if you use Windows 7 or earlier, you will find that simply plugging a drive with a garbage backup GPT may cause a BSOD.

You will find quite a few similar issues if you [search for FreeNAS in the issue tracker](https://github.com/pbatard/rufus/issues?utf8=%E2%9C%93&q=FreeNAS).

Considering that Windows 7 is reaching End of Life in a couple of weeks, and that this is really a Windows bug, I am not currently planning to add a workaround for this in Rufus (which could be done by copying the main GPT to the backup GPT location after writing a DD image).

If you test with Etcher or any other utility, you should find the same issue.

I would also encourage you to complain to the FreeNAS people, because there is very little reason to want to use GPT over MBR for a DD image, and, since this is an issue that will manifest itself regardless of the utility used to write the image, if they had some consideration for the people trying to create a FreeNAS flash drive on Windows 7, they __should__ fix this on their side...

weworkhere, 2020-01-02:
Hello pbatard,

Thank you for your prompt reply.

After reading your answer, I installed a Debian 10 box and used "gdisk" to wipe the 5 USB drives (GPT as you refer). "gdisk" utility presented a message in each first usage:
"
Caution: invalid backup GPT header, but valid main header; regenerating backup header from main header.
"
(I used option 'x' and then 'z' to wipe USB drives)

Now, three USB drives do not crash my Windows 7 and are back to life.
Two of them (one is a "4GB cheap not branded model" and the other a "8GB Verbatim") are still dead - I assume it is a coincidence and they would die anyway in a near future.

 
That said, I thank you for taking the time to explain the issue (which allowed me to rescue 3 of my 5 USB drives).
Since this issue is not related to Rufus, I will close this issue.

closed 2020-01-02 by weworkhere

CyberWeather, 2020-01-02:
I just happen to stop by because (ironically etcher wasn't working) to use RUFUS. And just so happened to see that someone had same issue as me about a year ago [(Issue #118](https://github.com/pbatard/rufus/issues/1188)). Maybe try downgrading in FreeNAS of some sorts? Also yes as @pbatard stated, upgrade to windows 10 when you can.

Just my two cents! ;)
~ Weather

weworkhere, 2020-01-04:
Hello CyberWeather,

I gave up from FreeNAS and I installed XigmaNAS instead.
Yes, in fact Windows 10 is not a bad option, I will definitively upgrade my desktop and laptop.

Thanks for your two cents :)
Paulo