pbatard/rufus #1432: confusing message: "Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings."

sjvudp, 2020-01-02:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I don't understand the relationship between "enabling updates" and "ISO downloads". Actually what does "ISO downloads" refer to? Getting images from the network?

I've omitted the messages after the message in question (a long list of filenames being written which is unrelated to the issue)

Log
---
```
Rufus x86 v3.8.1580
Windows version: Windows 10 64-bit (Build 18363)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0407 (de-DE)
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
...
```

pbatard, 2020-01-02:
> I don't understand the relationship between "enabling updates" and "ISO downloads".

Some people don't like the software checking remote websites on startup.

Both the ISO download feature and the check for update require sending requests to web servers on startup to obtain remote data.

Because Rufus asks, the first time you run it, whether you want to allow it to access remote data, we consider that if someone answered "No", we should disable the ISO download feature as well as the check for update feature.

> Actually what does "ISO downloads" refer to?

https://github.com/pbatard/rufus/wiki/FAQ#Help_I_dont_see_the_ISO_download_button

>  Getting images from the network?

Yup. This allows the download of __official retail Windows ISOs__ straight from Microsoft's servers.

I believe I have answered your question, therefore I will close this issue.

closed 2020-01-02 by pbatard

sjvudp, 2020-01-02:
It seems you missed what stands between the lines:
If 'Check for updates' controls whether any download is allowed, better name it "Allow downloads".
Currently it controls whether to (check and) download updates (to the program itself, I guess), and whether to allow remote images (which need downloading). However (as it seems) it does *not* control whether additional components are downloaded (like a newer syslinux).

So I guess either have _one_ setting for all, or have _three_; having one for two seems confusing at least.

pbatard, 2020-01-02:
>  better name it "Allow downloads".

Nope, that's not what it controls. Syslinux and GRUB downloads are still allowed, so your proposal of "Allow downloads" would be confusing, since even if you chose to say No, downloads would still be requested.

Also, please understand that being pedantically accurate is the luxury of people who don't have to ask 30+ translators to translate every new string they add to the UI (a process that usually takes __months__ of extensive follow up just to get to 90% translations covered, even for a single string).

If I didn't have to care about translations, yeah, I'd probably have tried to add a new option for "Enable ISO download" (along with a tooltip that explains that this options makes Rufus check a small file on the internet on startup). So that's 2 translations + a UI change for a setting that, I will argue, few people will want to set differently than the check for update. It is therefore __much__ smarter to just group it with the check for update, even if a couple complain that it would be more accurate to have a different setting, and move on to spending time on the many other matters that require a developer's attention.

TL;DR: Sorry but what you request is not going to happen. Too much effort for too little benefit.