pbatard/rufus #1436: after making my USB Device bootable - it acts like it does not exist

miloszowi, 2020-01-14:
After successful booting my USB Device via RUFUS, it made the USB not visible everywhere, I can not see it in device manager, through disk part in windows, nowhere. 
I have no idea what happened, this USB was used for years without any problems till booting with Rufus. 
Do you have any ideas how can I restore my USB to make it work again?

gdegliesposti, 2020-01-14:
Hello.
This is a FAQ.
https://github.com/pbatard/rufus/wiki/FAQ#Help_my_USB_drive_is_not_detected

ciao
G

miloszowi, 2020-01-14:
@gdegliesposti - this is not helping either, no USB shows up in the dropdown

pbatard, 2020-01-15:
Can you see it under Linux or MacOS?

Also, what was the exact image you used to write your USB? Some images (e.g. Chrome OS x86) play weird tricks with Windows (Microsoft bug), that can make it look like the flash drive is unusable.

However, it does look to me, from what you described, that your flash drive simply failed and that you will have to replace it...

miloszowi, 2020-01-15:
I've checked in MacOS - without positive results - it is not visible there too, not sure about Linux.
I used Windows 10 image.

I am aware that flash memory has a data retention of about 20 years but it was not even close. I am just wondering how did it come to entirely kill the device.

pbatard, 2020-01-15:
> I am just wondering how did it come to entirely kill the device.

Because hardware failures are random, and don't happen right at the MTBF that the manufacturer specifies... Consumer USB flash drive memory is __a lot__ less reliable than people imagine, and I've seen score after score of users asking themselves _"How can my flash drive be dead when it was almost brand new?"_ (and not just people using Rufus) to be able to state this as a fact.

miloszowi, 2020-01-15:
I've never encountered such a thing previously. 
If it is what you have up in your head - then it definitely should display some kind of notice/error to the end-user because right now I am confused - Rufus did not display anything other than the process went successful.

pbatard, 2020-01-15:
> Rufus did not display anything other than the process went successful.

That's because the failure intervened after Rufus got to write the drive.

It occurred sometime between when you used the drive to boot and when you plugged it back to your computer to reuse it for something else.

There's nothing Rufus can do to alert users about failures that occur __after__ it's done its job.

pbatard, 2020-01-20:
I will close this issue now, as I believe I answered your question.

closed 2020-01-20 by pbatard