pbatard/rufus #1440: bug downloading ldlinux.sys

brunovila, 2020-01-27:
When I try to create a usb with Debian 10, this error appear

```
Using image: firmware-10.2.0-amd64-DVD-1.iso (3.7 GB)
Downloading https://rufus.ie/files/syslinux-6.04/20190226/ldlinux.sys
Unable to send request: This system's SSL library is too old to be able to access this website.
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.sys
Unable to send request: This system's SSL library is too old to be able to access this website.
Could not download the file - cancelling
```
---
```
Rufus x86 v3.8.1580 (Portable)
Windows version: Windows 10 64-bit (Build 18363)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0C0A (es-ES)
Will use default UI locale 0x0C0A
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'es-ES'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 3.0 device 'SanDisk Ultra USB Device' (0781:5581)
Using autorun.inf label for drive E: 'Install Debian GNU/Linux'
1 device found
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 14955, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {5238CE1E-4DFC-449F-AE60-B926DA262603}
Max parts: 128, Start Offset: 17408, Usable = 123010513408 bytes
Partition 1:
  Type: {EBD0A0A2-B9E5-4433-87C0-68B6B72699C7}
  Name: 'Microsoft Basic Data'
  ID: {8C18D271-3B52-4220-A5EC-7DF819EB2626}
  Size: 114.6 GB (123009482240 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20190226 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Debian 10.2.0 amd64 1'
  Size: 3.7 GB (Projected)
  Has a >64 chars filename
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: firmware-10.2.0-amd64-DVD-1.iso (3.7 GB)
Downloading https://rufus.ie/files/syslinux-6.04/20190226/ldlinux.sys
Unable to send request: This system's SSL library is too old to be able to access this website.
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.sys
Unable to send request: This system's SSL library is too old to be able to access this website.
Could not download the file - cancelling
```

pbatard, 2020-01-27:
Didn't follow the checklist (that explicitly tells you to __search__ for similar issues that have been reported, of which there are plenty) &rarr; Closing.

pbatard, 2020-01-27:
See https://github.com/pbatard/rufus/issues?utf8=%E2%9C%93&q=SSL+library+is+too+old

closed 2020-01-27 by pbatard

JonnyTech, 2020-01-27:
>Windows version: Windows 10 64-bit (Build 18363)

@pbatard I thought that only applied to Windows 7 without latest patches. Can you clarify please?

pbatard, 2020-01-27:
It's a system/environmental issue.

The OS is reporting that the system SSL libraries can't access the GitHub website because they are missing the relevant levels of TLS support.

I'm afraid I'm not going to investigate how exactly OP has managed to configure their Windows 10 system (most likely they are missing some updates or, worse, they have manually altered their networking libraries) so that it's not compatible with modern SSL. It's up to them to fix it, as the alleged "fix" I could apply in Rufus (_"But I can access the web site with Chrome or Firefox! Why can't you do the same with Rufus??!"_) would mean completely exploding the size of the download by adding a multi MB SSL library (and maintaining it... You want to maintain a critical security library and make sure vulnerabilities are fixed as they are being reported, coz, on top of everything else, that sound like something that's not going to be time consuming at all...).

If other people with vanilla Windows 10 environments don't have that issue, but OP does, then it's up to OP to find out how they altered their system (or failed to update it) to make it behave in such a manner that it can't access SSL webistes it should be able to access.

Also, OP deliberately chose not to use the checklist, which immediately makes me a lot less inclined to want to help them...