pbatard/rufus #1441: Doesn't work under

vorob1, 2020-01-27:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

vorob1, 2020-01-27:
So i've tried the latest build under Windows XP. Clicking on exe does nothing. Nothing happens. I've downloaded older version 2.7.855 and it works, but it took already 25 min on usb 2.0 flash to create bootable flash. Why?

JonnyTech, 2020-01-27:
Please read the FAQ: https://github.com/pbatard/rufus/wiki/FAQ#Windows_XP_Windows_Vista_NO_LONGER_SUPPORTED

pbatard, 2020-01-27:
2.8 (which you can find [here](https://rufus.ie/downloads/)) is the last ever version of Rufus to support Windows XP. From version 3.0, Rufus does not support Windows XP.

And as @JonnyTech pointed out, I no longer support questions related to Rufus usage with Windows XP.

Closing issue.

closed 2020-01-27 by pbatard