pbatard/rufus #1448: Could not download the file - ldlinux.sys

jorjik, 2020-02-06:
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.
---
```
Rufus x86 v3.8.1580 (Portable)
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0419 (ru-RU)
Will use default UI locale 0x0419
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'ru-RU'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found non-USB removable device ' GOODRAM SCSI Disk Device' => Eliminated
Found USB 2.0 device 'Kingston DataTraveler 2.0 USB Device' (0930:6544)
1 device found
Disk type: Removable, Disk size: 16 Гб, Sector size: 512 bytes
Cylinders: 1884, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x475038F1
Drive has a Windows 7 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.4 GB (15499526144 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20190226 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Kubuntu 19.10 amd64'
  Size: 2.1 Гб (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: kubuntu-19.10-desktop-amd64.iso (2.1 Гб)
Downloading https://rufus.ie/files/syslinux-6.04/20190226/ldlinux.sys
Unable to send request: [0x00000002] Не удается найти указанный файл.
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.sys
Unable to send request: [0x00000002] Не удается найти указанный файл.
Could not download the file - cancelling

```

ALexeyAL1, 2020-02-10:
У меня такая же проблема. Пробовал качать эти файлы с https://mirrors.edge.kernel.org/pub/linux/utils/boot/syslinux/Testing/6.04/
И пытался подбросить руками, но не получилось, все равно Руфус пытается выкачать это сам

pbatard, 2020-02-10:
I'm afraid that if you can't download the files, it's an environmental issue, not a Rufus issue.

Rufus expects both to be able to work in a connected environment, as well as your environment to be properly configured to download files.

As far as I know (from the lack of reports I would get otherwise), up to date vanilla installations of Windows have no trouble downloading the files. Therefore, if you can't download them, it's either that you aren't up to date or altered your Windows configuration in a manner that prevents download (e.g. it could be an antivirus that prevents non whitelisted applications from downloading files).

I'm afraid there's nothing for me to do here, so I will close this issue.

closed 2020-02-10 by pbatard

ALexeyAL1, 2020-02-10:
Дело в том что я пытался руками скачать файл: https://rufus.ie/files/syslinux-6.04/20190226/ldlinux.sys
Но сервер выдает 404 ошибку! Я это сделал на разных компьютерах, но ошибка повторяется!

pbatard, 2020-02-10:
https://github.com/pbatard/rufus/wiki/Usage-Notes#Where_to_manually_copy_the_SyslinuxGRUB_files_if_Rufus_cant_download_them

pbatard, 2020-02-10:
Also be mindful that some 404 are __NORMAL__, because there is a fallback mechanism.

In other words, Rufus first tries to see if the distro needs a __ditsro-specific__ version (e.g. https://rufus.ie/files/syslinux-6.04/20190226/ldlinux.sys) and if that is not the case (404), it falls back to the mainline version (https://rufus.ie/files/syslinux-6.04/ldlinux.sys).

If you get a 404 on distro specific, it needs you need to use mainline. But again, I will not help you further than this because:
1. Rufus expects to be able to work in connected mode.
2. If it can't, you're on your own.

ALexeyAL1, 2020-02-10:
Спасибо! Скачал руками нужные файлы
https://github.com/pbatard/rufus-web/tree/gh-pages/files/syslinux-6.04
Заменил руками [rufus*.exe DIR]\rufus_files\syslinux-6.04\20190226\*
После этого все заработало. С чем это связанно буду разбираться. Но отключение антивируса и фаервола не помогло