pbatard/rufus #1450: just 1 more linux suggestion (please look before dismissing)

Tcll, 2020-02-10:
﻿Checklist
---------
- [✓] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [✓] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [X] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [X] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

I can't run Rufus, so I don't have any logs to give...
but anyways, I've taken the time to show you your well due respect and have read everything needed.

As someone working on `cbin`, a rewrite of `ctypes` for python, I do very much understand the hassles of supporting multiple platforms.
what I'm doing specifically with it, is including wrappers for native base binaries that bundle the functionality to make things more easily accessible across Windows and Linux.
(this would be your kernel32, user32, etc. binaries, except you don't have to research what names are available like you have to with ctypes)

I would include mac support, but I can't build a VM that works as advertised...

while I understand you're tightly integrated with the windows API (I assume user32/gdi32 functions as I'm doing with my own apps)
I think you might be overestimating the workload it would take to write a linux build.

I'm taking your time into consideration with this
but if you could primarily focus on the windows version
and then work on a side port for a linux-native app or appimage
(appimage should only be used for the case you can't make a distro agnostic build and need to bundle the environment)
it would very much be highly appreciated :)

not only that, but as linux is slowly gaining in user share
a linux build is slowly becoming more and more urgent
I understand I might be much more free to do what I want, and have the ability to work on such things when it comes to having the ability of freedom to work on a project...
actually I think that's plainly obvious given how much I have on my plate with all of my projects, and the fact I'm going the many extra miles with them... lol

but yeah, I just want to point out, linux might not be as big of an issue to port over a build as you might think :)
take it from someone who actually does this stuff ;)

I respect whatever you do from here out
but I just hope you'll take another look back and maybe start working on a side build for us :)

pbatard, 2020-02-10:
> I assume user32/gdi32 functions as I'm doing with my own apps

You are assuming wrong. If it was just the UI where the bulk of the effort was required, I'd have produced a Linux port a long time ago.

It's the whole formatting for NTFS, FAT, ReFS, UDF + device detection + partitioning + WIM support that __is__ the issue. We are tightly integrated with the Windows formatting, partitioning and WIM libraries because they are available and avoid having to reinvent the wheel (or find a squared wheel that we need to chip away to make round for our purpose).

Granted, you're going to tell me that (besides ReFS), file system support "should" be easy: Just hook into or invoke mkfs and mount behind the scenes. And likewise, just need to add wimlib for WIM support... But then you get all the corner cases such as:
* Can't invoke `bcdboot.exe` on Linux to finalize Windows To Go. &mdash; Oh well, you're gonna tell me, who needs Windows To Go anyway...
* No PowerShell to run the Windows ISO download script &mdash; Bah, who cares about conveniently downloading Windows ISOs from Rufus... It's not like Linux users would expect the same functionality as the Windows version anyway (or, one can just convert the FIDO PowerShell script to a shell script, coz it shouldn't take that long)
* No convenient `FormatEx` invocation for FAT/NTFS/exFAT/UDF/ReFS support in a single call &mdash; No big deal, just rewrite the whole thing to hook into `mkfs` and `mount`. And besides changing the whole formatting invocation, properly detecting (and validating) whether a system does have exFAT or NTFS formatting support, and reporting that to the user if it doesn't, is definitely not something that's going to take more than a couple of days at best...
* Detecting when a drive is plugged? &mdash; Easy: simply hook into the Linux calls that do the equivalent work from Window, and make sure to hook that to the UI refresh... If you need more than a week to do that, you must be doing it wrong!
* And so on, and so forth... I hope you see where I'm going with this, coz I'm barely scratching the surface here...

Even if it takes 1 month to get a super basic version of Rufus, as opposed to the 4 months minimum that I estimate are needed (and that estimate assumes full days of work, of ~8-10 hours/day), that's one month that I simply can't justify myself allocating when I have tons of other things that I know are going benefit people a lot more than a half-assed version of Rufus hastily put together. I'm actually quite happy to see that the woeusb folks are trying to take over the task of creating Windows drives on Linux because between what most people hope a port of Rufus would be like (something that would be very close to the Windows version in terms of features) and the reality, there would be such a big gap that I'd expect a huge backlash from simply calling that application "Rufus".

> I think you might be overestimating the workload it would take to write a linux build.

I'm not. I've created Windows + Linux + Mac UI applications in the past. Even when using common libraries (e.g. games using OpenGL), you __are__ going to run into all the corner cases (Want sound on Linux? Sure, just use ALSA. After all, there's no good reason why many people claim that using ALSA is akin to diving into the depth of hell. None at all... And if you're going to tell me I'd have been better off using Godot or Unity or whatever, I'll counter that I would have required a whole costly rewrite of the existing code, and that, for all its flaws, going for ALSA was still the less demanding option).

Oh, and for the record, reworking the Rufus UI between 2.18 and 3.0 is something that literally took about 3 months of full time work. And that was a "simple" UI rework that didn't even require switching libraries and stuff or abstracting things around (and no, that was not just done for eye-candy. It was something that was very needed to improve the pain points from UI usage, and that I had to delay for about 4 years before I could allocate the required amount of time... which I found I had seriously underestimated). So, yeah, I don't believe that my estimate of the work needed for a Linux port is an exaggeration, because there really is a lot more happening behind the scenes in Rufus than meets the eyes.

Besides, reuse of a library, even mature ones, is never ever trouble free. For the record, it took me about 2-3 months of extensive work to be able to reuse libcdio in Rufus, and more recently it also took about 2 months to integrate ext2fs for `ext2`/`ext3` formatting... and I still have [__major bugs to fix__](https://github.com/pbatard/rufus/issues/1396)... which I've actually spent the last 3-4 months trying to find a way to get to a point where I could dedicate a measly single week to be able to start properly looking into it. And that's a critical Rufus issue as far as I'm concerned.

So, while, I appreciate that you would like nothing more than for me to have some "time on the side" to start new projects (coz I sure would like that too!!) such as a Linux or Mac port, the reality is very different. I already have a list long like the arm of stuff I'd like to do, provided I could find __and justify__ the time and as such, a project such as a Linux port of Rufus is something that I'd need to plan years in advance, just like the UI redesign, because, after decades of being a software developer, I can tell from experience that this project is not a simple as a matter of being able to allocate a few weeks to work on it __on the side__.

closed 2020-02-10 by pbatard

Tcll, 2020-02-10:
> you're going to tell me that (besides ReFS), file system support "should" be easy

haha, took the words right out of my mouth, fair enough

- > who needs Windows To Go anyway

actually no, because if it were my project, I'd re-invent the wheel to make something work if needed
depending on how complex that may be, that may have to come later, but it would still come regardless.

- > Bah, who cares about conveniently downloading Windows ISOs from Rufus...

this I can agree with as it seems to go too far out of the way for users...
buuuut then again, it's not like I'm not simplifying a 3D modeling API to where even the noobest of noobs can make sense of it...

- > No big deal, just rewrite the whole thing

while this is something I would do, I agree with you here, you shouldn't HAVE to as it's going too far out of your way.

- > and make sure to hook that to the UI refresh...

actually I'd more make this an event callback rather than a complete refresh
sure the callback would still refresh the UI through the event system, but still...

but yeah, I understand you, like I said, I respect whatever you do from here on out
seeing you say you've worked on linux UIs in the past, I'd actually like to retract what I previously said
I can see this is a bigger issue for you than I made it out to be, and you certainly do have your experience to make your decision final.
sorry if I seemed a bit insulting previously, and sorry for challenging what clearly shouldn't be challenged.
guess I got a little swelled headed there thinking everything was easy

dang, if Wine allowed drive management, this would certainly be a non-issue >.<

also, thank you for making me see the bigger picture :)