pbatard/rufus #1462: Program detects custom WinPE as Windows installation

gtumanyan, 2020-02-23:
Here is the [link](https://s150vla.storage.yandex.net/rdisk/7bf87b989543ac9d644cca20bf6c12d2703c173fe7b82b2acb07c1004b3b4b20/5e528f73/3_kMTKZ-LikS4O7xBuuHCsvXXdvTTsWbth8QOqc9FcSWf3uwGlBtJoQNT59LuGfTqEzutqnb-wUiXDRINFAi8A==?uid=0&filename=Win10PE_x86x64%28EFI%29_mini%282019%29.iso&disposition=attachment&hash=NDIeFqxe7R3lxmtSo/ih1dMlShrfyPulGvTijqqni1s%3D%3A/Win10PE_x86x64%28EFI%29_mini%282019%29.iso&limit=0&content_type=application%2Fx-iso9660-image&owner_uid=92672393&fsize=375472128&hid=e496b5135c5ab33fa2d382040ad5ed6a&media_type=diskimage&tknv=v2&rtoken=K3gXruxKwP9Z&force_default=no&ycrid=na-0be98a5a6bc476cef88fe3d2f056682f-downloader17e&ts=59f3f455c82c0&s=3a7f887abae13ed59ecee6d3ce64e5b3ddce496d3977bf360440efc4dfb9e847&pb=U2FsdGVkX1_nXzDqP4KVftPzwnq8k4kQoBjDZ4CZVx8clVPLWIckhY0vp-WpdWE-BNpYmH8RfaAkTFkssbc2a0Dsj8TQo1LgSHmSLHgYpCw) of that PE

pbatard, 2020-02-23:
You ignored the checklist and did not provide a log. Also, you will understand that I'm not planning to add support for all the __custom__ ISOs that people can produce. I just don't have time for that. Unless you see an issue with an __official__ Microsoft ISO, I'm not going to look into it.

&rarr; Closed.

closed 2020-02-23 by pbatard

gtumanyan, 2020-02-23:
[image]
Rufus x86 v3.9.1622 (Beta) (Portable)
Windows version: Windows 10 64-bit (Build 17763.292)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0419 (ru-RU)
Will use default UI locale 0x0419
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'ru-RU'
Found USB 2.0 device 'USB DISK 30X USB Device' (13FE:1D00)
Using autorun.inf label for drive D: 'WIN10PEX32X64MINI'
1 device found
Disk type: Removable, Disk size: 4.1Гб, Sector size: 512 bytes
Cylinders: 494, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0076FCE0
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 3.8 GB (4063232000 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'WIN10PEX32X64MINI'
  Size: 262.0 Мб (Projected)
  Uses: EFI
  Uses: Bootmgr (BIOS only)
Using image: Win10PE_x64_EFI_(2020).iso (264.9 Мб)

Format operation started
Requesting disk access...
Will use 'D:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive1':
● Partition 1 (offset: 1048576, size: 3.8 Гб)
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Rufus Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 3.8 GB)
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Set bootable USB partition as 0x80
Using Windows 7 MBR
Found volume \\?\Volume{68ec3dea-53ae-11ea-946f-001a7dda7113}\
Disabling file indexing...
Waiting for access on \\?\Volume{68ec3dea-53ae-11ea-946f-001a7dda7113} [\Device\HarddiskVolume4]...
Opened \\?\Volume{68ec3dea-53ae-11ea-946f-001a7dda7113} for exclusive write access
Writing Partition Boot Record...
Using Standard NTFS partition boot record
Confirmed new volume has an NTFS boot sector
Successfully remounted \\?\Volume{68ec3dea-53ae-11ea-946f-001a7dda7113}\ as D:
Extracting files...
Image is an ISO9660 image
This image will be extracted using Joliet extensions (if present)
Extracting: D:\Boot\1064.wim (256 MB)
Extracting: D:\Boot\BCD (28 KB)
Extracting: D:\Boot\Fonts\boot.ttf (46.3 KB)
[... 136 lines cut ...]
No further action is required.

   3967999 KB total disk space.
    273508 KB in 24 files.
        80 KB in 21 indexes.
      9339 KB in use by the system.
      8544 KB occupied by the log file.
   3685072 KB available on disk.
      4096 bytes in each allocation unit.
    991999 total allocation units on disk.
    921268 allocation units available on disk.
NTFS Fixup completed.

Found USB 2.0 device 'USB DISK 30X USB Device' (13FE:1D00)
Using autorun.inf label for drive D: 'WIN10PEX32X64MINI'
1 device found
Disk type: Removable, Disk size: 4.1Гб, Sector size: 512 bytes
Cylinders: 494, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x001BD454
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 3.8 GB (4063232000 bytes)
  Start Sector: 2048, Boot: Yes