pbatard/rufus #1466: Rufus 3.8  Could not delete partitions: [0x80070032] The request is not supported.

JonnyGilchrist, 2020-03-01:
Attempting to create bootable UEFI USB Drive for Ubuntu 18.04 LTS
-----------------
Rufus outputs the above error when attempting to create bootable USB with Ubuntu ISO

Rufus x86 v3.8.1580
Windows version: Windows 10 64-bit (Build 18362)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 3.0 device 'SanDisk Ultra USB Device' (0781:5581)
1 device found
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 14961, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: FAT (0x00)
  Size: 114.6 GB (123060879360 bytes)
  Start Sector: 0, Boot: No
Scanning image...
ISO analysis:
  'C:\Users\Jonny\Documents\Downloads\boot\grub\efi.img' doesn't look like an ISO image
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
Using image: efi.img (2.4 MB)
Search for conflicting processes was interrupted due to timeout

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive3':
● Partition 1 (offset: 0, size: 114.6 GB)
Could not delete partitions: [0x80070032] The request is not supported.
Opened \\.\PhysicalDrive3 for exclusive write access
I/O boundary checks disabled
Writing Image...
Remounted \\?\Volume{c73f873f-4ef5-11ea-beb9-001a7dda7113}\ as E:
\\?\Volume{c73f873f-4ef5-11ea-beb9-001a7dda7113}\ is already mounted as E:
Re-mounted volume as E: after error

Found USB 3.0 device 'SanDisk Ultra USB Device' (0781:5581)
1 device found
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 14961, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: FAT (0x00)
  Size: 114.6 GB (123060879360 bytes)
  Start Sector: 0, Boot: No
Found USB 3.0 device 'SanDisk Ultra USB Device' (0781:5581)
1 device found
Disk type: Removable, Disk size: 128 GB, Sector size: 512 bytes
Cylinders: 14961, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or Unpartitioned
Partition 0:
  Type: FAT (0x00)
  Size: 114.6 GB (123060879360 bytes)
  Start Sector: 0, Boot: No

pbatard, 2020-03-01:
This is a known issue when dealing with multiple partitions, mostly because is an ass when it comes to relinquishing access.

I've tried for __YEARS__ to find a workaround for that, but I don't think I have one.

Things you should try:

* Unplug and replug the device (preferably to a different port)
* Try to use VDS mode with <kbd>Alt</kbd>-<kbd>V</kbd> (the whole VDS mode is actually me trying to find a way to get Windows' off let go of the USB when trying to delete partition)
* Try to cycle the device with <kbd>Alt</kbd>-<kbd>C</kbd>

Usually this is a transient error, so if you insist long enough, especially after plugging the drives around, Windows elevated services will stop accessing the drive.

By the way, since this tends to happen with persistence, please be very mindful that due to a major Ubuntu bug, persistence will not work unless you use Ubuntu 19.10 or later. Especially, it does not work with 18.04 LTS.

Hope that helps.

pbatard, 2020-03-18:
No update from OP &rarr; Closing.

closed 2020-03-18 by pbatard