pbatard/rufus #1469: Add Always on top

Sopor, 2020-03-06:
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.

It would have been nice if there was an option to make Rufus always on top. Rufus use a small window, so when doing other things i would like to have Rufus at the top to monitoring it. It would also make it easier to drag & drop files in to Rufus in some cases.

pbatard, 2020-03-06:
Sorry, I'm afraid I'm going to have to decline that yet again...

It just seems to be like something you would like to help your workflow, but that not that many people will have that much of an absolute need for. It's certainly not a blocking feature if you don't have it, so it's hard for me to see it as something that Rufus should provide. Please bear in mind that, in order to minimize points of failure, my default stance on adding new features is to reject the ones that are unlikely to be that necessary, even if they do qualify as _"nice to have"_. The reason behind that stance is that, no matter how small and seemingly inncouous, any new feature is a new potential point of failure, as well as something that needs to be taken care of when modifying the code so that it doesn't break from a release to another, which means that ideally, I'd need to test that "Always on top" still works with each new release, which requires A LOT more time than people realize (things like these always look like a 30 seconds tops matter... for anyone who hasn't to actually carry them out, because then it's usually a whole different story).

So next time you wish like proposing a feature, please try to consider whether is can qualify and/or something that will __drastically__ improve the majority of Rufus' users' lives.

I will also assert that there's more than enough of an indication from the task bar when it comes to progress, and would further advise that, if you want to have something on screen for monitoring, the easiest way is to buy a cheap second monitor and place whatever you want to monitor on the second screen.

&rarr; Rejected.

closed 2020-03-06 by pbatard

Sopor, 2020-03-06:
So you are saying we can buy a second monitor if we want to monitor Rufus? Sometimes i don't really understand how you are thinking ...

pbatard, 2020-03-06:
I'm not saying you are made of money. I am simply saying that, if you truly care that much about being able to monitor an application, then instead of asking somebody else to invest their time, you can try to invest your own money. In the end it's all the same thing, because, despite the application being free, my time isn't, so, to satisfy your request one way or another, someone, somewhere has to invest their __own__ money.