pbatard/rufus #1471: Add ISZ support

tormento, 2020-03-06:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I'd like to see native zipped ISO support, such as the files created from UltraISO.

Now, if I want to use Rufus, I have to unzip and feed the saved ISO.

pbatard, 2020-03-06:
I'm going to defer this until I see if there is actual demand for this. Considering that working around isz support is very easy, this will obviously never be a high priority issue.

Now, if someone wants to do the work and send me a patch, that's a different story.

pbatard, 2020-03-06:
I am also curious as to where exactly you find bootable systems that publish their images as `.isz`.

Can you point me to a __public__ example of people who publish media as `.isz`?
Because if you can't do that, I'm going to assume that `.isz` support will only apply to custom images, in which case I'm simply going to reject the request.

pbatard, 2020-03-08:
I'm going to assert that there exist no public repositories of people publishing their media as `.isz`.

As such, because this request appears to specific to helping someone's custom workflow, rather than a large number of Rufus users, I will close it.

&rarr; Rejected.

closed 2020-03-08 by pbatard

tormento, 2020-03-29:
Dear @pbatard 

sorry for the delay.

ISZ specification can be found [here](http://www.ezbsystems.com/isz/iszspec.txt).

It's enough widespread to be supported by Alcohol 120%, DAEMON Tools and WinMount (at least, perhaps other apps I don't know).

It can be uncompressed 7-zip or Winrar easily. It's main advantage, beside compression, is ISO spanning thru multiple files.

I know it is a low priority for you but loading and perhaps saving support could be of use.

Thanks.