pbatard/rufus #1472: Rufus stuck at "Formatting: 100%" when using the persistent partition

andriykrefer, 2020-03-10:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Rufus seems to be stuck at "Formatting: 100%" when using the persistent partition feature.
It works flawlessly when the persistent partition is not enabled.
I tried multiple times and waited for as much as 20 minutes.

Log
---
```
Rufus x86 v3.9.1624
Windows version: Windows 10 64-bit (Build 17134.1130)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 3.0 device 'Kingston DataTraveler 3.0 USB Device' (0951:1666)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3772, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 2
Disk ID: 0x3A9FCD50
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 2.9 GB (3111123456 bytes)
  Start Sector: 2048, Boot: Yes
Partition 2:
  Type: GNU/Linux (0x83)
  Size: 26.0 GB (27917277696 bytes)
  Start Sector: 6078461, Boot: No
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.04/20190226 (from '/isolinux/isolinux.bin')
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 19.10 amd64'
  Size: 2.3 GB (Projected)
  Uses: Syslinux/Isolinux v6.04
  Uses: EFI
  Note: This ISO uses symbolic links, which will not be replicated due to file system limitations.
  Because of this, some features from this image may not work...
Using image: ubuntu-19.10-desktop-amd64.iso (2.3 GB)
Will reuse 'ldlinux.sys' and 'ldlinux.bss' from './rufus_files/rufus_files/syslinux-6.04/20190226/' for Syslinux installation

Format operation started
Requesting disk access...
No drive letter was assigned...
Will use 'F:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive2':
● Partition 1 (offset: 1048576, size: 2.9 GB)
● Partition 2 (offset: 3112172032, size: 26.0 GB)
Opened \\.\PhysicalDrive2 for shared write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 2.9 GB)
● Creating Linux Persistence Partition (offset: 3112172032, size: 26.0 GB)
Closing existing volume...
Waiting for logical drive to reappear...
Using Ubuntu-like method to enable persistence
Formatting (ext3)...
425983 possible inodes out of 6815741 blocks (block size = 4096)
340787 blocks (5.0%) reserved for the super user
Creating 208 inode sets: [1 marker = 2.6 set(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Creating 16384 journal blocks: [1 marker = 204.8 block(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

```

pbatard, 2020-03-10:
1. Please run a bad blocks check as advised.
2. Please try with a different USB Flash Drive if you have one

This could be symptomatic of a defective USB Flash Drive, so the tests above are quite significant to isolate that possibility.

Also can you try with a smaller persistent partition (e.g. less than 4 GB)?

I'm afraid I have never experience this issue during testing and nobody else seems to be running into that problem either, so it would look like this is linked to your specific environment.

setBoolean, 2020-03-10:
> Also can you try with a smaller persistent partition (e.g. less than 4 GB)?

Ran into the same problem today (Rufus 3.9.1624), but setting a smaller persistent partition fixed the issue. It failed with 13GB and succeeded with 4GB. Didn't do any further testing though.

pbatard, 2020-03-10:
Thanks. I'll try to test with large-ish partition to see if I can replicate the issue.

andriykrefer, 2020-03-10:
> Also can you try with a smaller persistent partition (e.g. less than 4 GB)?

Succeeded with 1GB. Then, to fit my needs, I resized the partition with gparted, which also took quite some time (~20 minutes, but had a more accurate progress indicator, so I knew it wasn't stuck). Thus, it seems to be a expensive operation regardless.

A more accurate progress indicator would be nice enhancement, because I still don't know how long it would take to finish in Rufus.

Thanks for the support @pbatard

closed 2020-03-10 by andriykrefer

vsoraas, 2020-08-09:
Same issue. Rufus 3.11.1678. Brand new USB stick. Stuck on 100% formatting for 1+ hour.

```
Found USB 2.0 device 'USB DISK 3.1 USB Device' (13FE:6300)
NOTE: This device is a USB 3.0 device operating at lower speed...
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3773, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x483103AD
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 28.9 GB (31000886784 bytes)
  Start Sector: 72192, Boot: No
Downloading https://rufus.ie/files/syslinux-6.04/20191223/ldlinux.sys
Unable to access file: 404
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.sys
File length: 66.5 KB
Successfully downloaded 'ldlinux.sys'
Download signature is valid ✓
Downloading https://rufus.ie/files/syslinux-6.04/20191223/ldlinux.bss
Unable to access file: 404
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.04/ldlinux.bss
File length: 512 bytes
Successfully downloaded 'ldlinux.bss'
Download signature is valid ✓

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Deleting partitions (This may take a while)...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive4':
● Partition 1 (offset: 36962304, size: 28.9 GB)
Opened \\.\PhysicalDrive4 for shared write access
Analyzing existing boot records...
Drive has a Windows 2000/XP/2003 Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Initializing disk...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 12.9 GB)
● Creating Linux Persistence Partition (offset: 13857968128, size: 16.0 GB)
Closing existing volume...
Waiting for logical drive to reappear...
Using Ubuntu-like method to enable persistence
Formatting (ext3)...
524287 possible inodes out of 4194303 blocks (block size = 4096)
209715 blocks (5.0%) reserved for the super user
Creating 128 inode sets: [1 marker = 1.6 set(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
Creating 8192 journal blocks: [1 marker = 102.4 block(s)]
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
```

skulldealer, 2020-09-19:
Don’t even complicate it by logs n stufff

Here is a simple solution which allows u to create a persistence partition as large as u want! 1 Tb 100 Tb ....n so on
Only the limitation is ur physical disk space not a software bug

Note-By viewing the scenario it is clear that there is definitely a problem with the formatting module of rufus.

Now here is a solution

1. Set the persistence partition to 1GB
2. Within a few minutes your file will be flashed on the usb .
3. Supposedly I want to create a 100 gb partition
4. I have a 128 gb usb
5. 28 Gb a reserved for my iso boot sector, 1 gb as persistence and 99 gb unallocated memory
6. Now our goal is to merge the 99 gb to 1 gb persistence to make it 100 gb persistence which we could not have done otherwise on rufus as it would be stuck on 100% formatting

7.So either arrange a vm or casually boot another iso in an Minimum of 4gb stick. Note- don’t boot in the stick you have to operate on.

8.now install gparted (Linux partition manager for gnome btw) sometimes preinstalled just check and open it.

9.now at the top right u select your 128 gb stick.

8. Right click on the persistence 1gb sector and unmount it

9.then right click again on the same 1 gb persistence sector to resize and give it a size of 100gb by moving the start and ending black arrows

10.now click on the tick icon and apply the changes (located under the word help btw)

11.it will take some time and then show u that u have a boot partition of 28 gb and persistent partition of 100 gb exactly what u wanted !!

Took me 1 hour to add 50 gb.

Note just to confirm if everything went right, just boot up the usb after resizing and make sure that there is no persistence folder by default on the desktop
If there is a persistence folder then redo the process

In the above example u did not mess with the initial 28 go of boot file so don’t worry it is safe!

Thanks.