pbatard/rufus #1475: No % in Progress Bar

TobiasGrueninger, 2020-03-12:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
There is no % Info seen in progress bar when flashing

Log
---
Rufus x86 v3.9.1624
Windows version: Windows 10 64-bit (Build 17134.1304)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0407 (de-DE)
Will use default UI locale 0x0407
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
Found USB 3.0 device 'Generic STORAGE DEVICE USB Device' (05E3:0743)
Device eliminated because it appears to contain no media
Found USB 3.0 device 'Verbatim STORE N GO USB Device' (18A5:0247)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3775, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 4
Disk GUID: {387D34FD-A7BA-4EDE-A6CB-FF2071F07913}
Max parts: 128, Start Offset: 17408, Usable = 31054592512 bytes
Partition 1:
  Type: Linux Data Partition
  Name: 'msdos'
  ID: {F7B82D63-18F0-4632-BB31-12A9C6E3B558}
  Size: 33.6 MB (35201024 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000004
Partition 2:
  Type: Linux Data Partition
  Name: 'trainit-image'
  ID: {0A79A73B-A82A-42B0-B65E-D91235CA89CB}
  Size: 229.6 MB (240750592 bytes)
  Start Sector: 71680, Attributes: 0x0000000000000000
Partition 3:
  Type: Linux Data Partition
  Name: 'trit-payload'
  ID: {8C1636BD-ED73-445D-BE57-EA00253F5F01}
  Size: 1.5 GB (1659683840 bytes)
  Start Sector: 544768, Attributes: 0x0000000000000000
Partition 4:
  Type: Microsoft Basic Data Partition
  Name: 'trit-op'
  ID: {0DB0119C-FC6B-4AD0-8278-6F3ED08FC836}
  Size: 612 MB (641729536 bytes)
  Start Sector: 3788800, Attributes: 0x0000000000000000

Image provided: 'C:\10.Projekte\_ISR\15.TOS-Images\ISR-usb-installerMEN-dev2020-03-12_12-00#24.img'
Scanning image...
ISO analysis:
  'C:\10.Projekte\_ISR\15.TOS-Images\ISR-usb-installerMEN-dev2020-03-12_12-00#24.img' doesn't look like an ISO image
Disk image analysis:
  Image has a Zeroed Master Boot Record
  Image is a bootable disk image
Using image: ISR-usb-installerMEN-dev2020-03-12_12-00#24.img (2.4 GB)
Search for conflicting processes was interrupted due to timeout

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive2':
● Partition 1 (offset: 1048576, size: 33.6 MB)
● Partition 2 (offset: 36700160, size: 229.6 MB)
● Partition 3 (offset: 278921216, size: 1.5 GB)
● Partition 4 (offset: 1939865600, size: 612 MB)
Opened \\.\PhysicalDrive2 for exclusive write access
Writing Image...
\\?\Volume{edb33bdf-6454-11ea-be34-acfdcefe9980}\ is already mounted as G: instead of F: - Will now use this target instead...
Remounted \\?\Volume{edb33bdf-6454-11ea-be34-acfdcefe9980}\ as G:

Found USB 3.0 device 'Generic STORAGE DEVICE USB Device' (05E3:0743)
Device eliminated because it appears to contain no media
Found USB 3.0 device 'Verbatim STORE N GO USB Device' (18A5:0247)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3775, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 4
Disk GUID: {387D34FD-A7BA-4EDE-A6CB-FF2071F07913}
Max parts: 128, Start Offset: 17408, Usable = 31054592512 bytes
Partition 1:
  Type: Linux Data Partition
  Name: 'msdos'
  ID: {F7B82D63-18F0-4632-BB31-12A9C6E3B558}
  Size: 33.6 MB (35201024 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000004
Partition 2:
  Type: Linux Data Partition
  Name: 'trainit-image'
  ID: {0A79A73B-A82A-42B0-B65E-D91235CA89CB}
  Size: 229.6 MB (240750592 bytes)
  Start Sector: 71680, Attributes: 0x0000000000000000
Partition 3:
  Type: Linux Data Partition
  Name: 'trit-payload'
  ID: {DB5D52A6-F485-4800-BA34-8A5A0D2AB7E4}
  Size: 1.5 GB (1659683840 bytes)
  Start Sector: 544768, Attributes: 0x0000000000000000
Partition 4:
  Type: Microsoft Basic Data Partition
  Name: 'trit-op'
  ID: {B6E988CB-6535-45FF-89D1-75E0B480EC81}
  Size: 612 MB (641729536 bytes)
  Start Sector: 3788800, Attributes: 0x0000000000000000
[image]

TobiasGrueninger, 2020-03-12:
The % stays at 0.0% the whole time

pbatard, 2020-03-12:
This is a regression that only affects the German translation and that was introduced in e522ef6c551a8170cec8f1c90a28a1d4eaf37bc6. It has been addressed in cd4ab21e3ca7482b06254a7bc2e7fdcc9b18e9d2 and will be fixed in the next release.

closed 2020-03-12 by pbatard