pbatard/rufus #1476: Graphical Bug, Percentage not shown

TheWinHelper, 2020-03-13:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Sometimes (allmost randomly)
[image]
 when writing an DD Image File to a Stick, the current status remains at 0.0%, while the image is being written. This has happend on different Sticks, but only with images written with dd

Log
---
```
Rufus x86 v3.9.1624
Windows version: Windows 10 64-bit (Build 19041.113)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Detected installed Windows Language Pack for 0x0407 (German (Deutsch))
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'de-DE'
Found USB 2.0 device 'USB2.0 Flash Disk USB Device' (090C:1000)
Found USB 2.0 device 'Verbatim STORE N GO USB Device' (18A5:0302)
2 devices found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 487, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x2A9C5687
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 3.7 GB (4009689088 bytes)
  Start Sector: 128, Boot: Yes
Found USB 2.0 device 'USB2.0 Flash Disk USB Device' (090C:1000)
Found USB 2.0 device 'Verbatim STORE N GO USB Device' (18A5:0302)
2 devices found
Disk type: Removable, Disk size: 4GB, Sector size: 512 bytes
Cylinders: 487, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x2A9C5687
Drive has an unknown Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 3.7 GB (4009689088 bytes)
  Start Sector: 128, Boot: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Detected Grub version: 2.04~manjaro
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'MANJARO_KDEM_1902'
  Size: 2 GB (Projected)
  Uses: GRUB2
  Uses: EFI
Note: ISO mode will be disabled because this ISOHybrid is not compatible with ISO boot.
Using image: manjaro-kde-19.0.2-minimal-200311-linux54.iso (2 GB)

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive2':
● Partition 1 (offset: 65536, size: 3.7 GB)
Opened \\.\PhysicalDrive2 for exclusive write access
Writing Image...

```

JonnyTech, 2020-03-13:
>  I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.

Please follow the checklist, you would have found issue #1475

pbatard, 2020-03-13:
Indeed. Closed as duplicate.

closed 2020-03-13 by pbatard