pbatard/rufus #1477: Windows 1909 installer error (0xC0000225)

epsilon44, 2020-03-14:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Rufus creates bootable Windows 10 1909 Single Language x64 installer with no errors for a GPT+UEFI laptop. When booting installer, it shows a error screen. Missing file (\windows\system32\boot\winload.efi) with error code 0xc0000225. Laptop is on UEFI mode and secure boot off. (See first log)

Same situation when creating a MBR+(BIOS or UEFI+CSM), same error code, missing file is \windows\system32\boot\winload.exe. Same laptop switched to legacy mode and secure boot off. (See second log)

Asked on reddit, rufus developer (_Akeo_) responded. More details on link below:
https://www.reddit.com/r/windows/comments/f860zt/windows_10_installer_unable_to_boot/

Log (UEFI)
---
```
Rufus x86 v3.9.1624 (Portable)
Windows version: Windows 10 64-bit (Build 18363.720)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0C0A (es-ES)
Will use default UI locale 0x0C0A
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'es-ES'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3818, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {20C585D2-58B0-4190-BAE6-1305CB68719B}
Max parts: 128, Start Offset: 17408, Usable = 31406914048 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {07F8E563-AF26-486E-95B7-9781E1D35AB3}
  Size: 29.2 GB (31404796416 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  ID: {6253B06F-6A5D-4AA7-AD48-E95C15A7DFDB}
  Size: 512 KB (524288 bytes)
  Start Sector: 61339541, Attributes: 0x0000000000000000
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CCCOMA_X64FRE_ES-ES_DV9'
  Size: 5.1 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.0.0)
Using image: Win10_1909_Spanish_x64.iso (5.1 GB)

[... 2074 lines cut ...]

  30669823 KB de espacio total en disco.
   5337600 KB en 949 archivos.
       368 KB en 98 �ndices.
     44567 KB en uso por el sistema.
El archivo de registro ha ocupado      41936 kilobytes.
  25287288 KB disponibles en disco.
      4096 bytes en cada unidad de asignaci�n.
   7667455 unidades de asignaci�n en disco en total.
   6321822 unidades de asignaci�n disponibles en disco.
NTFS Fixup completed.

Found USB 2.0 device 'SanDisk Cruzer Blade USB Device' (0781:5567)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3818, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0001CD8C
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.2 GB (31405899776 bytes)
  Start Sector: 2048, Boot: Yes

```

pbatard, 2020-03-14:
I installed Windows 10 1909 using a media created by Rufus very recently (using the official MS ISOs), and nobody else appears to reporting this error, so the first thing I need to point out is that this is not a general issue that Rufus introduced with Windows 1909 ISOs, because, by now, there probably has been thousands and thousands of people who installed 1909 using Rufus without any issue, including people using the official Spanish image (you __are__ using the official 1909 Spanish ISO, right? If so, it might be worth checking its SHA-1 as explained [here](https://github.com/pbatard/rufus/wiki/FAQ#How_can_I_validate_that_a_Windows_ISO_is_a_genuine_retail_version)).

Therefore, there __must__ be an environmental component to your issue, even more so as the `\windows\system32\boot\winload.[xxx]` files you mention do not need to be explicitly copied by Rufus, but come from `install.wim`. You can validate this by opening the install.wim from your USB with 7-zip, then opening `sources\install.wim` and picking one of the version index (the `1`, `2`, etc you'll see in 7-zip after opening the `.wim`) and then navigating to `windows\system32\boot\`.

If the files are there, then the Windows installer should be able to access them, and if it can't, this is unlikely to be a problem caused by Rufus, because all Rufus does, __really__ is copy the whole content from the ISO onto the NTFS partition, and, apart from installing the UEFI:NTFS bootloader to a second small partiton (but this cannot have any incidence on the files being seen by Windows, because once Windows boots the installer, it can access the full content of the NTFS partition as well &mdash; Plus, this only applies to UEFI and you indicate that you see a similar issue in BIOS mode, where the second partition and extra bootloaders are not used, which pretty much confirms that this logically is exceedingly unlikely to be caused by Rufus).

Now, since you seem to have trouble with one of the early efi loaders, I know that some weird UEFI firmwares try to bypass regular UEFI entries (`\efi\boot\bootx64.efi`) when setting up boot entries for Windows, and instead try to create boot entries that point directly to the Windows bootloader (such as `EFI\Microsoft\Boot\bootmgfw.efi`). So I am wondering if maybe you have a UEFI system that bypasses the regular bootloaders and tries to look for a specific one... that may no longer exists.

Of course, the fact that you are observing the same thing in BIOS/CSM mode would tend to discredit that theory, but once again, I can only point out to the fact that myself and a large number of users have managed to perform Windows 1909 installations without any trouble, using drives generated by Rufus, and therefore that there is an environmental component to your issue that needs to be identified, because it certainly is not Rufus on its own that is the cause of your issue.

Another thing that I'm thinking could be the source of your issue is if you have some early anti-malware protection enabled in your UEFI settings, such as something that prevents boot records fro being modified (which the Windows installer might need to do), coz this could also explain why you are seeing this issue whereas everybody else seems to be fine.

I guess that, at the very least, so I can go through a installation of Windows 1909 Spanish (coz there have been instances where Microsoft screwed up the ISO for a specific language, though if they did that for Spanish, I'd have more than one report by now), I will need to know:
1. The SHA-1 of the ISO you used
2. The version you picked, if asked, during the install process.

Still, I would encourage you to look for specific environmental settings/"BIOS" options to play with as well as look for manufacturer documentation or support forum for your motherboard, as testing will most likely show that it does work for other people creating their image in the same manner you did, and therefore that there really won't be much that can be done in Rufus to hep you out.

blackcrack, 2020-03-14:
and the 3rd .. first Win10 Iso error.. of this long time .. interesting  :)

pbatard, 2020-03-20:
Just ran a test with the latest official Windows 10 1909 Spanish (SHA-1: `6f561da1cef683a93b67493a7049ed134ad44124`) and a Windows Pro installation on a UEFI platform and saw no issue:

```
Rufus x86 v3.9.1624
Windows version: Windows 10 64-bit (Build 18363.720)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found VHD device 'Microsoft Virtual Disk'
Device eliminated because it's a Windows Sandbox VHD
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
Using autorun.inf label for drive F: 'Ubuntu 18.04.3 LTS amd64'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1946, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x05112EAC
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.9 GB (16012894208 bytes)
  Start Sector: 2048, Boot: Yes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CCCOMA_X64FRE_ES-ES_DV9'
  Size: 5.1 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Win10_1909_Spanish_x64.iso (5.1 GB)

Computing checksum for 'D:\Incoming\Win10_1909_Spanish_x64.iso'...
  MD5:	 6f53c6e8841c5bb555e5a03d4b9cc181
  SHA1:	 6f561da1cef683a93b67493a7049ed134ad44124
  SHA256: 96d3754c75fccd41a79885ca6aaf4bb4acedc502425ab7d05d2be28fce7160c6

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive9':
● Partition 1 (offset: 1048576, size: 14.9 GB)
Opened \\.\PhysicalDrive9 for exclusive write access
Analyzing existing boot records...
Drive has a Syslinux Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (GPT)...
● Creating Main Data Partition (offset: 1048576, size: 14.9 GB)
● Creating UEFI:NTFS Partition (offset: 16012830208, size: 512 KB)
Writing UEFI:NTFS data...
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Using Rufus protective MBR
Found volume \\?\Volume{09481406-6ab9-11ea-9b09-f0038c9a2e6f}\
\\?\Volume{09481406-6ab9-11ea-9b09-f0038c9a2e6f}\ is already mounted as F:
[... 982 lines cut ...]
Device eliminated because it's a Windows Sandbox VHD
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1946, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {05AF58AC-D766-48C7-87E0-8A513CB500D7}
Max parts: 128, Start Offset: 17408, Usable = 16013908480 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {42D3968D-1736-4C4A-AA35-3804B5008E11}
  Size: 14.9 GB (16011781632 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  ID: {DD7795A1-07DC-4D1C-AC0C-3735FB417BAA}
  Size: 512 KB (524288 bytes)
  Start Sector: 31275059, Attributes: 0x0000000000000000
```

Considering that nobody else is reporting any similar issue, I have to conclude that this is an environmental issue with OP's target computer, and not something related to Rufus.

I will therefore close this issue.

pbatard, 2020-03-20:
The one thing I am seeing, which makes me think that OP is __NOT__ using an official image, is that they get:
```
Uses: Install.wim (version 0.0.0)
```
whereas, with the official Microsoft ISO I get:
```
Uses: Install.wim (version 0.13.1)
```
In other words, the `install.wim` version from OP's ISO is not properly detected, which is typically what I'd expect to happen with a custom or corrupt `install.wim`.

closed 2020-03-20 by pbatard