pbatard/rufus #1478: Cross compile appears to fail in modern mingw environment.

dyeske, 2020-03-14:
﻿

Checklist
---------
- [ ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ ] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ ] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ ] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>

Log
---
```
<FULL LOG>
```

pbatard, 2020-03-14:
Well, that has to be a difference that you'll need to isolate between your Linux/MinGW environment and a Windows/MinGW environment, because the latter (fully up to date as well) is what I use to produce the version of Rufus I release.

Especially your statement:

> This appears to be a difference between the native windows toolchain and mingw package.

is wrong, because I am certainly compiling with MinGW almost on a daily basis and do not need to define `PROCESSOR_ARCHITECTURE_ARM64` for builds to work.

Therefore, it's hard to give credence to the idea that this is a "modern MinGW environment" issue, because if that was the case, I would experience it too with Windows/MinGW.

My advice therefore is this: Considering that you are buidling a pure Windows application, you do have access to a Windows environment, therefore you should be able to install an up to date MinGW in its default configuration (through msys2). Once you have that, you should be able to confirm that it compiles as expected, and you may be able to isolate the options you use on Linux that may be causing you trouble (because it it does compile on Windows then it should work on Linux, and if it doesn't then this is a problem that is external to Rufus). For one thing, I'm seeing quite a few `/usr/lib/x86_64-linux-gnu` options in your compilation log that seem like they could cause trouble (MinGW on Linux should not be poking anything from the regular `linux-gnu` libraries).

Oh, and for what is worth, you will see that an up to date MinGW compiled just fine 3 days ago, as per AppVeyor (https://ci.appveyor.com/project/pbatard/rufus/build/job/jqkc35ipvdxwdi3g), without producing any error relating to the use of `PROCESSOR_ARCHITECTURE_ARM64`, therefore, unless there is a pure MinGW bug between Linux/MinGW and Windows/MinGW (which again, would not be for us to fix or work around in Rufus) `PROCESSOR_ARCHITECTURE_ARM64` is properly defined when MinGW is properly set up, therefore, if your compilation fails on Linux in a MinGW environment, then it has to be because you didn't use the right options or because you misconfigured your environment.

Now, as much as I'd like to be able to help everyone that asks for it, I have to set some limits to how far I will go and I'm afraid that, unless you can disprove the points I made above, I have to conclude that this issue is due to your environment or configuration options, and therefore not something I want to invest any more time on. Unless you can replicate the same issue on Windows/MinGW, I'm afraid this is something that you will have to sort out on your own, sorry.

&rarr; Closing.

closed 2020-03-14 by pbatard

dyeske, 2020-03-15:
It appears Ubuntu 18.04.4 has the following mingw which is too old.
ii  mingw-w64-common                     5.0.3-1

I tried again on Debian 10 which has the following.
ii mingw-w64-common                     6.0.0-3

This compile worked fine on Debian 10 wsl.

pbatard, 2020-03-15:
Happy to hear you found your issue. And yes, when you face errors like this, the first thing you should try is see if it still happens with a more recent version of the toolchain.