pbatard/rufus #1480: Would it be possible to not ask for application update policy at the start of Rufus?

BoQsc, 2020-03-16:
Could there be command line option to stop showing application update policy window at the start of Rufus?
[image]

pbatard, 2020-03-16:
Rename the application to `rufus.exe` and you won't see it (update check will be enabled by default).

closed 2020-03-16 by pbatard

JonnyTech, 2020-03-16:
@pbatard don't you mean `rufusp.exe`?

pbatard, 2020-03-16:
No. I do mean `rufus.exe`: https://github.com/pbatard/rufus/wiki/FAQ#Rufus_connects_to_the_internet_but_I_never_allowed_it_to__why

pbatard, 2020-03-16:
That's actually the reason why, on the old server, I used to have 2 symbolic links, one for the portable version (`rufusp-#.##.exe`) and one for the "no-prompt" version, which is always named `rufus.exe`, that all linked to the same binary. In other words, we're using the "check the application name" trick for more than portable vs. non-portable...

JonnyTech, 2020-03-16:
Gotcha. It's just that I do not ever recall the portable version asking the question. Thanks for the clarification.

pbatard, 2020-03-16:
The portable version does ask. Easiest way to confirm this is to launch Rufus, then press <kbd>Alt</kbd>-<kbd>R</kbd> (this deletes __all__ the registry keys created by Rufus) as well as delete any `rufus.ini` file you may have in your directory. Then you will see the default behaviour, and you can validate things like `rufus.exe` not prompting for the update check, whereas any other version will do so.