pbatard/rufus #1484: Change Window Title if writing has been started

BoQsc, 2020-03-16:
I would like to ask if you approve to append to the Window Title of Rufus some kind of "in-progress" string, so that it would be possible to see if Rufus is doing any work and should not be disturbed.

pbatard, 2020-03-16:
Sorry, no. You get the exact information you seek from the taskbar, so this is completely superfluous:

[image]

Can I ask you to please very carefully read __[this](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want)__ before you decide whether you want to open another feature request, bearing in mind that, because my development time is limited, you will never get all the bells and whistles you'd like to see in Rufus.

Please make sure to ask for something that:
1. Isn't redundant
2. Is going to help more than a handful of users
3. Is actually impairing the ability to use the application or effectively preventing users to perform some tasks (as opposed to something that's just cosmetic enhancements).

Thank you.

closed 2020-03-16 by pbatard

BoQsc, 2020-03-16:
@pbatard I completely understand. There are no ways to handle Rufus from command line.  
That means Rufus can't be used in automation. Adding progress to the Window Title would provide easy access to the progress of the Rufus.

Nobody asking these functionalies purely for entertainment and aesthetics.
I might be forced to simply search for alternative to Rufus or make a fork to resolve such simple request in relation.

BoQsc, 2020-03-16:
@pbatard  _"Rufus is NOT designed as a Sysadmin/Power User-friendly application"_

You should just write that in bold at the start of Rufus readme and stop wasting everyones time.

Thank you for your work.