pbatard/rufus #1485: Error: Access to the device is denied

kingehh, 2020-03-18:
Hello i have a problem with Rufus, every single time i try to make my USB Flashdrive bootable using rufus i get this Error: Access to the device is denied. A while ago i used Rufus without any problems(i used it on the same computer) i don't know how i can fix this error

Log

Rufus x86 v3.9.1624
Windows version: Windows 8.1 64-bit (Build 9600)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 2.0 device 'Generic USB Flash Disk USB Device' (3538:0901)
1 device found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 984, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'ESD-ISO'
  Size: 4.0 GB (Projected)
  Note: File on disk is larger than reported ISO size by 570 KB...
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.esd (version 0.14.0+)
Using image: Windows.iso (4.0 GB)

Format operation started
Requesting disk access...
Will use 'H:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive1':
● Partition 1 (offset: 0, size: 7.5 GB)
Opened \\.\PhysicalDrive1 for exclusive write access
I/O boundary checks disabled
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
write_sectors: Write error [0x00000005] Access is denied.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Retrying in 5 seconds...
write_sectors: Write error [0x00000005] Access is denied.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Retrying in 5 seconds...
write_sectors: Write error [0x00000005] Access is denied.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Retrying in 5 seconds...
write_sectors: Write error [0x00000005] Access is denied.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Could not reset partitions
Re-mounted volume as H: after error

Found USB 2.0 device 'Generic USB Flash Disk USB Device' (3538:0901)
1 device found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 984, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned
Found USB 2.0 device 'Generic USB Flash Disk USB Device' (3538:0901)
1 device found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 984, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned

Format operation started
Requesting disk access...
Will use 'H:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive1':
● Partition 1 (offset: 0, size: 7.5 GB)
Opened \\.\PhysicalDrive1 for exclusive write access
I/O boundary checks disabled
Analyzing existing boot records...
Drive has an unknown Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
write_sectors: Write error [0x00000005] Access is denied.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Retrying in 5 seconds...
write_sectors: Write error [0x00000005] Access is denied.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Retrying in 5 seconds...
write_sectors: Write error [0x00000005] Access is denied.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Retrying in 5 seconds...
write_sectors: Write error [0x00000005] Access is denied.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Could not reset partitions
Re-mounted volume as H: after error

Found USB 2.0 device 'Generic USB Flash Disk USB Device' (3538:0901)
1 device found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 984, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned
Found USB 2.0 device 'Generic USB Flash Disk USB Device' (3538:0901)
1 device found
Disk type: Removable, Disk size: 8 GB, Sector size: 512 bytes
Cylinders: 984, Tracks per cylinder: 255, Sectors per track: 63
Partition type: SFD (Super Floppy Disk) or unpartitioned

pbatard, 2020-03-18:
The check list is not optional. If you follow it, you will find similar issues to yours as well as a dedicated entry from the FAQ.

I'm afraid I don't have anything new to add to these existing entries, besides that you may also want to try <kbd>Alt</kbd>-<kbd>V</kbd> (use VDS) to see if it helps. However, this error is caused to an external application or system service preventing access to the drive, so you either need to try to find which it is, or unplug and replug your drive to force the application to relinquish access.

I will close this issue now

closed 2020-03-18 by pbatard

kingehh, 2020-03-18:
Could it be the pc's antivirus?

pbatard, 2020-03-18:
Yes, it could. The FAQ, that the check list advised you to read, has a list of software that is know to cause issues, such as some Antivirus software. That list is not exhaustive.

kingehh, 2020-03-18:
Ok i will check it thx

kingehh, 2020-03-18:
It worked finally! The problem was that i had iOBit Malware Fighter.

pbatard, 2020-03-18:
...which is explicitly listed in the FAQ. From what I can see, you should have easily found exactly the information you were after had you chosen to simply follow the check list, as requested. So can I ask you to please do that next time. The check list is not just there to annoy people!

kingehh, 2020-03-18:
Alright, Thank you,