pbatard/rufus #1487: Warning: Could not read file pointer [0x00000006] Descripteur non valide

Fl3xX, 2020-03-19:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Thank you for helping..!

Log
---
```
[rufus.log](https://github.com/pbatard/rufus/files/4353576/rufus.log)
```

pbatard, 2020-03-19:
1. You truncated the log, so I am missing CRUCIAL INFORMATION. The check list __explicitly__ tells you not to truncate the log. I cannot help you unless you provide the full log, and not just the part after `Format operation started`.

2. This looks like an error from a corrupted ISO. Are you using an official ISO? If so did you validate its checksum as asked? Where exactly did you get that ISO?

Fl3xX, 2020-03-19:
I don't use an official ISO ... Could I ask you where to get one? I would really like to go back to windows 7

pbatard, 2020-03-19:
> I don't use an official ISO

Well, I can only support official ISOs.

> Could I ask you where to get one?

https://www.microsoft.com/fr-fr/software-download/windows7

> I would really like to go back to windows 7 pro

Sorry, but I won't help you there. Windows 7 has reached end of life earlier this year so you're pretty much on your own if you want to install it as neither myself nor Microsoft will provide support for you to do so any longer.

I will close this issue.

closed 2020-03-19 by pbatard