pbatard/rufus #1491: Seems like there may exist a core bug with Rufus

TAbdiukov, 2020-03-23:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [-] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [-] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] ~~If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.~~ Yes, but with Nirsoft's utility. The hash matched the online-known values (MS unfortunately doesn't give that information publicly)

Issue description
-----------------
* ISO used: [link](https://www.microsoft.com/en-au/download/details.aspx?id=40745) (x64 one, Win 8.1 Industry Pro, which is **different** to consumer Win 8.1)
* Mode: MBR, NTFS formatted

The issue with the installation is resolved now, via using "Windows 7 USB DVD Download Tool", and it's definitely no rush. But from the course of events, it seems like there may be a core issue with Rufus

I tried burning the same image 3 times on 2 different machines (1 time on one machine; 2 on the other). Unfortunately, no logs saved since I never thought the issue is with Rufus.

The first and the second time, it would fail on "windows could not complete the installation. To install windows on this computer restart installation". The error itself is generic and appears to refer to the failure to execute "msoobe". The <kbd>Shift</kbd>-<kbd>F10</kbd> workaround posted online wouldn't work too, as the file would not be executed due to the "signature verification problem" or something, seemingly implying some internal hashing checks had been failed

The second time I burnt, I tried a different machine, and with bad-blocks checking, same deal. The third try, I used a different USB port (still connected to the motherboard), and that time, it would abort the installation on "Copying files 1-2%".

Since them I successfully managed to burn the image via "Windows 7 USB DVD Download Tool"  on the 2nd machine without issues. 

I understand my description sounds vague, and no logs provided (although the third time I checked logs, there was nothing alarming), but from my "analogue" experience, it seems like there may be some core bug to Rufus, perhaps a fresh one. Perhaps this info is to be found useful in the future :)

pbatard, 2020-03-23:
You seem to be under the impression that Rufus does something fancy, but the way we create Windows installer for BIOS mode (which I assume is what you use, because MBR and NTFS alone are not enough to differentiate between UEFI and BIOS, since UEFI can also use MBR and NTFS to boot, even in non CSM mode), really is that we copy the whole content of the ISO, __unaltered__, to an NTFS formatted drive, and then apply the __official__ boot records from Microsoft.

Heck, if you want to, you can do exactly what Rufus does by:
* Repartitioning your drive to MBR
* Create a single partition and format it to NTFS
* Extract the whole content of the ISO to the NTFS partition
* Use ms-sys (or something else) to install the standard MBR and partition boot records.

Granted, we could somehow have introduced a bug in any of the steps above, but if that was the case, that would be a very generic bug, which, considering that Rufus gets downloaded between 1 and 3 million times every month, and is used in large part to install Windows, would most certainly have been found a long time ago, especially since we haven't changed much of anything that pertains to Windows creation for quite a few releases.

All this to say that, without a lot more evidence from your side, it's quite unlikely that you have stumbled upon a major bug like the one you described, especially as hundred of thousands of people have already downloaded and used Rufus since the last release...

Which means that the explanation for your issue is likely to lie elsewhere. For instance it could have to do with the fact that, unlike what the MS tool does, Rufus does not enable dual UEFI+BIOS boot, so maybe you're not installing in the same mode as the MS tool, and your environment has an issue with that.

At any rate, I would __really__ like to see the full log of how you created your drive because as much as you think that the issue "should" be easy to replicate, you should always assume that this is not the case at all, and provide as much data as you can. Especially, you created the drive 3 times, so please do that a 4th time and provide a log. Oh and since I have reasons to believe that your USB might be defective (see below)

Furthermore, the failure on "Copying files 1-2%" is generally indicative of a flash drive issue, and as much as I'd like to take you at your word, unless I see the bad blocks check report from the log, I have doubts that you genuinely ran that test as it tends to be very time consuming. Again, there's really nothing fancy in the way Rufus extracts the files from the ISO, so, if the Windows installer complains that it can't copy data, it pretty much means that it has trouble reading the drive. I also see that you left the "I tried with a different USB" part from the check list unchecked, which adds even more fuel to the bad flash drive theory.

Finally, if you do believe that this is a fresh bug introduced in a recent version of Rufus (not sure what makes you think that, unless you were able to successfully use Rufus with the same flash drive in the past, and it's only recently that you are experiencing problems, which would add more credence to the failing flash drive theory), then you should be able to demonstrate it by reverting to an older version and testing what happens.

I know that this is not what you want to hear, and that you were hoping that 3 failed installations would seem like enough of a justification to point to a potential bug that needs urgent investigation, but when put in light of the downloads since last release and other things, I'm afraid that the burden of truth will have to be mostly on you, especially as there are 2 crucial things that you can easily do, which is provide a full log, and make sure you run a bad blocks check or try with a different flash drive.

That doesn't mean to say that I won't try to test a Windows 8.1 installation when I get a chance, but I'm afraid that your report alone is not enough at this stage to lead me to believe that there is an actual issue in the manner Rufus creates Windows 8.1 drives, so it may be a few weeks before I look into that...

TAbdiukov, 2020-03-24:
Hmm. It is all good. I'm glad it's taken reasonably seriously, despite the lack of log. Maybe it's just some really bad luck on my side. I may try to test a little more if I get a chance (as per nothing to do due to corona) and I'll let you know.

TAbdiukov, 2020-03-26:
@pbatard hey again! So far I burned 2 Windows USB's, and they appear to work well. It seems like there was a motherboard problem on my computer on my side, the USB ports rarely output incorrect voltage, perhaps it made the USB temporarily malfunction or something? Regardless, I think I need to apologise. On the bright side, now there is more information online on, "windows could not complete the installation. To install windows on this computer restart installation", which appears to be one of otherwise mysterious problems to Microsoft OSs

pbatard, 2020-03-26:
No worries, I'm glad you figured it out and I understand that it's logical to think that the application you used might have an issue if you create a bunch of drives and they all seem to fail to install.

And yeah, the Windows error messages are so generic that it's difficult to find out details about the actual cause behind a problem.

I will close the issue now.

closed 2020-03-26 by pbatard