pbatard/rufus #1493: Is Rufus to extend the given single Boot mode to both UEFI and Legacy BIOS?

pstein, 2020-03-26:
Assume I have an ISO file which was originally created for exactly one Boot mode (e.g. UEFI). The ISO is WinPE based.
If you need an example: Aomei Backupper Pro

Now I load this ISO into Rufus v3.9 and change the parameters as follows:

Partition Scheme: MBR
Target System: ALT+E   and then BIOS or UEFI
File System: FAT32

Does Rufus replace the existing UEFI-only boot mode by a dual-boot mode UEFI or BIOS?
Or does it add the missing necessary Legacy BIOS boot files?

Or will Rufus NEVER change the originally ISO embedded boot mode?
In this case Rufus should block the "BIOS or UEFI" drop down menu for selection.

Thank you
Peter

pbatard, 2020-03-26:
Provide a log with the ISO you think there is a problem with. Thank you.

pbatard, 2020-04-01:
No update &rarr; Closing.

And to answer what I think is OP's question, __NO__ Rufus cannot convert a BIOS-only or UEFI-only image to BIOS+UEFI. It also doesn't touch the original UEFI boot loader. Rufus' goal is to produce a bootable USB that is as close as possible to how the optical media would behave and, whereas we did choose to turn dual BIOS+UEFI bootable Windows media to single mode only, after seeing users struggle with Windows installations that were enacted in a mode they didn't want (but that users can easily switch back to dual BIOS+UEFI with <kbd>Alt</kbd>-<kbd>E</kbd>), we can't turn single BIOS or UEFI bootable media into dual BIOS+UEFI.

closed 2020-04-01 by pbatard