pbatard/rufus #1495: Newer version does not support bodhi-5.1.0-64 and antiX-19.1_x64-full

duytrung, 2020-03-28:
Hello,

I got rufus executable files but not able to update the syslinux update due to firewall. Could you please consider fixing this? Thank you very much

[image]

pbatard, 2020-03-28:
Rufus is designed to work online so that it can download files.

The only help I can offer if you can't go online is this:
https://github.com/pbatard/rufus/wiki/Usage-Notes#Where_to_manually_copy_the_SyslinuxGRUB_files_if_Rufus_cant_download_them

If you need more, you're on your own.

closed 2020-03-28 by pbatard

duytrung, 2020-04-01:
Thank you very much @pbatard. I think that it's better if those can be embedded into one executable file. Whenever new things released so people just download the executable and run it without having to worry about online accessibility. 

Have a great day

pbatard, 2020-04-01:
The problem is that these files can change at short notice.

If Ubuntu 20.04 decides to use an incompatible version of SysLinux or GRUB tomorrow, then I can upload the relevant files to the site tomorrow, and all users of Rufus can install Ubuntu 20.04, even if they are using an old version.

Plus, whereas other utilities seem to have no trouble asking users to download an installation package that is 10's of MB large, we try to keep Rufus' size to the minimum.

I'm afraid that it's only a minority that has a problem with online access, and if I start to design Rufus for all the annoyances various people report, I can say goodbye to sleep, or to doing anything else but modify and maintain Rufus 24/7, ever.

Sorry, but I specifically designed Rufus for online access. If you can't get online, then I will kindly ask that you use the established workaround I pointed above, especially as, if you are working offline and run into a distro that requires one of the new or updated GRUB or Syslinux files we sometimes update from one day to the next on the server, you are going to have to go online to download a new version of Rufus anyway, so, one way or another, you do need some form of online access to be able to use Rufus as expected. And again, I'm not going to make the application's size explode and annoy tons of users by making them wait longer than they should for the download to complete, just so that a handful of users don't have to go through a guide that explains step by step what they should do if Rufus can't access the online files.

duytrung, 2020-04-11:
Ah, I have read the news related to ubuntu on your issue tab. I'm not a professional developer but can understand the hardship of developers some ways. Keep release the best software. I wish you all the best

Thank you very, very much