pbatard/rufus #1500: Error: Access to the device is denied

dvs8438, 2020-04-07:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
<Please describe your issue here>
App doesnt write windows 10 image to USB device. I get error message "Error: Access to the device is denied" No other app is using the drive in that time

Log
---
```
<FULL LOG>
Rufus x86 v3.9.1624 (Portable)
Windows version: Windows 10 64-bit (Build 18363.720)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found non-USB removable device 'WDC WD40EFAX-68JH4N0' => Eliminated
Found USB 3.0 device 'JetFlash Transcend 8GB USB Device' (8564:1000)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 7.9GB, Sector size: 512 bytes
Cylinders: 960, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 0
Disk GUID: {31508265-40E8-4E3B-8816-35A29C344A43}
Max parts: 128, Start Offset: 17408, Usable = 7897840128 bytes
Scanning image...
ISO analysis:
  Image is an UDF image
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'CCCOMA_X64FRE_EN-US_DV9'
  Size: 5 GB (Projected)
  Has a >4GB file
  Uses: EFI
  Uses: Bootmgr (BIOS and UEFI)
  Uses: Install.wim (version 0.13.1)
Using image: Win10_1909_English_x64.iso (5 GB)
Search for conflicting processes was interrupted due to timeout

Format operation started
Requesting disk access...
Will use 'F:' as volume mountpoint
Deleting partitions...
No partition to delete on disk '\\?\PhysicalDrive2'
Opened \\.\PhysicalDrive2 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
write_sectors: Write error [0x00000005] Access is denied.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Retrying in 5 seconds...
Search for conflicting processes was interrupted due to timeout
write_sectors: Write error [0x00000005] Access is denied.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Retrying in 5 seconds...
Search for conflicting processes was interrupted due to timeout
write_sectors: Write error [0x00000005] Access is denied.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Retrying in 5 seconds...
Search for conflicting processes was interrupted due to timeout
write_sectors: Write error [0x00000005] Access is denied.
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Could not reset partitions
Re-mounted volume as F: after error

Found non-USB removable device 'WDC WD40EFAX-68JH4N0' => Eliminated
Found USB 3.0 device 'JetFlash Transcend 8GB USB Device' (8564:1000)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 7.9GB, Sector size: 512 bytes
Cylinders: 960, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 0
Disk GUID: {31508265-40E8-4E3B-8816-35A29C344A43}
Max parts: 128, Start Offset: 17408, Usable = 7897840128 bytes
Found non-USB removable device 'WDC WD40EFAX-68JH4N0' => Eliminated
Found USB 3.0 device 'JetFlash Transcend 8GB USB Device' (8564:1000)
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 7.9GB, Sector size: 512 bytes
Cylinders: 960, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 0
Disk GUID: {31508265-40E8-4E3B-8816-35A29C344A43}
Max parts: 128, Start Offset: 17408, Usable = 7897840128 bytes
Found non-USB removable device 'WDC WD40EFAX-68JH4N0' => Eliminated
Found USB 3.0 device 'JetFlash Transcend 8GB USB Device' (8564:1000)
Ignoring autorun.inf label for drive F: No media
1 device found
No volume information for drive 0x82
Disk type: Removable, Disk size: 7.9GB, Sector size: 512 bytes
Cylinders: 960, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 0
Disk ID: 0x00000000
read_sectors: Read error [0x00000015] The device is not ready.
  Read: 0, Expected: 512
  StartSector: 0x00000000, nSectors: 0x1, SectorSize: 0x200
Drive does not have an x86 Master Boot Record
Found non-USB removable device 'WDC WD40EFAX-68JH4N0' => Eliminated
0 devices found
Found non-USB removable device 'WDC WD40EFAX-68JH4N0' => Eliminated
0 devices found

pbatard, 2020-04-07:
Did you really [look at the FAQ](https://github.com/pbatard/rufus/wiki/FAQ#Error_0x00000005_Access_is_denied) or search for [previous similar issues](https://github.com/pbatard/rufus/issues?q=Access+to+the+device+is+denied) as you indicate you did?

closed 2020-04-07 by pbatard

dvs8438, 2020-04-07:
YES I DID !!! The error code and the nature of the error are different from the other similar issues. If you weren't a lazy ass, you would have taken a closer look at the problem and wrote a better user guide. Your product MUST do its job without any special settings from OS. And if its  don't - that's your  job to find it out why and create a better user experience. Best!

pbatard, 2020-04-07:
From your log:
> ```
> Write error [0x00000005] Access is denied.
> ```

How is that not the same error as the one I describe in the FAQ, or the one mentioned in other issues. It's not because you get subsequent errors after that that it invalidates the `[0x00000005] Access is denied` you are getting prior to it.

I believe you are disputing the idea that __other software__ can interfere with Rufus operations, and that it is just not possible for a software developer to detect every single possible software that might interfere, and add a popup that says _"Ah, I see that you are using XYZ... You're going to have to either change its settings to abc or disable it altogether if you want to use Rufus"_.

Or do you really believe that I should drop everything, and just spend my life searching for the most obscure piece of software that may interfere with Rufus, just so that __you__ don't have to? How judgemental do you have to be to delude yourself into thinking that I have all the time in the workd to do just that but not doing it because I'm lazy.

So, again, as pointed out in the FAQ, you may want to take a look at the [non-exhautive list of software that is incompatible with Rufus](https://github.com/pbatard/rufus/wiki/FAQ#Incompatible_Software).

Or you can just berate other people when they are telling you that not everything in life comes for free, especially after they already gave you something (an application) entirely for free.