pbatard/rufus #1502: Please provide a workable exe/zip build

xalaineus, 2020-04-08:
Hi,

Please provide a workable exe/zip build (also if portable too better)
after each verified commit so can test them and report then easily before
they are released.
Github with AppVeyor already provided an option for this and other software use this.
It will make life of a tester very easy so please strongly consider this.

EG of this automatic is here

URL: https://github.com/notepad-plus-plus/notepad-plus-plus/compare/v7.8.5...master

scroll to Update copyright 3de506b 

Click the green tick near the commit

details

Environment: platform=Win32, archi=x86, platform_input=Win32; Configuration: Unicode Release
Environment: platform=Win32, archi=x86, platform_input=Win32; Configuration: Unicode Debug
Environment: platform=x64, archi=amd64, platform_input=x64; Configuration: Unicode Release
Environment: platform=x64, archi=amd64, platform_input=x64; Configuration: Unicode Debug
Environment: platform=mingw-w64_810_X64, platform_input=mingw; Configuration: Unicode Release
Environment: platform=mingw-w64_810_X64, platform_input=mingw; Configuration: Unicode Debug

select any one

eg Environment: platform=Win32, archi=x86, platform_input=Win32; Configuration: Unicode Release

Jobs Console Messages Tests Artifacts  

now click artifacts tab

here the final exe is available to download and quickly test

Sorry for the bad English
Hope this will be available soon

Regards
JN
@pbatard

pbatard, 2020-04-08:
This has been requested before.

You have to realize that, unlike Notepad++, which is a usermode application, Rufus requires elevation, and because of this, for obvious security reasons, I do not want to publish non-digitally signed executables for Rufus (because I consider that ANY application that requires elevation should be signed to ensure a minimum level of trust and accountability).

The problem is that, if I were to start to publish "official" unsigned versions of Rufus executables, then I would normalize the idea that it's okay to use versions of Rufus from which the publisher cannot be authenticated (after all, if the official developer publishes them, it means they should be okay), which would open the door for some ill-intentioned people to publish their own unsigned versions of Rufus, riddled with malware, and pass them as an official build to unsuspecting users.

I am already seeing the Rufus executable being repackaged inside larger executable malware applications, with, fortunately, it easy to detect since the people doing this cannot alter the embedded executable without breaking the signature. So I __really__ don't want to ever give the idea that using non-digitally signed executable versions of Rufus is okay.

On the contrary, I very much want all Rufus users to know that, if they are trying to run a version of Rufus that is not digitally signed, or one where the signature does not say 'Akeo Consulting', they are NOT using an official release and should consider that this is potential malware.

Which brings us to AppVeyor. The only way I would ever consider publishing AppVeyor artefacts is if I could digitally sign them... but of course, to be able to do that, I would have to upload my private signing key somewhere or automate the signing process somehow, which are both VERY BAD IDEAS in terms of security.

Therefore, because I cannot publish signed artifacts and I don't want to normalize the idea that it is ever okay to use an unsigned executable that requires elevation (If you value security, this is NEVER okay!), I have only one choice, which is not publishing any artifacts.

So again, if it was a user-level application, like Notepad++, that does not require extended privileges to run, I would be happy to publish artifacts. But because it requires elevation, I can't see myself ever doing that, for security reasons.

&rar; Rejected.

closed 2020-04-08 by pbatard

xalaineus, 2020-04-08:
> 
> 
> This has been requested before.
> 
> You have to realize that, unlike Notepad++, which is a usermode application, Rufus requires elevation, and because of this, for obvious security reasons, I do not want to publish non-digitally signed executables for Rufus (because I consider that ANY application that requires elevation should be signed to ensure a minimum level of trust and accountability).
> 
> The problem is that, if I were to start to publish "official" unsigned versions of Rufus executables, then I would normalize the idea that it's okay to use versions of Rufus from which the publisher cannot be authenticated (after all, if the official developer publishes them, it means they should be okay), which would open the door for some ill-intentioned people to publish their own unsigned versions of Rufus, riddled with malware, and pass them as an official build to unsuspecting users.
> 
> I am already seeing the Rufus executable being repackaged inside larger executable malware applications, with, fortunately, it easy to detect since the people doing this cannot alter the embedded executable without breaking the signature. So I **really** don't want to ever give the idea that using non-digitally signed executable versions of Rufus is okay.
> 
> On the contrary, I very much want all Rufus users to know that, if they are trying to run a version of Rufus that is not digitally signed, or one where the signature does not say 'Akeo Consulting', they are NOT using an official release and should consider that this is potential malware.
> 
> Which brings us to AppVeyor. The only way I would ever consider publishing AppVeyor artefacts is if I could digitally sign them... but of course, to be able to do that, I would have to upload my private signing key somewhere or automate the signing process somehow, which are both VERY BAD IDEAS in terms of security.
> 
> Therefore, because I cannot publish signed artifacts and I don't want to normalize the idea that it is ever okay to use an unsigned executable that requires elevation (If you value security, this is NEVER okay!), I have only one choice, which is not publishing any artifacts.
> 
> So again, if it was a user-level application, like Notepad++, that does not require extended privileges to run, I would be happy to publish artifacts. But because it requires elevation, I can't see myself ever doing that, for the security reasons exposed above.
> 
> → Rejected.

Thank you for the very nicely explained reply

So is there any way for you to provide a build(to test)
DigiSigned by you after a commit or alpha build kind?

Just want to help please don't be offended
Thanks

pbatard, 2020-04-08:
I understand your request. But the only way I can see to provide test builds is if I can sign them, which means I would have to manually build and upload them every time (because the private key I use to digitally sign the executable is password protected and sits on a dedicated device that is not plugged unless I need to sign something). Doing this manually would be very time consuming, so I would have to automate the signing and upload process, which of course is doable, but then if someone were to hijack or steal my development machine, they would be able to sign whatever application they want using my signature.

So I would have to drastically reduce the level of security I apply to ensure that people can trust that, when they received a signed Rufus application, it is not malicious. That's simply not something I can see myself doing. I will not be degrading security for the sake of convenience.

I will also point out that I am always release a BETA version of Rufus at least one week before a release, so, if you are interested in testing, I would advise you to focus on checking the BETA versions. And also you have to understand that from time to time, I am going to break things with some commits, and, even if there are breakages that I might miss, I'd rather not have 10 people immediately send an e-mail to let me know that something that I am already in the process of fixing needs to be fixed.

I hope that makes sense. I wish I could produce signed releases for every commit, but it's just simply not possible without me having to invest a lot of time doing so, or degrading security.

pbatard, 2020-04-08:
If you do want your own builds however every time there is a new commit however, you should be able to achieve just that by:
- Having a clone of the Rufus repository with your own AppVeyor script
- Using the GitHub APIs along with an automated process, to check if a new commit has occurred on my repository and then update and launch a build on your repo.

So, if you really want to, you can have automated builds for each Rufus commit. But I would strongly advise you to keep those for your own usage and not publish them openly, because unsigned elevated applications should simply not be normalized as something that's ever okay to do, and I certainly do not want to have any part in a process that would be normalizing that idea (even if you were to ask users to do something like validate that the SHA-256 of the published executable matches the one you would generate as part of the AppVeyor process). It's just too risky as a behaviour.

xalaineus, 2020-04-08:
> 
> 
> I understand your request. But the only way I can see to provide test builds is if I can sign them, which means I would have to manually build and upload them every time (because the private key I use to digitally sign the executable is password protected and sits on a dedicated device that is not plugged unless I need to sign something). Doing this manually would be very time consuming, so I would have to automate the signing and upload process, which of course is doable (by no longer using a dedicated hardware device to store signing keys, which of course is possible), but then if someone were to hijack or steal my development machine, they would be able to sign whatever application they want using my signature.
> 
> So I would have to drastically reduce the level of security I apply to ensure that people can trust that, when they received a signed Rufus application, it is not malicious. That's simply not something I can see myself doing. I will not be degrading security for the sake of convenience.
> 
> I will also point out that I always release a BETA version of Rufus at least one week before a release, so, if you are interested in testing, I would advise you to focus on checking the BETA versions. And also you have to understand that from time to time, I am going to break things with some commits, and, even if there are breakages that I might miss, I'd rather not have 10 people immediately send an e-mail to let me know that something that I am already in the process of fixing needs to be fixed.
> 
> I hope that makes sense. I wish I could produce signed releases for every commit, but it's just simply not possible without me having to invest a lot of time doing so, or degrading security.

Thank you again for the beautiful reply way

> 
> 
> If you do want your own builds every time there is a new commit however, you should be able to achieve just that by:
> 
>     * Having a clone of the Rufus repository with your own AppVeyor script
> 
>     * Using the GitHub APIs along with an automated process, to check if a new commit has occurred on my repository and then update and launch a build on your repo.
> 
> 
> So, if you really want to, you can have automated builds for each Rufus commit. But I would strongly advise you to keep those for your own usage and not publish them openly, because unsigned elevated applications should simply not be normalized as something that's ever okay to use, and I certainly do not want to have any part in a process that would be normalizing that idea (even if you were to ask users to do something like validate that the SHA-256 of the published executable matches the one you would generate as part of the AppVeyor process). It's just too risky as a behaviour.

will try this