pbatard/rufus #1503: feature request - eject after writing

gitcnd, 2020-04-09:
fussing with ejecting USB drives is annoying - it would be great if this tool had a button to do the next thing that everyone needs to do after using this tool...

pbatard, 2020-04-09:
This has been requested before. There's a button to do that on the Windows toolbar, so I'm afraid there is no real need of duplicating this feature. I'd rather concentrate on features that can't be easily accessed elsewhere (it's literally already on the screen) and that will benefit a lot more users, as per: https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want

Also, Rufus makes sure that all accesses to the drive have been synced before indicating that it's done so, really, you could just yank the drive without going through safe eject if you want.

&rarr; Rejected.

closed 2020-04-09 by pbatard

gitcnd, 2020-04-09:
CURA has this feature - and it's the single best thing they added.
Maybe you don't use windows?  That wretched tiny icon is hard to find, takes 5 to 10 seconds to pop up the dialog, makes you think about which of the many things to eject, and then always fails and make you do all that time-waste again if something goes wrong...

If I make the fix and do a pull request, would you include it then?

pbatard, 2020-04-09:
> If I make the fix and do a pull request, would you include it then?

I don't think so, because for one thing not everybody will want to eject the drive right after creating it, and even using a setting like always/never eject the drive doesn't seem like that great an idea to me. Which means, the only satisfying way to provide this feature is to popup a dialog the the first time a drive has been completed during a session asking if the user if they also want to eject the drive (preferably with a checkbox they can check to indicate if they always want to eject it for this session), and now you're in the process of adding something that will potentially be seen as an annoyance by many people who don't care about the feature, as well as require 35+ translations for the new dialog.

Plus, as I said, Rufus does ensure all activity on the drive is completed before reporting that it's done, so you don't really need to go through a safe eject. This means that, if you are annoyed by having to go through a "safe" eject, I don't see why you don't just yank the drive and be done with it, apart from a belief that, somehow, not doing so may result in corruption in the data written by Rufus, which wouldn't be the case.

gitcnd, 2020-04-10:
Nah - it's a single button "Eject SD" that's on the existing dialog - no popups etc.

You're putting more effort into fighting against this, than it would have taken to accept something useful by now, and proposing increasingly unlikely/inventive/bogus reasons for why not... what's the REAL reason you're against this?  Like - *Honestly.* ?

pbatard, 2020-04-10:
1. You are making the assumption that because __you__ want this, everybody else will want this. I don't see it that way. Especially I don't want to force a new specific behaviour that, as I pointed out, many people might not want at all!
2. You are making a big deal of __duplicating__ a feature that's readily available and easy to access. 
3. You don't seem to have properly read [this](https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want), or, if you did, you assume that this is bullshit, because _"some other software did it!"_ and that adding __and maintaining and supporting__ features is a lot less work than it really is (which I have to conceade is an easy assumption to make as long as you aren't the person having to do the support & maintenance).

Considering that I can spare time by NOT having to duplicate something that Windows already provides and that I don't see as a necessity, or even a desirable feature, it should be quite obvious what my choice is.

Maybe the people from other projects have time to waste maintaining & supporting duplicated features from Windows. I don't.

Does that answer your question?