pbatard/rufus #1506: Error: Invalid device label

adima73, 2020-04-10:
﻿When creating a boot disk WITHOUT LABEL, for example "Freedos", Rufus sets the volume label, for example "New 7.8GB volume label (set by the size of the disk)" after process which it gives an "Error: Invalid device label" and you have to change it manually. If I remember correctly, on FAT and FAT32 the volume label cannot contain the following characters (this restriction does not apply to NTFS):
*? / \ | . ,; : + = [] <> "

pbatard, 2020-04-10:
We do sanitize labels (see [here](https://github.com/pbatard/rufus/blob/1e6e38b18075adc8527b08e7ab8a6c99946cf415/src/format.c#L244-L312)), so your assumption that we simply forgot to filter unallowed characters is wrong.

Especially I am not seeing any issue with a FreeDOS drive labelled `New 7.8GB volume label (set by the size of the disk)`, be it when creating the drive in Rufus or when booting FreeDOS.

Please provide a log, because I have no idea where you are getting your error from.

You are __not__ above following the checklist, even if you (erroneously) think you know what the problem is. At the very list, please indicate where exactly you are seeing that `Error: Invalid device label` being generated, and what the precise label you tried to use was, as I'm afraid your report is currently way to vague and impossible to replicate, for me to be able to do anything about it (which is also the reason why I didn't bother to reply to your earlier e-mail btw).

adima73, 2020-04-10:
I tried formatting with a volume label of 7.8 in Windows when I entered 7.8 , I received the warning "Invalid input: volume names FAT and FAT32 the volume label cannot contain the following characters: *? / \ | . ,; : + = [] <> "

The reason is found: In the Russian version of the interface, rufus tries to write the label 7.8 on that - it is indicated in the "Volume label" field and we get an error, but in the English version of the interface the program displays the label in the "Volume label" 7.8 field and writes label 7_8 and the error is not generated.

pbatard, 2020-04-10:
I still don't see it even when switching to Russian

Please provide your full log (you can access it with <kbd>Ctrl</kbd>-<kbd>L</kbd> or by clicking the last small icon left of 'START'), and tell me exactly what you enter in the 'Volume label' field.

adima73, 2020-04-10:
Rufus x86 v3.9.1624 (Portable)
Windows version: Windows 10 64-bit (Build 18363.752)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0419 (ru-RU)
Will use default UI locale 0x0419
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'ru-RU'
Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
1 device found
Disk type: Removable, Disk size: 7.8Гб, Sector size: 512 bytes
Cylinders: 945, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0152E022
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.2 GB (7779385344 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
1 device found
Disk type: Removable, Disk size: 7.8Гб, Sector size: 512 bytes
Cylinders: 945, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0152E022
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.2 GB (7779385344 bytes)
  Start Sector: 2048, Boot: Yes
Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
1 device found
Disk type: Removable, Disk size: 7.8Гб, Sector size: 512 bytes
Cylinders: 945, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0152E022
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.2 GB (7779385344 bytes)
  Start Sector: 2048, Boot: Yes
1 device found
Disk type: Removable, Disk size: 7.8Гб, Sector size: 512 bytes
Cylinders: 945, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x0152E022
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.2 GB (7779385344 bytes)
  Start Sector: 2048, Boot: Yes

Format operation started
Requesting disk access...
Will use 'E:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive1':
● Partition 1 (offset: 1048576, size: 7.2 Гб)
Opened \\.\PhysicalDrive1 for exclusive write access
Analyzing existing boot records...
Drive has a Zeroed Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 7.2 GB)
Closing existing volume...
Waiting for logical drive to reappear...
FAT label is mostly underscores. Using '7.8Гб' label instead.
Formatting (FAT32)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Bad label
[0x00000032] Такой запрос не поддерживается. - Retrying...
[0x00000032] Такой запрос не поддерживается. - Retrying...
[0x00000032] Такой запрос не поддерживается. - Retrying...
[0x00000032] Такой запрос не поддерживается. - Retrying...
Format error: The volume label is invalid.
Re-mounted volume as E: after error

Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 7.8Гб, Sector size: 512 bytes
Cylinders: 945, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x01798236
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.2 GB (7779385344 bytes)
  Start Sector: 2048, Boot: Yes
Closed Windows format prompt
Found USB 2.0 device 'Generic Flash Disk USB Device' (058F:6387)
1 device found
No volume information for drive 0x81
Disk type: Removable, Disk size: 7.8Гб, Sector size: 512 bytes
Cylinders: 945, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x01798236
Drive has a Zeroed Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 7.2 GB (7779385344 bytes)
  Start Sector: 2048, Boot: Yes

pbatard, 2020-04-10:
Thanks. Now I'm starting to see a bit clearer at last. I wish you had provided your log from the start, because the one important line is this:
```
FAT label is mostly underscores. Using '7.8Гб' label instead.
```
I need to look into it some more, but at least now I have a starting point to try to understand the issue.

pbatard, 2020-04-10:
This will be fixed in the next release.

commit 3c75ca92 (2020-04-10): [core] fix potential invalid label for non western locales

* If a converted label contains mostly underscore, the proposed
  label is used for FAT32 instead. However this label still has
  the KB/MB/GB symbols localized so it may be invalid.
* Ensure that we use a non-localized version of the size when
  using such a label.
* Closes #1506.
* Also fix a VS2019 static analysis warning in net.c.
closed 2020-04-10 by pbatard