pbatard/rufus #1507: Protective MBR error message overwrites primary GPT header

sveith, 2020-04-13:
Checklist
=========
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
=================

I have been trying to boot a system with an ASUS P8Z77-V PRO motherboard into the Ubuntu 18.04 and Ubuntu 20.04 (beta) installers, using the trusty Rufus utility to create a USB drive with the GPT partitioning scheme.

When selecting the recommended ISO image mode, I have found that the target system would not boot successfully from the USB drive, instead giving the following error messages and dropping me to a GRUB prompt:

```plain
error: unknown filesystem.
error: no such device: /.disk/info.
```

The same installer ISOs do work on that target system, however, if I instead select the DD image mode in Rufus.

I decided to further debug the failures in the ISO image mode, and I believe that I have found a bug in Rufus that is exposed by *another* bug in the UEFI firmware of my motherboard to trigger the boot failure.

Hypothesis
==========

When initializing a drive in the GPT format, Windows places the protective MBR in LBA sector 0, and the primary partition table in LBA sector 1, with a backup partition table in the last LBA sector of the drive.

Rufus, when writing an image in ISO image mode, will overwrite the primary GPT header in sector 1 in the `WriteSBR()` function with a `BT_NON_BOOTABLE`-type Secondary Boot Record, containing the "legacy boot of UEFI-only media" message, thus rendering the GPT header unusable.

When booting the resulting drive, most UEFI firmware can detect this situation and will silently restore the primary partition table from the backup at the end of the drive, then continue booting. Thus, in most cases, the GPT medium created by Rufus will still work.

Reproduction steps
==================

The following steps require Powershell, [QEMU for Windows](https://qemu.weilnetz.de/w64/), as well as the Hyper-V platform services and its Powershell modules to be installed. (The actual hypervisor is not required.) They were written for Windows 10 1903 and require administrator privileges.

1. Create a VHD in GPT format
   ```powershell
   $vhd = New-VHD -Path test.vhd -SizeBytes 4GB
   $disk = $vhd | Mount-VHD -PassThru | Get-Disk
   $disk | Initialize-Disk -PartitionStyle GPT
   $vhd | Dismount-VHD
   ```
2. Dump MBR (sector 0) and SBR (sector 1) of the initially partitioned drive
   ```powershell
   & 'C:\Program Files\qemu\qemu-img.exe' dd if=test.vhd of=header-before.bin count=2
   Format-Hex -Path header-before.bin
   ```
3. Observe that there is a valid EFI partition table with the magic `EFI PART` signature at offset 0x200 (sector 1)
   ```plain
   [...]
   00000200   45 46 49 20 50 41 52 54 00 00 01 00 5C 00 00 00  EFI PART....\...
   00000210   C6 D8 56 81 00 00 00 00 01 00 00 00 00 00 00 00  ÆØV.............
   00000220   FF FF 7F 00 00 00 00 00 22 00 00 00 00 00 00 00  ........".......
   00000230   DE FF 7F 00 00 00 00 00 FE 0E 59 A4 BB 83 FA 4D  Þ.......þ.Y¤».úM
   00000240   AA AE AC D5 65 74 DC A6 02 00 00 00 00 00 00 00  ª®¬ÕetÜ¦........
   00000250   80 00 00 00 80 00 00 00 0A 2A AB DF 00 00 00 00  .........*«ß....
   [...]
   ```
4. Mount the VHD
   ```powershell
[... 907 lines cut ...]
Extracting: E:\ubuntu (0 bytes)
  Ignoring Rock Ridge symbolic link to '.'
Created: E:\syslinux.cfg → /isolinux/isolinux.cfg
Finalizing, please wait...
Created: E:autorun.inf
Created: E:autorun.ico

Found USB 2.0 device 'Lexar WorkflowSR2 USB Device' (05DC:BA06)
NOTE: This device is a USB 3.0 device operating at lower speed...
Device eliminated because it appears to contain no media
Found VHD device 'Microsoft virtueller Datenträger'
Using autorun.inf label for drive E: 'Ubuntu 18.04.4 LTS amd64'
1 device found
Disk type: FIXED, Disk size: 4.3GB, Sector size: 512 bytes
Cylinders: 522, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 1
Disk GUID: {15D7275D-CFF8-4E1C-A90B-A56BA0AF08A2}
Max parts: 128, Start Offset: 17408, Usable = 4294932992 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {0FB68C07-D177-4BC3-ACC6-944E4F628FE2}
  Size: 4.0 GB (4293901824 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
```

pbatard, 2020-04-13:
Thanks for the extensive report.

You are right. I forgot that the primary GPT started at LBA 1 and the new protective message added in 3.9 gets written there.

Should be simple enough to move that message to LBA 34, i.e. after the primary GPT, so that there is no conflict. I will fix that for 3.10.

pbatard, 2020-04-13:
This will be fixed in 3.10, for which I am a planning to publish a BETA later this week, since I have to release a new version of Rufus for the Ubuntu 20.04 release anyway.

Whereas the problem is definitely something we introduced and that needs to be fixed in Rufus, it's quite interesting to see that there are many systems and software that don't seem to be able to handle fall back to the secondary GPT.

The thing is, GPT provides 2 partition table: One at the very beginning of the disk (which is the one we overwrote) and a secondary one at the very end (which should still be usable), and I would expect systems and software that are UEFI compliant to detect when a primary GPT is corrupt and switch to the secondary GPT then. So, if anything, I guess it's a good indication of which systems have proper GPT fallback, and which don't...

closed 2020-04-13 by pbatard

sveith, 2020-04-13:
Thank you for the prompt fix. I can confirm that as of commit 2dd538b7cbbe82039e3257f81d5a633e25ae1da2, both Ubuntu 18.04 and Ubuntu 20.04 beta USB media generated by Rufus boot successfully on my target system.

commit 2dd538b7 (2020-04-13): [core] fix MBR protective message overwriting the primary GPT

* We distractedly chose to populate the message from our protective MBR
  for GPT/UEFI-only boot media into the 4KB that directly followed the
  MBR, which of course is space that is being used by the primary GPT.
* This resulted on systems having to fall back to using the secondary
  GPT, which not all appear to be designed to do.
* Alter the code to ensure the protective message is written at LBA 34,
  after the primary GPT.
* Closes #1507