pbatard/rufus #1514: Wrong USB HDD drive shown in UI

kaelef, 2020-04-17:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

[image]
Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Rufus 3.9 reporting wrong USB HDD size.  3.8TB HDD attached via USB shows as 1.8TB in UI.  Windows Disk Management and Rufus log indicate correct size is detected.

Log
---
```
Rufus x86 v3.9.1624
Windows version: Windows 10 64-bit (Build 18363.778)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'HGST HDN 724040ALE640 USB Device' (05E3:0718)
No logical drive found (unpartitioned?)
Device eliminated because it was detected as a Hard Drive (score 15 > 0)
If this device is not a Hard Drive, please e-mail the author of this application
NOTE: You can enable the listing of Hard Drives under 'advanced drive properties'
0 devices found
Found USB 2.0 device 'HGST HDN 724040ALE640 USB Device' (05E3:0718)
No logical drive found (unpartitioned?)
1 device found
No volume information for drive 0x83
Disk type: FIXED, Disk size: 1.8TB, Sector size: 512 bytes
Cylinders: 219051, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {E44A70A0-F9EE-3446-9D74-F6B33DAD3EB8}
Max parts: 128, Start Offset: 17408, Usable = 4000786995712 bytes
Partition 1:
  Type: Apple ZFS Partition
  Name: 'zfs'
  ID: {3321373C-D623-B748-AC13-D91EB80FE158}
  Size: 3.6 TB (4000776716288 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2:
  Type: Solaris Reserved Partition
  Name: ''
  ID: {B85EA9FA-3EA7-054D-996C-AB7A455C4260}
  Size: 8 MB (8388608 bytes)
  Start Sector: 7814019072, Attributes: 0x0000000000000000

```

pbatard, 2020-04-17:
Thanks for the report. Let me look into that.

kaelef, 2020-04-17:
Now that I look at this more closely, there's a good chance this is an issue with the USB-SATA adapter, as Windows Disk Management can't delete the partitions it finds, returning a "Parameter Incorrect" error when I try.  I'll try a different USB-SATA adapter when I get a chance.  It's interesting the Rufus still reports the wrong size, but I think it's probably not its fault...

pbatard, 2020-04-17:
You may be right because, at first glance, there seems to be a discrepancy between the disk size Windows reports for `IOCTL_DISK_GET_DRIVE_GEOMETRY_EX` and `IOCTL_DISK_GET_DRIVE_LAYOUT_EX`.

The `1.8 TB` we report in the UI as well as in the log with `Disk type: FIXED, Disk size: 1.8TB, Sector size: 512 bytes` comes from `IOCTL_DISK_GET_DRIVE_GEOMETRY_EX`. On the other hand the `4 TB` we report in the log with `Max parts: 128, Start Offset: 17408, Usable = 4000786995712 bytes` comes from `IOCTL_DISK_GET_DRIVE_LAYOUT_EX`.

In other words, if for any reason `IOCTL_DISK_GET_DRIVE_GEOMETRY_EX` and `IOCTL_DISK_GET_DRIVE_LAYOUT_EX` report different things for the disk size (this is something that's internal to Windows and that we have no control on), then that could explain your issue.

kaelef, 2020-04-17:
Yes, this adapter is working fine with many other drives I'm testing, but none of them are bigger than 2TB.  I'm guessing it's a firmware issue on the adapter.  I think it's fair to close this issue!  Thanks for looking into it.

pbatard, 2020-04-17:
Okay, I'll close it then.

closed 2020-04-17 by pbatard