pbatard/rufus #1516: Enable sparse_super EXT3 feature

marcosfrm, 2020-04-20:
Rufus creates EXT3 filesystem without the `sparse_super` feature enabled. EXT3 this way has superblock and group descriptors backups (2 blocks) on **every** block group.

With `sparse_super`, backups are created only if block group number is a power of 3, 5 or 7, which results in higher number of free blocks. Maybe it also speeds up filesystem creation.

No changes otherwise.

Thanks for the fantastic tool.

pbatard, 2020-04-20:
I'm going to have to test this again.

The thing is, I had `sparse_super` enabled for a while, but I [commented it out](https://github.com/pbatard/rufus/blob/0e3658df1a2f71b80042823bd06a708ce66298e5/src/format_ext.c#L330) because it seemed to produce errors with `e2fsck`.

As long as `e2fsck` doesn't complain, I'm fine with this request. Since other aspects of the `ext#` creation that were the source of errors have been fixed in the meantime, I'll run a few tests and see if I'm still seeing issues. If not, then I'll try to enable `sparse_super` for the 3.10 release.

marcosfrm, 2020-04-20:
Leave `resize_inode` disabled, otherwise it will waste a bunch of blocks (many more than non-`sparse_super` case) with the reserved GDT blocks. The ability to online resize filesystem Rufus has created is not important at all.

pbatard, 2020-04-20:
Yeah, I'm not planning to touch this one, as `e2fsck` definitely complains if I enable it.

marcosfrm, 2020-04-20:
I could not find any call to `ext2fs_create_resize_inode()`, perhaps that was it. Anyway, it does not matter. :-)

https://github.com/pbatard/rufus/blob/0e3658df1a2f71b80042823bd06a708ce66298e5/src/format_ext.c#L334

BTW, you probably do not need to set this, because it is only used when EXT4-specific `sparse_super2` feature is set.

pbatard, 2020-04-20:
> I could not find any call to `ext2fs_create_resize_inode()`, perhaps that was it.

Most likely. You have to realize that one of the issues I've had in adding the `ext#` formatting code in Rufus is that whereas the `lib/ext2fs` sources are available under LGPLv2 can can therefore be used into a GPLv3 project, the `mke2fs.c` source is GPLv2 __only__, which means I couldn't use it and had to guess a lot of the formatting sequence. So I probably missed a few things.

> BTW, you probably do not need to set this

Okay. I'll remove this then. Thanks for the pointers!

pbatard, 2020-04-21:
Testing does not show adverse effects from re-enabling `sparse_super` and the other proposed change, so I have applied those. Thanks again.

closed 2020-04-21 by pbatard

marcosfrm, 2020-04-23:
https://github.com/pbatard/rufus/blob/0085c4a464a4b0128f7e541a352973ba5448e283/src/ext2fs/nt_io.c#L749

`unix_io.c` does not try to change partition type in `io_channel_flush()`. At least for Rufus usage, this `NtDeviceIoControlFile(... IOCTL_DISK_SET_PARTITION_INFO ...)` call is unnecessary, no? (maybe broken too, as it appears to not cover GPT) Even for upstream e2fsprogs it simply does not look right -- format tool should not mess with partition type.

pbatard, 2020-04-23:
If it's anything like `IOCTL_DISK_GET_PARTITION_INFO` this IOCTL is ignored for GPT. And type `0x83` is what we are using as well. So I'm not seeing this call is hurting us any and, specifically, I have not seen it causing issues when testing.

Note that this is code that we got from upstream e2fsprogs, so if there is an issue, it should be addressed there (but then again, it looks like `nt_io.c` has pretty much been left in an abandoned state for some time from what I can see).

marcosfrm, 2020-04-23:
Yes, `nt_io.c` was in bad shape. Until your great cleanups took place! Do you have plans to upstream them?

pbatard, 2020-04-23:
I do... eventually. But I'm not entirely sure how tolerant project that's mostly aimed at POSIX (hence cygwin support and no MinGW) is going to be with regards to obvious MSVC'isms.

I actually have a bunch of 4 commits that could be used for upstreaming (after rebasing to latest) in https://github.com/pbatard/e2fsprogs/commits/msvc but I have no clue when I'll get a chance to get sorted on those.

marcosfrm, 2020-12-28:
https://www.spinics.net/lists/linux-ext4/msg75966.html

Maybe useful.

pbatard, 2020-12-28:
I subscribe to the linux-ext4@vger.kernel.org mailing list, so I did see that. But thanks for pointing it out.

At any rate, I guess this kills the need to upstream my `nt_io.c` changes, which I'm not unhappy about (less work for me!), though I expect code that is aimed at MinGW to require a few changes to be compiled for MSVC.

At this stage, I am waiting for this to be formally integrated to look into using it in Rufus.

commit cf52f61b (2020-04-21): [ext2fs] re-enable the sparse_super feature

* Closes #1516
* Also prevent protective message from being written in MBR mode