pbatard/rufus #1517: Strange device behavior after bootable creation

MrAdam1983, 2020-04-20:
﻿

Checklist
---------
- [ X] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ X] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ NA] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ NA] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ X] Using a different USB drive.
  - [ X] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
My issue is not with creation or with the bootable aspect of the process, my bootable works just fine on every other computer except the one I created it on; I think my problem is my instance of Windows 10 but I am looking to see if anyone else has had this problem. 

Again as far as I can tell, rufus works perfectly.

I can take ANY USB drive, I mostly have 32gb sandisk but I also have some no name 8GB; again creation goes fine and the drive will boot no problem.  I can go to any other Windows instance and access the drives file structure without issue. The problem is that when I plug these drives, after bootable is created, back into the computer that created them (Which right now is only my work laptop) the OS acknowledges the drive is inserted, starts to access the file structure, fails, indicates the drive was removed even though it wasn't, and occasionally will report the device is corrupt.  I can take it to literally any other computer in my lab (Windows XP, Windows 7, or any other Windows 10) and I can access the drive just fine.  If I do a deep reformat, the drive works on my laptop again.

If I was getting this problem on other devices i would assume it was rufus, but it only happens on one computer and it happens on all of the ports on that one computer. 

I have updated drivers
I have checked windows updates
I have looked for similar problems
I have successfully made bootable drives on this same machine before. 
I have the latest version of rufus but this was happen on the previous version as well ( I don't think rufus is at fault, but Windows 10 does not like something about these bootables)

I am booting a ChrisPE LiveOS made with PEbakery, but I made several ISOs that worked before this all started. 

and again, it's only one laptop that is doing this. 

Thoughts?
Log
---
```
<FULL LOG>
```

JonnyTech, 2020-04-20:
To rule out a hardware problem, boot the computer with a live Linux disk (ie [Linux Mint](https://linuxmint.com/download.php) and check to see if it can successfully read the disk(s). If that cannot then you have a hardware issue else your install copy of Windows is bad.

pbatard, 2020-04-20:
Can you try with 3.10 BETA to see if it makes any difference? (download [here](https://rufus.ie/downloads))

And, even though you __think__ this is superfluous, I sure would like to see a complete log of how you created the drive. As the developer, I can tell you one thing: You would be surprised how many times __critical__ information that actually helped with the troubleshooting the issue was produced from eventually attaching a log, even though the person who opened the issue thought it wasn't needed...

MrAdam1983, 2020-04-21:
i am now not able to replicate the error on a new Flashdrive and the two i was having issues with are now exhibiting the same issues on other systems.  just yesterday they only had issues on my laptop, now I can't even format them on another system.  I will report back what I find but i now suspect that both of theses drives are damaged.

MrAdam1983, 2020-04-21:
Finally replicated the error.  Creation went fine, drive will boot in another computer, put it back in the laptop that made and it cannot read. 

Found USB 2.0 device 'General UDisk USB Device' (ABCD:1234)
1 device found
Disk type: Removable, Disk size: 2GB, Sector size: 512 bytes
Cylinders: 243, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x004B9BC6
Drive has a Rufus Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 1.9 GB (2001731584 bytes)
  Start Sector: 2048, Boot: Yes

Format operation started
Requesting disk access...
Will use 'D:' as volume mountpoint
Deleting partitions...
Deleting ALL partition(s) from disk '\\?\PhysicalDrive2':
● Partition 1 (offset: 1048576, size: 1.9 GB)
Opened \\.\PhysicalDrive2 for exclusive write access
Analyzing existing boot records...
Drive has a Rufus Master Boot Record
Clearing MBR/PBR/GPT structures...
Erasing 128 sectors
Deleting partitions...
Partitioning (MBR)...
● Creating Main Data Partition (offset: 1048576, size: 1.9 GB)
Closing existing volume...
Waiting for logical drive to reappear...
Formatting (NTFS)...
Using cluster size: 4096 bytes
Quick format was selected
Creating file system...
Format completed.
Writing Master Boot Record...
Set bootable USB partition as 0x80
Using Rufus MBR
Found volume \\?\Volume{3c6869ff-7830-11ea-bdf8-b88a60daf01a}\
Disabling file indexing...
Opened \\?\Volume{3c6869ff-7830-11ea-bdf8-b88a60daf01a} for exclusive write access
Writing Partition Boot Record...
Using Standard NTFS partition boot record
Confirmed new volume has an NTFS boot sector
Successfully remounted \\?\Volume{3c6869ff-7830-11ea-bdf8-b88a60daf01a}\ as D:
Extracting files...
Image is an UDF image
Extracting: D:\2020-04-21-12.37.tag (0 bytes)
Extracting: D:\Project.Settings.ini (900 bytes)
Extracting: D:\boot\bcd (16 KB)
Extracting: D:\boot\boot.sdi (3 MB)
Extracting: D:\boot\bootsect.exe (93.8 KB)
Extracting: D:\boot\en-us\bootsect.exe.mui (16 KB)
Extracting: D:\boot\etfsboot.com (4 KB)
Extracting: D:\boot\fonts\chs_boot.ttf (3.5 MB)
Extracting: D:\boot\fonts\cht_boot.ttf (3.7 MB)
Extracting: D:\boot\fonts\jpn_boot.ttf (1.9 MB)
Extracting: D:\boot\fonts\kor_boot.ttf (2.3 MB)
Extracting: D:\boot\fonts\malgun_boot.ttf (173.9 KB)
Extracting: D:\boot\fonts\malgun_console.ttf (80.6 KB)
Extracting: D:\boot\fonts\malgunn_boot.ttf (171.5 KB)
Extracting: D:\boot\fonts\meiryo_boot.ttf (142.5 KB)
Extracting: D:\boot\fonts\meiryo_console.ttf (89.6 KB)
Extracting: D:\boot\fonts\meiryon_boot.ttf (140.8 KB)
Extracting: D:\boot\fonts\msjh_boot.ttf (161.8 KB)
Extracting: D:\boot\fonts\msjh_console.ttf (199.2 KB)
Extracting: D:\boot\fonts\msjhn_boot.ttf (159.9 KB)
Extracting: D:\boot\fonts\msyh_boot.ttf (153.6 KB)
Extracting: D:\boot\fonts\msyh_console.ttf (95.4 KB)
Extracting: D:\boot\fonts\msyhn_boot.ttf (151.8 KB)
[... 83 lines cut ...]
  Type: NTFS (0x07)
  Size: 1.9 GB (2001731584 bytes)
  Start Sector: 2048, Boot: Yes
0 devices found
Found USB 2.0 device 'General UDisk USB Device' (ABCD:1234)
Device eliminated because it appears to contain no media
0 devices found
Found USB device 'General UDisk USB Device' (ABCD:1234) [ID]
Device eliminated because it appears to contain no media
0 devices found
0 devices found
Found USB 2.0 device 'General UDisk USB Device' (ABCD:1234)
Device eliminated because it appears to contain no media
0 devices found
0 devices found
Found USB 2.0 device 'General UDisk USB Device' (ABCD:1234)
Device eliminated because it appears to contain no media
0 devices found
0 devices found
Found USB 2.0 device 'General UDisk USB Device' (ABCD:1234)
Device eliminated because it appears to contain no media
0 devices found
0 devices found
0 devices found
0 devices found

MrAdam1983, 2020-04-21:
and after a reboot it works fine, can't boot to drive due to UEFI and i don't have access to the BIOS.

if you got an idea let me know, otherwise I give up.

pbatard, 2020-04-22:
Sorry, you truncated the log, so I don't know what image, what version/build of Windows you are using and other details.

One thing's for sure, that `General UDisk USB Device' (ABCD:1234)` looks fake as heck, as `ABCD` is unlikely to be a valid vendor ID, especially followed by a `1234` PID. `ABCD` is what you use when you aren't a legitimate flash drive producer.

As such, I don't think there's much to investigate and I will close this issue.

closed 2020-04-22 by pbatard