pbatard/rufus #1518: Enable /DYNAMICBASE /guard:cf for VS builds

asvc, 2020-04-22:
Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Readme states that VS 2019 can (should?) be used to build the Rufus.
Currently, binary is being shipped with ASLR and CFG disabled:
```ps
> Get-PESecurity -File .\rufus-3.10_BETA.exe

ARCH             : I386
DotNET           : False
ASLR             : False
DEP              : True
Authenticode     : True
StrongNaming     : N/A
SafeSEH          : False
ControlFlowGuard : False
HighentropyVA    : False
```
Would you consider releasing x86_64 version with the above-mentioned flags enabled?
Alternatively, if gcc/MSYS2 is your preference, adding `-Wl,dynamicbase,highentropyva` might help.

Log
---
```
Rufus x86 v3.10.1642 (Beta) (Portable)
Windows version: Windows 10 64-bit (Build 18363.778)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0809 (en-GB)
Will use default UI locale 0x0809
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found VHD device 'Microsoft Virtual Disk'
Device eliminated because it's a Windows Sandbox VHD
0 devices found
```

pbatard, 2020-04-22:
> Would you consider releasing x86_64 version with the above-mentioned flags enabled?

I guess I can enable those in the VS2019 project files, sure.

> Alternatively, if gcc/MSYS2 is your preference

Well, it's not just a matter of preference. MinGW is what we use to build the release because, as opposed to VS2019, it's not a proprietary closed source compiler and therefore it makes it easier to find out if it has unwanted behaviour.

I don't have an issue with adding `-Wl,dynamicbase,highentropyva` unless it increases the size of the executable too much, since I'm trying to keep it as small as possible.

pbatard, 2020-04-23:
Apparently the best I can do in MinGW is enable ASLR (`-Wl,--dynamicbase`) since we're compiling for 32-bit and `highentropyva` seems to be a 64-bit only option.

However, you should know that, if you install Rufus from the `.appx`, you will get an x86-64 version installed that does have all the options you seek, since it comes from VS2019 and these options are enabled by default:

```
> Get-PEsecurity -File .\x64\Release\rufus.exe

FileName         : C:\Projects\rufus\x64\Release\rufus.exe
ARCH             : AMD64
DotNET           : False
ASLR             : True
DEP              : True
Authenticode     : True
StrongNaming     : N/A
SafeSEH          : N/A
ControlFlowGuard : False
HighentropyVA    : True
```

asvc, 2020-04-23:
> if you install Rufus from the .appx, you will get an x86-64 version installed that does have all the options you seek

You are right, DEP and ASLR are indeed enabled:
[image]

Would you consider CFG as well? Maybe as alpha/beta for those feeling adventurous.

pbatard, 2020-04-23:
> Would you consider CFG as well? Maybe as alpha/beta for those feeling adventurous.

This is not a web server application processing potentially malicious input. As much as I am happy to apply mitigation, the attack vectors when Rufus are relatively small and would usually require the machine to already have been compromised in one way or another, so that one could craft a malicious `.loc`, `.ini`, sideload a DLL and so on.

Also, as opposed to other applications, Rufus is not something that's expected to be running all the time, or even frequently (as opposed to applications like Notepad++), so it's hard to take the threat of someone deliberately targeting their compromising of a computer through an attack on Rufus as a real-life scenario.

We have already hardened the one element that we can see a remote attacker using as an attack vector, which is everything that Rufus downloads from the internet (see https://github.com/pbatard/rufus/wiki/Security#Digital_Signatures and https://github.com/pbatard/rufus/wiki/Security#Mitigated_risks), which should make it pretty difficult for someone not already having compromised the local machine to compromise the app, as Rufus shouldn't let through any download that doesn't have a signature that matches our private key.

So that pretty much leaves only maliciously crafted images as an attack vector, but then again, since you are creating a bootable drive from those, there's little point of creating an ISO that triggers malicious code execution in the app since you can just replace the content from the ISO with malicious content and infect the computer that is going to boot from that image.

All this to say that I don't think going as far as add CFG is something I want to do for the time being. Unless you are a web server app running 24x7 and handling a lot of private user data, there are degrees to blindly applying every security feature under the sun just because they are there...

pbatard, 2020-04-23:
Glad you liked it. Of course I'm always open to security improvements, and your earlier suggestion to apply ASLR for MinGW is one that I should have applied some time ago, since it doesn't cost anything and is something any modern executable should have, just in case. And I am also not saying no for CFG, but right now, I'm a bit worried that it may impact performance and I'm also seeing reports of CFG causing issues when trying to debug applications, which is why I'd rather wait on that.

> Shall this issue be closed then?

I will close it with the commit where I enable ASLR for MinGW, which should be coming soon.

commit 44a8e08b (2020-04-23): [misc] enable ASLR for MinGW builds

* ASLR is enabled by default for Visual Studio builds but that isn't the case
  for MinGW builds. Fix that and also add -Wformat-security while we're at it.
* Closes #1518
* Also ensure that we'll never write protective MBR message for non-bootable
  GPT drives, even as we are not calling WriteSBR() for those anyway.
closed 2020-04-23 by pbatard