pbatard/rufus #1523: Unable to start after crash - "Unable to synchronize with GUI application."

lauyuen, 2020-04-24:
﻿

Checklist
---------
- [x ] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [ x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [ \] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [ \] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ x] I also tried one or more of the following:
  - [ x] Using a different USB drive.
  - [ x] Plugging the USB into a different port.
  - [ \] Running Rufus on a different computer.
- [ \] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Trying to start Rufus with `win+r` `rufus` or `rufus` on the command line/powershell immediate outputs `Unable to synchronize with GUI application.` and exits. 

After some digging, it looks to be a problem of hogger.exe.
https://github.com/pbatard/rufus/blob/d4a4506b16eacee7f4c0bfeb50ecddd0b2a7d5be/res/hogger/hogger.asm

The reason is that the default path of Rufus is now being taken over by what I assume is the mutex flag generated by hogger? The solution was to delete `%userprofile%/rufus.com`. 

It may be worth an add in the wiki as it's not documented elsewhere on the internet. 

Log
---
```
<FULL LOG>
```

pbatard, 2020-04-24:
Hmm, I'm not seeing any issue with <kbd>Win</kbd>-<kbd>R</kbd> or commandline/powershell, so I'm afraid this is not the easy to replicate universal failure that you assume it to be, and there must be some environmental component or something you haven't disclosed with regards to your testing.

The only small hiccup I see, which is to be expected, is that when starting from a PS prompt, you get a `rufus.com : The term 'rufus.com' is not recognized as the name of a cmdlet, function, script file, or operable program.` which is to be expected since PowerShell doesn't run COM files. But otherwise, I am not able to see the `Unable to synchronize with GUI application` you report.

Care to provide a log? It may give some info about your environment that may help understand what your issue is.

Otherwise, I'm afraid you'll have to provide some more detailed steps on how one is supposed to reproduce this on a vanilla Windows machine?

That is unless what you are saying is that Rufus crashed (Can you indicate how? I'm actually more interested in that), and therefore left the `rufus.com` hogger running behind, which of course may create some issues.

lauyuen, 2020-04-26:
You're absolutely right. It did happen after Rufus stalled. I had to exit the program by killing the process. I didn't save the logfile when I reinstalled Rufus, so I went back into Rufus a few times to try and replicate what happened without success. I also wished I went and examined the content of what was in the `Rufus.com` file. I did notice that if I were to try and start hogger.exe https://github.com/pbatard/rufus/tree/d4a4506b16eacee7f4c0bfeb50ecddd0b2a7d5be/res/hogger from the resource directory without Rufus running, I get the exact same output, `Unable to synchronize with GUI application.`. All in all, maybe it was a freak accident that Rufus stalled at exactly the right time that caused this problem, but it seems like I can't replicate it for now. Please consider this solved. Thanks again.

closed 2020-04-26 by lauyuen

pbatard, 2020-04-26:
Okay. At the very least people getting this message should be able to find this issue by just searching the error they got, and see that they need to delete `rufus.com` if it exists.