pbatard/rufus #1531: Rufus cannot download ldlinux.sys

jrubinos, 2020-05-04:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
I wanted to create a bootable drive for ubuntu 16.04. However I keep on getting download failed error.
I have also tried running rufus as administrator, I have also turned-off firewall.

I was able to download ldlinux.sys manually, however rufus cannot detect the file. 

Log
---
```
Drive has a DOS/NT/95A (F2) Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.5 GB (15581806592 bytes)
  Start Sector: 64, Boot: Yes
Downloading https://rufus.ie/files/syslinux-6.03/20151222/ldlinux.sys
Unable to send request: The attempt to connect to the server failed.
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.03/ldlinux.sys
Unable to send request: The attempt to connect to the server failed.
Could not download the file - cancelling

Found USB 2.0 device 'USB Disk 2.0 USB Device' (23A5:5786)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1894, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x06A484DB
Drive has a DOS/NT/95A (F2) Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.5 GB (15581806592 bytes)
  Start Sector: 64, Boot: Yes
Downloading https://rufus.ie/files/syslinux-6.03/20151222/ldlinux.sys
Unable to send request: The attempt to connect to the server failed.
Extended version was not found, trying main version...
Downloading https://rufus.ie/files/syslinux-6.03/ldlinux.sys
Unable to send request: The attempt to connect to the server failed.
Could not download the file - cancelling

Found USB 2.0 device 'USB Disk 2.0 USB Device' (23A5:5786)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1894, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x06A484DB
Drive has a DOS/NT/95A (F2) Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.5 GB (15581806592 bytes)
  Start Sector: 64, Boot: Yes
```

pbatard, 2020-05-04:
Rufus requires the ability to go online. If you can't download files, then the issue is on your end, because, as you should be able to check, `https://rufus.ie/files/syslinux-6.03/ldlinux.sys` can definitely be downloaded from a browser if you have network connectivity and you don't have a firewall applied that prevents access to GitHub (which is where the file resides).

So you will need to check that:
- You aren't using a Security Solution that __restricts__ the ability of applications like Rufus to download files
- You aren't behind a firewall that prevents access to https://rufus.ie/files/syslinux-6.03/ldlinux.sys
- Your network connectivity is working.

If you can't sort any of the above, then [this](https://github.com/pbatard/rufus/wiki/Usage-Notes#Where_to_manually_copy_the_SyslinuxGRUB_files_if_Rufus_cant_download_them) might interest you, but that is the only help I will provide on the subject, as it is really up to you to sort your network access issues.

closed 2020-05-04 by pbatard

jrubinos, 2020-05-04:
thanks... the last link solved the problem.