pbatard/rufus #1533: small request

jmm8344562, 2020-05-09:
I created a FOLDER in a16GB  WIN 10 USB CREATED BY RUFUS. 

The 16GB_USB properties showed 11.7GB available after  the WIN 10 installation.  

I titled the FOLDER   --
                     "11GB-MAX FOLDER FOR FILES"

IT WOULD BE A PERK TO THE USERS OF YOUR PRODUCT IF A FILE FOLDER ( TITLED IN SIMILIAR TERMS ) BE AUTO CREATED AFTER THE OTHER OUTPUT OF RUFUS IS COMPLETED.  

After installing Win 10, I like to install a set of programs such as Jarte. 7zip, IMGBURN, etc.

Perhaps a revenue stream from other websites would occur when the folder is created and populated by any user selecting programs ( and even WIN10 apps ).

this concept is free and open source from me unless other restrictions, I do not know about ,apply to this concept.    ENJOY

This is my first experience with this issue web page and I am here at the recommendation of your RUFUS website link to this issue website. 
My first recommendation is  -  include the many details shown in this text window, on the RUFUS website link that directs to here.
I can say that this list was not a courteous introduction for someone with a simple non technical problem request.
I can see and understand why you present the list of required items, but for a first time user overwhelming, to say the least.

Thank you and your organizations involved for a user friendly and useful program.  The proposed addition of an automatic folder creation service would fit into that paradigm 

James M. McDonald
jmm8344562@gmail.com

pbatard, 2020-05-09:
> IT WOULD BE A PERK TO THE USERS OF YOUR PRODUCT IF A FILE FOLDER ( TITLED IN SIMILIAR TERMS ) BE AUTO CREATED AFTER THE OTHER OUTPUT OF RUFUS IS COMPLETED.

By design, Rufus does not apply any modification to the ISO content it has copied over unless it has to.

I'm also very much afraid that your request falls into point 2. of https://github.com/pbatard/rufus/wiki/FAQ#Why_cant_you_just_add_the_feature_I_want, in that it is something that you appear to want, but that, when everything is considered, not as many people as you imagine will want, because you do appear to want this for the very custom scenario where you want to add your own apps post install to the drive that has been created by Rufus. I'm afraid that I do foresee that, if I were to add the feature you request, I would instantly get tons of complaints from sysadmin and power users on _"Why is Rufus creating a folder I never asked it to create? Please remove that!"_.

> After installing Win 10, I like to install a set of programs such as Jarte. 7zip, IMGBURN, etc.

Yes, this is something __you__ like to do. But please try to be very conscious that something that sounds very desirable or even logical to want to have to you may not be viewed the same by other people, no matter how much you think the whole world may benefit from your proposal...

> Perhaps a revenue stream from other websites would occur when the folder is created and populated by any user selecting programs ( and even WIN10 apps ).

In case this hasn't been made clear, and despite the fact that there are ads on the web page, I'm not motivated by money.

> I can say that this list was not a courteous introduction for someone with a simple non technical problem request.

This is intended that way. I get many many requests, and no matter how many people complain that I should be "more courteous" or "nicer", my only goal is to __HELP__ people in the most effective way, which means ensuring that a minimum level of triage has occurred and that relevant information is provided. You get the product entirely for free, and the process of trying to help as many people as I can means I __must__ reduce the time it costs me to process each request. And I'm afraid I get a lot fewer requests to add features, and that I very much expect people who produce such a request to read the FAQ, and especially the link I pointed to above, which is the reason why, no, even if you are asking for a feature, you still should not consider that the check list should be entirely optional.

Now, I understand that this refusal to add your request and this reply will probably not fit into what you expected when you placed your request, and may actually make you angry, so I will please ask you to try to carefully consider things from my perspective and understand that it's not because a developer provides an Open Source application for free that they have to bend over backwards to add every user request, especially if, from their experience, the feature requested will most likely only help a very limited number of individuals (the requester and maybe a few others) and actually create more problems than it will solve.

closed 2020-05-09 by pbatard

jmm8344562, 2020-05-09:
Clear