pbatard/rufus #1534: 3 partitions created on flash drive

muv1, 2020-05-09:
After I create a bootable flash drive, if I boot I get this:

~~~
UEFI: Sandisk Cruzer 8.02, Partition 1
UEFI: Sandisk Cruzer 8.02, Partition 2
Sandisk Cruzer 8.02
~~~

I tested all 3. The first 2 seems to work fine, with the second taking some extra steps.
The last option fails.

Why are 3 options available, and what are they for?

Sorry if this has been answered, I searched issues and repo code and didnt see this addressed.

pbatard, 2020-05-09:
Can't help you without a log. I have absolutely __no idea__ how you created the drive.

The check list is __NOT__ optional.

muv1, 2020-05-09:
Is this what you need?

~~~
Rufus x86 v3.10.1647
Windows version: Windows 7 SP1 64-bit
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Notice: The ISO download feature has been deactivated because 'Check for updates' is disabled in your settings.
Found USB 2.0 device 'SanDisk SanDisk Cruzer USB Device' (0781:5530)
1 device found
Disk type: Removable, Disk size: 32 GB, Sector size: 512 bytes
Cylinders: 3894, Tracks per cylinder: 255, Sectors per track: 63
Partition type: GPT, NB Partitions: 2
Disk GUID: {4647BD0B-1EB5-4446-A6FA-8D48FA2D762C}
Max parts: 128, Start Offset: 17408, Usable = 32031832064 bytes
Partition 1:
  Type: Microsoft Basic Data Partition
  Name: 'Main Data Partition'
  ID: {4683D705-9D32-4730-B220-34862715F7FD}
  Size: 29.8 GB (32029724160 bytes)
  Start Sector: 2048, Attributes: 0x0000000000000000
Partition 2 (UEFI:NTFS):
  Type: Microsoft Basic Data Partition
  Name: 'UEFI:NTFS'
  ID: {306CFB91-2273-4381-90AC-58E03A60A975}
  Size: 512 KB (524288 bytes)
  Start Sector: 62560103, Attributes: 0x0000000000000000
~~~

pbatard, 2020-05-09:
Well, I would have preferred the log from how you created the drive, but I'm going to assume that you created a bootable drive from a Windows ISO.

The explanation you seek is [here](https://github.com/pbatard/uefi-ntfs).

The reason why you get 3 options is because your UEFI system supports native NTFS boot, which means UEFI:NTFS is not really needed, and either of the 2 partitions created will work. And it looks like your UEFI adds a 3rd option for the whole device (which is not something that Rufus controls and which if you need further explanation, you will need to ask your computer manufacturer about).

closed 2020-05-09 by pbatard