pbatard/rufus #1537: Page not found 404

ricber, 2020-05-12:
﻿Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [ ] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [ ] I also tried one or more of the following:
  - [ ] Using a different USB drive.
  - [ ] Plugging the USB into a different port.
  - [ ] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
These links are broken:
https://rufus.akeo.ie/files/syslinux-6.03/20171017/ldlinux.sys
https://rufus.akeo.ie/files/syslinux-6.03/ldlinux.sys

Log
---
```
Rufus version: 2.5.799
Windows version: Windows 10 64-bit (Build 18362)
Syslinux versions: 4.07/2013-07-25, 6.03/2014-10-06
Grub versions: 0.4.6a, 2.02~beta2
Locale ID: 0x0410
Found USB 3.0 device 'SanDisk Extreme USB Device' (0781:5580)
1 device found
Disk type: Fixed, Sector Size: 512 bytes
Cylinders: 3892, TracksPerCylinder: 255, SectorsPerTrack: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x85DAA222
Drive has an unknown Master Boot Record
Partition 1:
  Type: NTFS (0x07)
  Size: 29.8 GB (32015998976 bytes)
  Start Sector: 2048, Boot: No, Recognized: Yes
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux/isolinux.cfg' for Syslinux
  Detected Syslinux version: 6.03/20171017 (from '/isolinux/isolinux.bin')
Could not get ISO-9660 file information for file boot/grub/i386-pc/normal.mod
  Could not read Grub version from 'boot/grub/i386-pc/normal.mod'
  Could not detect Grub version
Disk image analysis:
  Image has an unknown Master Boot Record
  Image is a bootable disk image
ISO label: 'Ubuntu 18.04.4 LTS amd64'
  Size: 2126544896 bytes
  Has a >64 chars filename: Yes
  Has Symlinks: Yes
  Has a >4GB file: No
  Uses Bootmgr: No
  Uses EFI: Yes
  Uses Grub 2: No
  Uses Grub4DOS: No
  Uses isolinux: Yes (6.03)
  Uses KolibriOS: No
  Uses ReactOS: No
  Uses WinPE: No
Using image: ubuntu-18.04.4-desktop-amd64.iso
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.03/20171017/ldlinux.sys
Unable to access file: 404
Extended version was not found, trying main version
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.03/ldlinux.sys
Unable to access file: 404
Could not download the file - cancelling
Checking for Rufus updates...
Checking release channel...
Updates: Unable to access version data
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.03/20171017/ldlinux.sys
Unable to access file: 404
Extended version was not found, trying main version
Downloading 'ldlinux.sys' from http://rufus.akeo.ie/files/syslinux-6.03/ldlinux.sys
Unable to access file: 404
Could not download the file - cancelling

```

pbatard, 2020-05-12:
> `Rufus version: 2.5.799`

Your version is way too old. Please update it to latest (https://rufus.ie), as I no longer provide support for 2.x versions.

closed 2020-05-12 by pbatard

ricber, 2020-05-12:
Oh sorry, it was telling me that there weren't new updates.

pbatard, 2020-05-12:
Yeah, the problem is we've had to switch servers because of repeated DDoS, and this effectively brought down some of the old links and the ability to check for update with older versions.

If you're running Windows 10 with 3.10, you should find that the `ldlinux.sys` download should work (the first link will still fail, because it's only meant to be used __if__ the general 6.03 version, i.e. the second link, cannot be applied, which in almost 99.9% of cases it can).