pbatard/rufus #1538: Chinese language mouseover text on bottom left icons in English Lanuage Mode

Underestimated, 2020-05-13:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [x] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
This is a relatively small bug but I thought it best to notify you in case it was unnoticed.
With Rufus 3.10.1647 (windows) the icons in the lower left corner of the window are showing what I believe are Chinese characters. The Language of my system is English and the program I believe is just the standard English one. happens with all of the blue icons, and the text looks to be identical.
Probably an oversight, but thought I'd inform you.
Doesn't impact direct usage, just UI usability.

[image]

Log
---
```
Rufus x86 v3.10.1647
Windows version: Windows 8.1 64-bit (Build 9600)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found USB 2.0 device 'Verbatim STORE N GO USB Device' (18A5:0302)
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1887, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0x6D893C96
Drive has a Windows 2000/XP/2003 Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.5 GB (15521087488 bytes)
  Start Sector: 8064, Boot: No

Image provided: 'C:\Users\David\Downloads\VMware-VMvisor-Installer-7.0.0-15843807.x86_64.iso'
Scanning image...
ISO analysis:
  Image is an ISO9660 image
  Will use '/isolinux.cfg' for Syslinux
  Detected Syslinux version: 3.86 (from '/isolinux.bin')
Disk image analysis:
  Image does not have an x86 Master Boot Record
ISO label: 'ESXI-7.0.0-15843807-STANDARD'
  Size: 350.2 MB (Projected)
  Uses: Syslinux/Isolinux v3.86
    With an old menu.c32
  Uses: EFI
Using image: VMware-VMvisor-Installer-7.0.0-15843807.x86_64.iso (350.3 MB)
Checking for Rufus updates...
Checking release channel...
No new release version found.
Downloading https://rufus.ie/files/syslinux-4.07/menu.c32
File length: 53.8 KB
Successfully downloaded 'menu.c32'
Download signature is valid ✓
[... 145 lines cut ...]
Extracting: G:\vmx.v00 (31.1 MB)
Extracting: G:\vsan.v00 (13.9 MB)
Extracting: G:\vsanheal.v00 (487.6 KB)
Extracting: G:\vsanmgmt.v00 (3.9 MB)
Extracting: G:\weaselin.v00 (1 MB)
Extracting: G:\xorg.v00 (1.4 MB)
Created: G:\syslinux.cfg → /isolinux.cfg
Finalizing, please wait...
Created: G:autorun.inf
Created: G:autorun.ico

Found USB 2.0 device 'Verbatim STORE N GO USB Device' (18A5:0302)
Using autorun.inf label for drive G: 'ESXI-7.0.0-15843807-STANDARD'
1 device found
Disk type: Removable, Disk size: 16 GB, Sector size: 512 bytes
Cylinders: 1887, Tracks per cylinder: 255, Sectors per track: 63
Partition type: MBR, NB Partitions: 1
Disk ID: 0xEFB91D69
Drive has a Syslinux Master Boot Record
Partition 1:
  Type: FAT32 LBA (0x0c)
  Size: 14.5 GB (15524167680 bytes)
  Start Sector: 2048, Boot: Yes

```

pbatard, 2020-05-13:
Thanks for the report. I'm afraid I can't replicate this at all (so, no, this is not a mere oversight that every user of the English version of Rufus will experience):

[image]

So I'm afraid that there is something environmental, that is related to the specific system you are using, to this issue, and that will make it very difficult to troubleshoot. I did create a VMvisor image as well, just in case there was any chance this had to do with the selected ISO, and I also tested with a vanilla installation of Windows 8.1 in case this was related to the OS (the screenshot is from that platform), but still didn't see what you report at all.

As per your check list, you say you ran Rufus on more than one computer (and allegedly saw this issue there too then), so can you please indicate what version of Windows that other computer was using? And since there has to do with the specifics of your environment, can you please provide any details on whether the OSes were you are seeing this issue were set up in a fashion that may deviate from a standard installation (e.g. Corporate environment or extra Windows local policies being applied).

Also, the fact that the text doesn't change from icon to icon (it should) and that you are seeing a 2 characters Chinese string that does not appear to come from any of the Chinese translation messages indicates that this is not a simple matter of somehow picking the message from a different locale, but more likely Rufus getting some semi-random data, and trying to interpret it as a Unicode string (which, in a lot of cases, would produce Chinese characters). So another question I have is: Do you always seem to see the same Chinese characters (including on that other machine you tested with)? Or do they change from one Rufus session to the next? And do you get the same string for all of the small icons (including the checksum icon and the save icon that is displayed when _advanced drive properties_ is showing)?

Of course, I'll be looking at the code to see if there's something like a bad pointer being passed for that message (the process to generate a localized message string is rather generic though, so I would expect that if it fails in one place, it would fail in others), but since none of the environment I have tested allow me to replicate this issue, I will need as many details as you can provide on the circumstances where you see these messages appear...

pbatard, 2020-05-20:
In the absence of a reply from OP, and since I have not seen anything that allows me to identify where this behaviour might be coming from, I will close this issue.

closed 2020-05-20 by pbatard

Underestimated, 2020-05-20:
Apologies, I’ve been laid up for the past week with an injury.
I could only replicate it on the source machine, tried on the other computer and it’s not coming up anymore. Have to conclude you’re right and it must be something in my environment.

pbatard, 2020-05-20:
> Apologies, I’ve been laid up for the past week with an injury.

I'm sorry to hear that. I hope that your injury is healing and that you are getting better.

> I could only replicate it on the source machine, tried on the other computer and it’s not coming up anymore. Have to conclude you’re right and it must be something in my environment.

Thanks for following up. I'll definitely keep that issue in mind if someone reports a similar problem, as it's could be a combination of Rufus + environment and maybe there's something I can do about it. But of course, without being able to replicate the problem in my environments, it's difficult to pinpoint the possible cause.