pbatard/rufus #1539: Rufus 3.10 not detecting drives (elimination them as options)

L0VEFIRE, 2020-05-16:
﻿

Checklist
---------
- [x] I looked at https://github.com/pbatard/rufus/wiki/FAQ to see if my question has already been answered.
- [x] I performed a search in the issue tracker for similar issues using keywords relevant to my problem, such as the error message I got from the log.
- [x] I clicked the 'Log' button or pressed <kbd>Ctrl</kbd>-<kbd>L</kbd> in Rufus, and copy/pasted the log into the line that says `<FULL LOG>` below.
- [x] The log I am copying is the FULL log, starting with the line `Rufus version: x.y.z` - I have NOT removed any part of it.

Additionally (if applicable):
- [x] I ran a bad blocks check, by clicking _Show advanced format options_ then _Check device for bad blocks_, and confirmed that my USB is not defective.
- [x] I also tried one or more of the following:
  - [x] Using a different USB drive.
  - [x] Plugging the USB into a different port.
  - [x] Running Rufus on a different computer.
- [ ] If using an image, I clicked on the `(✓)` button to compute the MD5, SHA1 and SHA256 checksums, which are therefore present in the log I copied. I confirmed, by performing an internet search, that these values match the ones from the official image.

Issue description
-----------------
Rufus is not detecting my drives neither automatically or after I check the advanced option checkbox to show all drives. All other software shows and detects my drives fine, so does the OS.

Screen Recording software comparison detectable :
https://www.dropbox.com/s/92khjneqs0ulenh/20200515_211611.mp4?dl=0

Screen Recording: Rufus Logs
https://www.dropbox.com/s/mxipiqfmh8g62ek/20200515_213110.mp4?dl=0

Log
---
```
<Rufus x86 v3.10.1647
Windows version: Windows 10 64-bit (Build 18363.418)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
0 devices found
0 devices found
Found non-USB removable device 'Samsung Portable SSD X5' => Eliminated
0 devices found
Found non-USB removable device 'Samsung Portable SSD X5' => Eliminated
0 devices found
0 devices found
0 devices found
0 devices found
Found non-USB removable device 'Samsung SSD 970 EVO Plus 250GB' => Eliminated
0 devices found
Found non-USB removable device 'Samsung SSD 970 EVO Plus 250GB' => Eliminated
0 devices found
Found non-USB removable device 'Samsung SSD 970 EVO Plus 250GB' => Eliminated
0 devices found>
```

pbatard, 2020-05-16:
Your devices are non USB __internal__ SSDs/HDDs and, as opposed to other software, Rufus is designed to never list those.

See [this relevant FAQ entry](https://github.com/pbatard/rufus/wiki/FAQ#Do_you_plan_to_support_internal_HDDsSSDs)

closed 2020-05-16 by pbatard

L0VEFIRE, 2020-05-16:
They were listing them before this morning and yesterday, now suddenly they stopped, then if I reinstall the app it will show them again.

L0VEFIRE, 2020-05-16:
is Rufus, not meant for External HDDs / SSD's ??

pbatard, 2020-05-16:
As per the log:
`Found non-USB removable device 'Samsung Portable SSD X5' => Eliminated`

Rufus sees these drives as non USB devices, therefore it won't list them. Maybe you changed something on your system to make them appear as non-USB, whereas they were seen as USB before. But we're only following what Windows tells us here. If Windows says these aren't USB devices, then they won't be listed whether you have external listing of USB HDD checked or not.

For more info, you can try <kbd>Alt</kbd>-<kbd>.</kbd> and post the log.

L0VEFIRE, 2020-05-16:
**Samsung X5 Portable SSD Thunderbolt 3:**

<Rufus x86 v3.10.1647
Windows version: Windows 10 64-bit (Build 18363.418)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Found non-USB removable device 'Samsung Portable SSD X5' => Eliminated
0 devices found
Processing Hub 1:
Processing Hub 2:
  Hub[1] = '\\?\usb#root_hub30#7&9b2e3aa&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[021]: USB\VID_0BDA&PID_8153\9056AB000000
Processing Hub 3:
  Hub[2] = '\\?\usb#root_hub30#5&2396db45&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[096]: USB\VID_05AC&PID_8233\0000000000000000
  Found ID[104]: USB\VID_05AC&PID_8514\CC2938702XRHNW127
  Found ID[116]: USB\VID_05AC&PID_8262\000000000000
  Found ID[119]: USB\VID_05AC&PID_8103\000000000000
  Found ID[122]: USB\VID_05AC&PID_0340\FM7940701WSHYYKA8+RRZ
  Found ID[092]: USB\VID_05AC&PID_8302\0000000000000000
  Found ID[187]: USB\VID_05AC&PID_8102\0000000000000000
  Found ID[160]: USB\VID_05AC&PID_8104\000000000000
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskApple___APPLE_SSD_AP1024717.'
Found non-USB non-removable device 'Apple APPLE SSD AP1024 SCSI Disk Device' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskNVMe____Samsung_PortableEXE7'
Found non-USB removable device 'Samsung Portable SSD X5' => Eliminated
If you *REALLY* need, you can enable listing of this device with <Ctrl><Alt><F>
0 devices found>

**Samsung 970 EVO Plus 250GB** _inside_ Thunderbolt 3 Enclosure

<Rufus x86 v3.10.1647
Windows version: Windows 10 64-bit (Build 18363.418)
Syslinux versions: 4.07/2013-07-25, 6.04/pre1
Grub versions: 0.4.6a, 2.04
System locale ID: 0x0409 (en-US)
Will use default UI locale 0x0409
SetLGP: Successfully set NoDriveTypeAutorun policy to 0x0000009E
Localization set to 'en-US'
Processing Hub 1:
Processing Hub 2:
  Hub[1] = '\\?\usb#root_hub30#7&9b2e3aa&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[021]: USB\VID_0BDA&PID_8153\9056AB000000
Processing Hub 3:
  Hub[2] = '\\?\usb#root_hub30#5&2396db45&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[096]: USB\VID_05AC&PID_8233\0000000000000000
  Found ID[104]: USB\VID_05AC&PID_8514\CC2938702XRHNW127
  Found ID[116]: USB\VID_05AC&PID_8262\000000000000
  Found ID[119]: USB\VID_05AC&PID_8103\000000000000
  Found ID[122]: USB\VID_05AC&PID_0340\FM7940701WSHYYKA8+RRZ
  Found ID[092]: USB\VID_05AC&PID_8302\0000000000000000
  Found ID[187]: USB\VID_05AC&PID_8102\0000000000000000
  Found ID[160]: USB\VID_05AC&PID_8104\000000000000
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskApple___APPLE_SSD_AP1024717.'
Found non-USB non-removable device 'Apple APPLE SSD AP1024 SCSI Disk Device' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskNVMe____Samsung_SSD_970_EXM7'
Found non-USB removable device 'Samsung SSD 970 EVO Plus 250GB' => Eliminated
If you *REALLY* need, you can enable listing of this device with <Ctrl><Alt><F>
0 devices found
Processing Hub 1:
Processing Hub 2:
  Hub[1] = '\\?\usb#root_hub30#7&9b2e3aa&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[021]: USB\VID_0BDA&PID_8153\9056AB000000
Processing Hub 3:
  Hub[2] = '\\?\usb#root_hub30#5&2396db45&0&0#{f18a0e88-c30c-11d0-8815-00a0c906bed8}'
  Found ID[096]: USB\VID_05AC&PID_8233\0000000000000000
  Found ID[104]: USB\VID_05AC&PID_8514\CC2938702XRHNW127
  Found ID[116]: USB\VID_05AC&PID_8262\000000000000
  Found ID[119]: USB\VID_05AC&PID_8103\000000000000
  Found ID[122]: USB\VID_05AC&PID_0340\FM7940701WSHYYKA8+RRZ
  Found ID[092]: USB\VID_05AC&PID_8302\0000000000000000
  Found ID[187]: USB\VID_05AC&PID_8102\0000000000000000
  Found ID[160]: USB\VID_05AC&PID_8104\000000000000
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskApple___APPLE_SSD_AP1024717.'
Found non-USB non-removable device 'Apple APPLE SSD AP1024 SCSI Disk Device' => Eliminated
Processing 'SCSI' device:
  Hardware ID: 'SCSI\DiskNVMe____Samsung_SSD_970_EXM7'
Found non-USB removable device 'Samsung SSD 970 EVO Plus 250GB' => Eliminated
If you *REALLY* need, you can enable listing of this device with <Ctrl><Alt><F>
0 devices found>

L0VEFIRE, 2020-05-16:
Also now that you brought this to my attention, how does a drive become recognized as a "non-USB removable device" if all I am doing is reformatting the drive? in my endless attempts to get windows 10 to work on them....

pbatard, 2020-05-16:
Well, the problem, again, is that these are not listed as using the USB protocol but direct SCSI, on account that they are Thunderbolt devices. If the device path starts with `SCSI` rather than `USB` then Windows is telling us that these are not USB. And because it's difficult to differentiate between internal SCSI disks (typical of internal SATA/NVMe HDD/SSDs) and external ones (typical of Thunderbolt), in order not to have people reformat internal drives by mistake, we will only list USB drives (and some VHD drives) by default, even with the _List USB HDD_ option enabled.

Now, as you can see from the log, I did add an extra __unsupported__ option that may help for these kind of situation, because most internal drives would also be identified as FIXED drives, and non internal ones are usually identified as REMOVABLE (though it is a VERY DANGEROUS assumption to assume that this is always the case, because an improperly configured BIOS/UEFI setup can easily declare an internal SATA drive as REMOVABLE), so you might be able to use <kbd>Ctrl</kbd>-<kbd>Alt</kbd>-<kbd>F</kbd> to list such drives.

But if you do that then you are 100% on your own, because I cannot guarantee that this will not list internal drives and that you won't end up formatting the wrong drive by mistake.

Maybe, if I ever get a Thunderbolt disk, I'll be able to figure out how exactly one is supposed to differentiate between regular SCSI and Thunderbolt. But I don't have that yet.

L0VEFIRE, 2020-05-16:
Thank you for this explanation, thunderbolt has been a trickery topic for me. I appreciate your time in explain all this. Would definitely be interesting to see what you find in comparison with thunderbolt 3. 

Thanks again!

pbatard, 2020-05-16:
Yeah, I'll probably get around to look into it some day, but even the cheapest thunderbolt enclosures I can find seem way more expensive than they should be, so I'm waiting for these to come down in price.

blackcrack, 2020-05-16:
@L0VEFIRE , you can now close the "issue", 
because it's not a issue because is a usb-stick Software 
like you see at the Logo and the executable as logo..

what other say, is not this what's stay in the FaQ and Description of the original Founder and programmer of the  Program. so, if this all, could you close this issue :)